MarlinFirmware/Marlin #2082: SD Card not use SPI mode!!!

rdacomp, 2015-05-15:
Refuse to use SPI mode when using an SD card. The reason is that it is impossible to find the SD card for sale with the regime. They are not !!! You endlessly obsolete!
The problem is very urgent and needs to be addressed rather than closing the eyes !!!

thinkyhead, 2015-05-15:
http://www.marlinfirmware.org/index.php/Reporting_Bugs

thinkyhead, 2015-05-15:
If you would like to help, please explain in more detail. We are all volunteers here also trying to figure out things.

nophead, 2015-05-15:
I think the standard requires SD cards to support SPI mode but some manufacturers like Kingston don't. There is no hardware for parallel mode so you just need to choose a card that supports SPI. I use Sandisk.

rdacomp, 2015-05-15:
I use kit sainsmart.
http://www.sainsmart.com/mega2560-smart-lcd-2004-controller-a4988-ramps-1-4-3d-printer-kit-for-reprap.html
Later configuration.h and attach files that you have requested.
I have tried several types of cards but no one sees it.
Run and look SanDisk absurd. My proposal to solve all problems in software.

nophead, 2015-05-15:
You can solve the problem in software. The non-SPI mode needs different hardware. It is a four bit wide parallel bus.

boelle, 2015-06-01:
i will close this one as its not a marlin problem

closed 2015-06-01 by boelle

rdacomp, 2015-06-01:
The problem is not solved and requires solutions!!!

boelle, 2015-06-01:
its NOT a problem with marlin

bkubicek, 2015-06-01:
then please solve it and come back to us with a solution.

Bernhard

boelle, 2015-06-01:
and if sit pack and keep complaining and expect us to fix everything then you are using the wrong firmware

AnHardt, 2015-06-01:
@rdacomp 
Let me try to say it in other words.

If there will be, in the future, a board, what can support non SPI SD.cards and we have a library for that, we'll try to support it.
As long we have no hardware to support, we simply can't do it. We make software here - not hardware.

rdacomp, 2015-06-01:
The problem is not solved and requires solutions

AnHardt, 2015-06-01:
@rdacomp 
We agree there is a problem. We also agree that this problem needs a solution.
But we disagree that we are the right ones who can provide a solution.