MarlinFirmware/Marlin #2085: Integration resurrection wasp mode

galeots, 2015-05-15:
You can enter in the firmware resurrection wasp mode?
So in case of power failure it can pause and resume printing the extruder to get up so as not to damage the piece and resume printing when you want?

https://www.youtube.com/watch?v=ZmfGlsUGVs0

thinkyhead, 2015-05-16:
I like the concept, and detecting loss of power without a dedicated circuit could be done in some creative ways, such as noticing a sudden switch in state among may components at once. The biggest caveat is that Marlin must support many boards, and they will vary very widely in the amount of time available to process before power to the CPU is lost.

http://www.wasproject.it/w/en/resurrection-system-a-3d-printing-revolution/

ntoff, 2015-05-16:
That looks like it would require extra hardware as well. Also it seems to power the arduino from the 12v supply, what if you've converted it to run off the +5vsb rail so the printer can turn the PSU on and off? ATX PSU's have a power good wire which you could monitor instead. There are so many ways this could be monitored and implemented.

thinkyhead, 2015-05-16:
@ntoff Yes, theirs has a diode in series with the power supply. Of course with an Arduino connected to USB you are correct. It will stay powered from the USB. And since most users now use laptops, it will probably stay powered for a long time. So this is still very useful for those cases where power is lost only to the high current circuits.

ntoff, 2015-05-16:
That's why OP said move the nozzle up and away but THAT would definitely require extra hardware (a UPS) to power the motors.

Still, it should be trivial to sand / file that one damaged spot on a 60+ hour print (someone on reddit just did a 65 hour print) rather than starting all over again from scratch.

Mine are both powered by the 5 volt standby line from the power supply so they work without the PC being hooked up but yeah with the laptop connected the arduino still gets power so there'd be plenty of time for that to save the data or even send the position back to the host and let the host do all the hard work.

thinkyhead, 2015-05-16:
@ntoff In the demo video the delta gets stuck and stays in one spot. Maybe it has a fast-cool nozzle. :8ball:

galeots, 2015-05-16:
I have downloaded, their firmware and I'm trying to figure out the pieces of code in more. if you look in firware in cardreader.cpp function it was planned but not implemented

if (store_location) {
     // future: store printer been, filename and position for continuing a stopped print
     // I know one can unplug the printer and continue printing the next day.
   }
}

galeots, 2015-05-19:
you could integrate into the firmware this thing !? It would be helpful to all in case of low or tebsione and resumed printing when you want.

thinkyhead, 2015-05-20:
@galeots Someone around here will experiment with it when we have more time.

boelle, 2015-06-01:
with our agreement that we should limit the list of supported hardware maybe this one is more easy? and if no one is going to implement it then i think i should be closed...after all we are only 4 that have made a comment about this

but i like the idea too

galeots, 2015-09-26:
I would like to take this project forward with you very helpful to all of you!

Roxy-3DPrintBoard, 2015-09-26:
> I would like to take this project forward with you very helpful to all of you!

This is very complicated for several reasons.  First, the G29 Correction Matrix and Zero would have to be stored so it can be recovered and used.  Without that, a recovery would not be possible.  And you can't re-probe the bed with a part already half built.

Are you thinking about doing the firmware to accomplish this?

galeots, 2015-09-26:
if you download from their site you understand where to add the code!
I'm not good at planning but we must add a paritore sensing the voltage drop ...
http://www.personalfab.it/download/19162/

KiteLab, 2015-09-26:
An here is the hardware part of the story.
http://www.wasproject.it/w/en/resurrection-system-a-3d-printing-revolution/

Roxy-3DPrintBoard, 2015-09-27:
One problem with this is it appears it is aimed at Power Failures.  It would be easier to get everybody to agree that the cost, effort, support issues, etc. made sense if it handled any generic failure causing the print to stop.   For example, a computer crash or a communication failure to the printer.

galeots, 2015-09-27:
as now in the firmware in case of power failure throw it all away! hours in case of printing is a waste! in their gaming system can resume printing the following day if it takes too long!

Roxy-3DPrintBoard, 2015-09-27:
> as now in the firmware in case of power failure throw it all away! hours in case of printing is a waste! in their gaming system can resume printing the following day if it takes too long!

I do appreciate the innovation and use.   Mostly, I'm asking "How hard would it be to make this handle the generic case of any failure?"   If it could be made to handle the generic case, I might personally jump in and help code it.   The big problem right now is we need to clean up and simplify Marlin.  So there are a lot of trade offs that have to be made.

When I get a chance...  I'll check out what they have done and see if there are things we can leverage.

KiteLab, 2015-09-27:
@galeots 
Are you sure you linked to the right software.
Can't find where an extra adc is read.
Can't find a 'resume'-menu.
Can't find writing or reading to/from the sd-card.
Can't find "resurrection" as a string.
Can't find anything relevant

ntoff, 2015-09-29:
The problem still exists that this feature will require at least some sort of power reserve in the form of either a small battery or capacitor bank. There's no way to implement this feature in *just* firmware.

Just hook a UPS to the board, power_good wire to one of the RAMPS pin and when the power good signal goes bad it shuts down the heaters and parks the head. Since power is technically never lost you wouldn't need to re-home (unless the UPS battery goes flat but if they can run a PC for a few hours surely it could keep an arduino  on for a day)

giutrec, 2016-03-07:
wait, the WASP team has just released the version of the firmware for delta wasp open (which is the smaller wooden version of the delta wasp) and do not believe in that version have implemented the resurrection, nor will the Lazarus that are present in commercial versions ....

thinkyhead, 2016-03-16:
Let's move this issue to the MarlinDev repository. It will require a lot of preliminary work generally, because Marlin doesn't have formal support for stopping and resuming a print yet, not at all. We need to develop some good, clean ways to accomplish that first, before even beginning to think about how to catch a power failure and respond meaningfully. For example, what if the print is using bed leveling compensation? Upon restoring power, that compensation would be disabled, so the print would resume in the wrong location. So… lots to do.

studiodyne, 2017-01-08:
Hi , i'm not sure if this thread is solved or anywhere else
I'm just a beginner and i don't have learned all the marlin code yet 
I don't want to be stupid here , but i need to contribute in a solution for all and for dummies
Not all are devellopers , not all understand GITHUB lol , not all are electronician 

I will offer my solution 'for dummies ' 

We have many sources that can apply a pause in this firmware  ok
We know that stopping the motors will not resume their perfect original position
We know that the homing will not resume the original position perfectly 

We have ONLY ONE POSSIBILITY ,  never stop the power , so , we recommand to avoir , hour and hour of programming that will make a dirty print after resuming

So , everybody have to make their own extra power hardware , it's not expensive , it's not needed to be enginner

My solution ,   
1-) An UPS ,  12v/24v battery with diodes on the power 
2-) A 220V/110V switch wired on the ' FILAMENT RUNOT SENSOR ' to make pause when loss
3-) Bed power stop when  FILAMENTRUNOUT active  ' option on the lcd  if we want resuming system '

Ups must be able to produce enough time to wait the power restart '24h is very sure 'in reality  10/12h is enough , but for high security i prefer to wire battery directly by diode on the out of the power supply , we can gain more hours just by this 

This solution is perfect , FOR DUMMIES too , the cost is not expensive , i thing if you have  a printer , you have a home for the printer , you have the money for the filament , you have a computer and INTERNET , you have enough money for one battery and a 220v switch

If stupid comment , please delete

Thks

Roxy-3D, 2017-01-08:
> 1-) An UPS or 12v/24v battery with diodes on the power

My FolgerTech only has a 12 volt supply.  Nothing else comes out of the power supply.   I'm not quite willing to do this, but if I put a small 6" x 4" x 4" tractor battery on the +12 volt power rail, it would probably handle the 20 minute power blinks I get.   Something like this:

http://www.ebay.com/itm/Mighty-Max-12V-12Ah-F2-SEALED-LEAD-ACID-DEEP-CYCLE-RECHARGEABLE-BATTERY/191115358342?rt=nc&_soffid=100&_soffType=SaleAndClearence&_trksid=p5731.m3795

 If I were to do this, I would probably want to adjust the power supply's output to be more like 13 volts just so it is always lightly charging the battery.   That would be a very simple and effective solution for printers with a single 12 volt power supply.

studiodyne, 2017-01-08:
Like  ThinkyHead said eternally lol ,  it's not so easy 
This rapid electronic wiring not offer security and endurance , you know that the printer will be used all your life , so , the electronics of this type of machine , must be very evaluate

If you put , a resistor parallel to the diode , you  have so many risks of problem , if battery is damaged , you can burn the resistor and make fire , if you 'OVERCHARGING' , the battery will heat ,  if the voltage of the battery increase , the diode will conduct and make oscillation alternate with the power supply

Let me compose a future secure schematics , that is  easy to made , secure in time and all fatalité that can appears
I need to find some solutions for non electronicians too
I will , offer the possibility of the battery of your choice , but , my preference is for a car batterie , not expensive and very powerfull ,  a true maker , a passionnate maker will not surrender the objective of the secure printing , i think lol

I promise to make the perfect system , soon, a totaly standalone system for many years of ' NO STRESS PRINTING' , and all in  DO IT YOURSELF PSYCHOLOGY
I have a creator pro , that i have customised ' GREAT and unique OPTIONS'

Now i have a marlin based printer , with MKSGEN7 , and a big buildplate , it's why , i need to secure the printing , because of very long time of printing

Don't worry , i will make it soon ,   and some software revision too ' not for the moment ' but in the future

thks

studiodyne, 2017-01-08:
Look at this
It's exactly what needed
https://fr.aliexpress.com/item/UPS-Uninterruptible-Power-Supply-AC-110V-220V-Battery-Charging-Module-Switching-Power-Supply-Voltage-Regulator-090102/1580759696.html?aff_platform=aaf&cpt=1483888350026&sk=3RRjqFqVV&aff_trace_key=7a40e12dda354ef39088130d02cce717-1483888350026-02554-3RRjqFqVV

But , i have to add the power loss interface , for the filament sensor 

I'm working , don't worry , i will make it soon

studiodyne, 2017-01-08:
The only problem is the low voltage stop system that , stop to produce  if voltage is low down to 10v

The 13.5v is not a diode conductive , but  a system that produce 13.5 with the 12v of the battery

This system is good , but , i want to gain the max time as possible , using the battery low voltage until it dies to 0 v  LOL

studiodyne, 2017-01-08:
http://www.ebay.com/itm/12V-3A-Uninterruptible-Power-Supply-Box-for-CCTV-Security-Camera-System-/311348367962?hash=item487dcf125a:g:5l8AAOSwPhdVPusj

studiodyne, 2017-01-08:
FOR BIG POWER MACHINE 

https://fr.aliexpress.com/item/Switching-Switch-Power-Supply-DC-60V-13-3A-800W-Driver-Transformer-110V-220V-AC-DC60V-SMPS/32706081085.html?spm=2114.30011508.3.10.w0PT4D&s=p&ws_ab_test=searchweb0_0,searchweb201602_5_10065_10068_10000009_10084_10083_10080_10082_10081_10060_10062_10056_10055_10037_10054_10059_10032_10099_10078_10079_10077_10000012_426_10103_10073_10102_10000015_10096_10052_10053_10107_10050_10106_10051,searchweb201603_10,afswitch_2_afChannel,single_sort_0_default&btsid=14b86f68-2ee8-4955-90a1-0c54eb497293

studiodyne, 2017-01-08:
I have made the unique , durable , secure and cheap system , it has all the quality for long time running without problems or surprises

Look the diagram , you need
One battery of your choice , of your choosen price 25$  to more
One low cost battery charger  5$ to more
One power 2 channels RELAY 12v  10$ approx
And some minor fixing and connecting components 

This system , is easy to mount , low cost , professional long time life , zero risks of problems ' in theory '

Look the diagram , it's a very sure system , all other economical system are not secure for the print and for your home  loll

I will use this soon
[image]

studiodyne, 2017-01-08:
after long reflexion about all the possibilities , the low cost requested , the long life without problems , i have decided this diagram , and all other economical system will not be sure . Printing need stable electrical system , no rapid composed diagram
I have gather all the conditions , cheap , easy for dummies , evolutive for each budget , no safety risks , long life durability , no code no upgrade , work with all printer immediatly ' or all have pausing pin '

1 batterie of your choice , 15$ to more
1 charger low cost  5$ or more
1 220v relay  10$ ' high current needed '
1 capacitor
and some connectors/wires easy to find every where

[image]

Here is for me , the unique diagram for the better ups securised home made system for 3d printers

studiodyne, 2017-01-12:
@thinkyhead  and other confirmed devellopers

If we add this code 'only for SD print' 
with ' a little UPS + Very sure mechanical motor 'no move without power' 

- Add Menu Item to activate RESUMING
- Store The name of the printed file  
- Store actual Z pos
- Store 'the number of the gcode line after the last z change layer '  ' Reset of this data on layer change and incrementation after each line for rapid processing between gcode executing
- Go to a  x and y pause location  defined
- Go to a z pause position with upper offset added to actual z stored before ' defined by firmware'

And wait REBOOT , because no idea of the time of the loss power , if UPS dye , a reboot will appear , but if it's a short loss power failure the ups will stay alive  , all is ready to restart 

If ups dye 👍 lol

Case 1 :  Autostart SDFILE file with filename stored    + apply  ' STORED ' X + Y + Z to the motherboard  + forwarding the SDCARD FILE to Z and LINE + Heating , and start gcode 

Case 2 :  ' EXPERIMENTAL '
If we create the reverse HOMING ,  a ENDSTOP Z MAX for homing without collision ?  and restart the forwarded files like i said before

Case3: UPS OR BATTERY and ultra economical mode , no processing all stopped , just the loop or ' interruption ' to wakeup , to have the maximum time of pausing

Can you comment this reflexions , i'm beginner in coding , and i will waste so many time in learning the marlin code ,  but i want to contribute to have the economical ups pausing / resuming  firmware

Thks

studiodyne, 2017-01-13:
@thinkyhead @Roxy-3D  and others

Here are my suggest , the ideas of non confirmed and experiment develloper , but develloper too !

- Mini Ups or 12/24 battery
- A very little power supply , ' Plugged on a free PIN to give the loss information'

The pin make the paused process , the UPS permit to have a good pausing process 
The pause is ok , and we can have the  gcode line number , the z height  , stabilized

- BACKUP 
Now it's possible to make a big dump on the SDCARD , Filename , z , line , Mesh , all to restart

- RESUMING
I have an idea ,  if we create the ' REVERSE HOMING' , we add on our printer , the ZMAX endstop , and we remake the home process , by probing the zmax too
We will have , all the data to make a HOMING by ZMAX endstop ,  the height between the two endstop is on the SDCARD 
You have a perfect homing , you have the date , you can make the exact resuming coordonates to restart perfectly

I don't have enough knowledge of MARLIN labyrinth for the moment , BUT I HAVE IDEAS !! 

@ Everybody 

You dump all what n

boelle, 2017-04-26:
@thinkyhead @Roxy-3D and others... i think this one should be taken away.

very little interest have been shown and it just adds more complexity than its worth.

I would say that if people want this kind of security an UPS is the way to go. you can get an APC unit with expansion port for almost nothing is you get them used. replace the internal batteries and add external ones to suit needs

Grogyan, 2017-10-31:
Recently, and since Prusa brought out their I3 Mk3 with resuming after loss of power, I feel it should be looked at again.
As a UPS is generally too expensive for most people.

Studiodyne, 2018-02-26:
My idea 👍 

**OUTAGE PROCESS**
1-) Ups with enough energy to finish the current layer
2-) A new gcode instruction M1000 , that i place in the layer change of my slicer
3-) A hardware outage detection , on a free pin 
4-) A pause program , that idle the machine , if OUTAGE detectir + M1000 ' layer finish '
5-) The layer is finished , Carriage in pause park 
6-) Program save last height in EEPROM
7-) If power back print restart

If UPS is down then **RESUMING**

1-) A program for resuming :
--- That ask to move manually  the nozzle to find a solid layer to probe z on the object
--- With babystep we will set it perfectly when print restart
**But now we have the problem to retrieve the perfect X/Y homing ...**

Solution 1 --- Ultra high end endstop with high repeatability and the normal homing give good results
Solution 2 --- Use very low feedrate and babystep in realtime for x/y , if the object is easy ok , but if a mad object , impossible to make it in a short time , and without make some traces ...

Solution 3 --- Place jauge comparator on the printer , and before the print , we place the carriage at x0y0 and we make the measurement. When we will resume , we will replace the carriage and with babystep we remake the same value , and print can begin

But , i need a good idea , to babystep without a ton of adventure or anything that is too much ...
Thks for all idea , all are welcome

Grogyan, 2018-02-26:
I don't see why a gcode is needed, when you could just add into the Configuration_adv.h that you have a UPS, so that would auto-magically finish the current layer, then store the next line onto an SD card.

If someone lives in an area where power outages occasionally happen, or regularly, then an EEPROM is a bad idea, as you will wear out said EEPROM.  Besides as the move towards 32bit, most boards do not include EEPROM.

anton-piliugin, 2018-02-27:
Can you test out the solution with "secret voltmeter", i just bought the cr-10s and can't test if it will work and how much time the mc can work without power?
Links on the code examples:
http://www.instructables.com/id/Secret-Arduino-Voltmeter/
https://code.google.com/archive/p/tinkerit/wikis/SecretVoltmeter.wiki

thinkyhead, 2018-02-27:
The CR-10S has a Mega2560, not a 168p or 328p, so it's not likely.

anton-piliugin, 2018-02-27:
I've posted a link on a code for 2560: http://www.instructables.com/id/Secret-Arduino-Voltmeter/

thinkyhead, 2018-02-27:
Aha! Well, that is cool.

Studiodyne, 2018-02-27:
@Grogyan 
Tell me How the machine knows when we make a layer change ... Because in a print , you have zlift , you have fwretract , it's impossible for the machine to know how it's a layer change
It's why we add the gcode instruction in the slicer , not a difficulty and sure and , now , we are sure that the current layer is FINISHED. We are sure that the object have a precise planned surface ,  the best starting point for recovery. When printing , if we have the outage detection , the machine continue to print while appears the M1000 gcode , and now the machine knows the layer is finished and can go in pause park
> Most board does not include EEPROM
I didn't know that ... 😱

Studiodyne, 2018-02-27:
@thinkyhead , my new idea is to use the new sensor or MK3 prusa , and to mount it under the carriage , just for X/Y homing. The optical tracker will scan a fix point of the machine when homing but can be used , to make detection of the edges of an alreadry printed object
The question is ... What is the precision/repetability

I have a ' silhouette cameo ' plotter , and the system is the same , an optical tracker scan the paper to find some printed marks before to start the cutting

Grogyan, 2018-02-27:
@Studiodyne
The machine could know by checking, once power loss has been detected, if the next gcode in the buffer has a Z change, then, you can decide if you want to do an X and Y home, or just move the tool head out of the way, the  to an X and Y home once power has been restored and print resumed. 

Homing accuracy on X and Y depends entirely on your particular endstops, optical tends to be more accurate than a microswitch.

Studiodyne, 2018-02-27:
@Grogyan 
In the situation of a print in dual extrusion mode , with ZLIFTING , and with separate printed object , where we have  Zlift up = G1 zxxx , the layer is not finish .
In my system , we ensure a finished layer , to have a plane surface , for more solving solution , if the object is simple , the user can make a new print of the rest , if the user want to resume , it is possible
And with a totally plane layer , it's possible to make a z homing everywhere

About the endstop... Yes , it's true , it's my last job , find or make some endstop ultra high precision. I'm sure that if I found/make a device with ' 0.05mm' repeatability , i'm winning
But i need the EEPROM + Human intervention for resuming , not automatic mode 
The autoresume will never exist as a system for all machines , only some custom machines can make this

In a case of CNC , the carriage cannot move , because of the guiding screws , and if stopped , we are sure to retrieve the same exact position of all the machine , and it's easy to resume , but on a chip printer , when power is down , the carriage moving by the force of the cable , the filament and more

My project is simple ' ALL MACHINES '  with EEPROM ' of course '+ Ups + OUtage detector hardware

- Power outage -> print while M1000 --> When M1000 go on pause and in reduced bed heat mode + nozzle stop heating

-If ups falls , when reboot the menu tells me 
```

Place manually the nozzle on the object to find the z0
Apply position 
Home x/y 
Restart print 
```

Grogyan, 2018-02-27:
@Studiodyne
You are making this more complicated than it should. 
WASP has working code for an old version of Marlin, that just needs to be worked into the current BugFix.

Using the EEPROM us still not a good idea. I personally do not know if there is enough space left for storing the bytes of the last/next line. Good for quick experiments, but probably not long term.

Checking for Z change once power loss is detected is relatively straightforward.
In my opinion, I am not a dev, nor can speak on their behalf, but I see no need for another gcode be added, just to get a resume going. The layer change means you know the Z height, no need to home that, would be dangerous if settings are not correct, may end up with a broken printer. 

FYI a 3D printer, is CNC (Computer Numeric Control)

Grogyan, 2018-03-06:
I have compared the Creality firmware that Naomi acquired
to
WASP
to
Marlin 1.0.1 (as this is the oldest I can find that is similar enough to compare)

All are about the same (more or less) with no specific functions/calls to a pin, or M112 (emergency quickstop)

So either the functions are already in Marlin, or there is proprietary code to execute saving the last gcode line, and resuming from said line.
However, these functions, by themselves, already exist in Marlin, just not implemented in a way to be automated in the case of power loss.

Studiodyne, 2018-03-07:
I have ordered some ultra high precision x and y endstop   ' METROL 1µ'

I will make my test like this

I make a print with cylinders and square towers , and every 25 layers ' 10mm ' , i will make a G28XY homing , and every  25 layers i will see , if the homing have bad effects on the print
If i have good result , i will start programming this

- One Gcode instruction to place in the slicer , in ' layer change tab ' , to inform Marlin when layer finished
- One feature named ' FL - Resuming '  ' finished layer resuming '

The feature is really simple and need hardware
-one UPS 
-One outage detector on a free pin

If outage 
- Finish the layer or wait the gcode instruction '  Gcode + outage pin = go to pause park and wait '
- When in pause , Z height is stored in eeprom 

if power resumue
- Back to print , delete eeprom Z stored

If UPS down
- On reboot , ask to resume
- If you resume , MARLIN ask you to place your carriage on a zone where it's possible to probe the object surface
- If no bltouch , or inductive or simple endstop , you will positionate yourself  to be in contact with the object
- You confirm the HEIGHT= CONTACT TO OBJECT
And you click resume print at EPPROM Z HEIGHT
And now , the carriage make a G28 X Y ,  and use current z as ' zero ' 
The print start at the beginning of the sdcard , and reject all moves gcode while the G1 z=Z in eeprom
AND PRINT START 

SIMPLE !

The convenient is ,  AN UPS , DIMAFIX GLUE ON THE BED to ensure no ' pop off '

I promise , i will make it soon 
But i will need thinkyhead , to make the storing , and the  SD CARD FEEDING until  z found
The rest i can !

Grogyan, 2018-03-07:
I have begun the slow process of writing up backup and restore functions last night, as there did not seem to be much progress in this area.

Studiodyne, 2018-03-07:
Why finish the layer ? Because , so easy to resume a print if  the current layer from zero , the probing of restart is so easy too 
It's why i  will use UPS ,   because large format printer , need a true solution !!!

Grogyan, 2018-03-07:
It's a good starting point.

Studiodyne, 2018-03-07:
I WILL MAKE IT
and thinkyhead will make it really functionnal for all machine
No need of special mechanicals system

FOR ALL MACHINES
but need

Sensor high end ........ 120$
UPS 200$
outage detector 5$

Grogyan, 2018-03-07:
That is very expensive to a lot of people.
But it is a good starting point

anton-piliugin, 2018-03-07:
Most UPS'es has USB data line to signal if power outage is detected, IMHO. You can try to use UPS connected to a Raspberry Pi + Octoprint for that. But it will be better if someone will test if atmega printer boards have enough power to save the progress on SD card

Grogyan, 2018-03-07:
M27 reports the current byte, getting an accurate location which can be done when SD printing is paused via M25.

Knowing also what the filename was when prematurely stopped is also important, which can be also stores as bytes in the EEPROM.
How much free EEPROM space is there on the Arduino MEGA 2560?

anton-piliugin, 2018-03-07:
EEPROM size in ATMega2560 is 4096Bytes

Studiodyne, 2018-03-08:
Yes .... expensive ... 
But this resume feature is for large format machine.  May be for little if has money.
Large format machine can burn 2 or 3 spools and can play more than 100 hours.
It's a lot of stress to have bo possibility to use the already printed object.

Note: 1 micron are really expensive and ... too much for me for the moment so i will wait a few weeks

Studiodyne, 2018-03-10:
## GOOD NEWS

marcio-ao, 2018-04-09:
@Grogyan: I would be more interested in seeing Prusa's TMC2130 crash detection and recovery working on upstream Marlin. That feature shouldn't require any new hardware!

thinkyhead, 2018-04-11:
Here are all the changes that Creality made for the CR-10S firmware:
https://github.com/brewpirate/CR-10S/pull/1/files?w=1

amrithmmh, 2018-04-13:
this should atleast be added as optional feature along with hardware to detect the power loss
I am from India and power loss is a common thing here so 3d printers are worthless here without  this feature

comps, 2018-04-13:
> Meanwhile, here are the changes that Creality made for the CR-10S firmware:

That looks like a decent SD card destroyer. It seems they don't use any backup files or renames, but rather hope the outage doesn't happen during write and if it does, that a random 8bit value saves the day and cancels the print entirely.

I've been playing with a relay on the 230V inputs, so I can send a logic signal as soon as the power goes out and there's about 1500ms of movement before the printer stops, using a "standard" cheap LED 12V 200W PSU, so there is potential to do something useful even without extra capacitors.

The problem I ran into is coldend mounting - after a short while, my extruder mount melted. So "recovery from power loss" sort of inherently relies of a machine built for it.

[43 more replies]

closed 2018-11-05 by thinkyhead