MarlinFirmware/Marlin #2102: Hardware supported by the community and HW supported by vendors in own fork

fmalpartida, 2015-05-17:
This is a topic covered over the last few Marlin HO where we recognised that a range of "new HW support" has crawled in into the main development baseline supported by the community.

The discussion took us to point to the fact, that supporting such "new HW" was diverting time, effort and resources to supporting identical boards and example configurations that are just customisations of already supported/existing boards.

The main idea behind supporting different boards is to have a wide offering of unique electronics in the base code and vendor specific customisations/configurations should be done on their own fork.

One of the main drivers of the discussion was the exponential raise of "variants" and "example" configurations that are currently finding its way into the main branch. Looking into the future, and if adopted as a trend, we could end up in an unrealistic situation where by the gross amount of files in the repo are "new variants" and "new examples". To make things a bit more fun, new variants introduced have broke the numbering scheme that was in place to identify boards by processor used.

In my mind, a good approach is to thin down the range of supported board to unique boards and have vendors address their particular customisations in their own fork that they publish to their customers. They can always keep up to date and maintain their forks/boards with the latest release branch. This way they are in a good position to maintain their boards and release when they see fit.

As far a examples are concerned, they should be kept as they were, limited to a cartesian, core XY/H-bot, delta, ... Perhaps also addressing Marlin specific feature configurations (autoleveling, LCD configurations, ...) which is more relevant to the community than showcasing a product.

ntoff, 2015-05-17:
I don't know why you can't just make RAMPS 1.4's pin numbering the "standard" and any vendor who makes a board that doesn't comply can go jump in the lake or they can supply their own pins.h for people to use.

fmalpartida, 2015-05-17:
Yes, the is the main idea. With just a that statement the number of versions thins down substantially.

CptanPanic, 2015-05-20:
Yes I agree with this also.

thinkyhead, 2015-05-21:
@fmalpartida The numbering scheme is kind of obsolete, and this was the point of making `BOARDS_MY_BOARD` defines instead of forcing people to use numbers. Yes, lots of posts all over the internet refer to the numbers. But more are appearing all the time that refer to the new defines. So, in the long run I think we should aim to hide the numbers from users and make it so they never need to be used. Then they can be changed to anything that makes sense with minimal disruption. (Another reason to bump the version to 1.1).

fmalpartida, 2015-05-21:
I agree with your statement @thinkyhead.

daid, 2015-05-21:
As a hardware vendor (Ultimaker) I agree with @fmalpartida here.

We maintain our own forks. I did merge basic hardware support for our boards into the mainline. But that's it. I didn't do anything setting wise. And especially the Ultimaker2 has lots of differences compared to the mainline (but still supports the same protocol for compatibility)

I currently have no intention to merge the mainline back into the Ultimaker Original or Ultimaker2 firmwares. Because those machines work fine right now for the users. But users are free to experiment, which is why I do like the board support in the mainline.

thinkyhead, 2015-05-21:
> users are free to experiment

Yet another reason to love UM, besides making the best value printer there is.

ntoff, 2015-05-21:
>But users are free to experiment, which is why I do like the board support in the mainline.

Then it should be up to those users to compare and do their own work. When I first got my 3drag / k8200 I had to figure out the weird pin mappings they'd used when I came over to regular Marlin from the Velleman customized version. I just copied their pins.h file over and voila. I never expected the Marlin devs to accomdate me especially seeing as there was actually a version of Marlin specifically configured for my board.

CONSULitAS, 2015-05-21:
@ntoff 
For pros like you this is correct. But there are many users out there who rely on what they get out of the box.

If you name Velleman: Have a look at this url in my repository https://github.com/CONSULitAS/Marlin-K8200/tree/Vellemann_firmware_k8200_v2.1.1.zip
You will see that their newest release is 911 commits behind RELEASE. They don't support any bug fixing for their users IMHO.
BTW: I backported the zips on their homepage and did the diffs on my own. They don't have any github repository AFAIK.

So if we want to support a broad mass of users we can only try to give out of the box support for the most important printers. But there is a similar discussion on on #1435  - Just my 2cent. 

*Disclaimer: I don't work for Vellemann, I just own one K8200 and try to share my work on it.* :smile:

thinkyhead, 2015-05-21:
@ntoff Well, you do make a good point there, in terms of whether people will feel that Marlin "out of the box" can be put on their printer, so they can try the latest…

ntoff, 2015-05-21:
@CONSULitAS It was still trivial to figure out how to make this Marlin work with my 3drag board. I googled, I read, I modified pins.h beyond all recognition (removed all of it except what my printer needed).

No, it wasn't a direct 'plug-n-play' solution but it *is* the kind of thing you would expect when moving away from a preconfigured solution into a 'bleeding edge' version of the firmware.

Whether or not they fix bugs isn't up to the poor devs here to worry about, nor should it be up to them to play customer service roles to every hardware vendor who just wants to take peoples money and run without supporting them (not a reference to velleman, they're quite helpful on their forums).

daid, 2015-05-21:
@ntoff so which boards should be support in the mainline? Boards that can be used in reprap builds?
As the UM board can be bought and used in other builds just fine (and people do this)

fmalpartida, 2015-05-21:
@daid I am too a HW designer and we could say a vendor of one of the boards. However, I prefer (and my customers for that matter) to have full control of when and how they are updated. The board is maintained up to date on its own fork with its configuration for direct download. Schematics, connections and setup guide to the board have a their wiki entry and also quick setup guide to get people going.

In my case, users are free to download and tailor the latest version from the board's own fork.

fmalpartida, 2015-05-21:
This is a snippet from an entry that I've shared on a similar topic with some insights. I've also posted an entry there to try and divert the attention to this thread whose intention is a bit broader.

---------- 8< --------
Hi folks,

let me bounce these ideas and thoughts over and see how they resonate.

First up we have to recognise that Marlin is a fully configurable firmware that can be customised and tailored to a good range of electronics, printer geometries and setups. 
Based on this fact, the number of possible configurations can grow exponentially and the growth of examples "vendor specific" configurations too. At some point, the core development Marlin team will start dedicating more time on updating vendor specific files than necessary and will end up giving up on them. Just in one weekend, with a simple well targeted email, we could end up with over a dacade "new custom boards" and example configurations. I can foresee too that any broken vendor specific configuration will get a gun report and have the community fixing it.

There are several ways around this that can help vendors help their customers and it very much depends on how they want to implement it:
1. Fork, merger the latest release branch (or any of the others, depending on how suicidel they feel), push changes, their web page link should always be up to date as soon as they push their changes.
2. Get the release branch once published, merge their changes and publish on their website.
3. Any other method that they see fit to support their products.

Following the current branching scheme that is being rolled out it should be very simple.

I can see a good range of advantages and very little disadvantages. To start with:
1. End-less support/maintenance of a deprecated configuration is not maintained forever. How is the marlin team notified that a particular board is no longer in use?
2. Vendors can decide when to pull in the latest changes and what to publish to their customers. They are in full control of the firmware update to their machines or electronics.
3. The Marlin team is not accountable for vendor's specific support. I do think that vendors are accountable for supporting their own products and what they sell more than relying on the Marlin community to do so.

I can think of a good range of additional reasons that do discourage the current practice, that has been recently adopted, and I see very little benefit for the community other than having a vendor specify printer configurations "showcased" in the Marlin repo. Also think that most "new hardware" that has gone into the repo is just a RAMPs with a similar pin mappings or a remix of a different board.

Most people that buy electronics board over internet will have to do a customisation of the firmware at some point anyway(calibration could be one example) or they will have their board shipped from their reliable vendor with its base configuration already in place for their setup.

Contributions, well here you will have to fork anyway if you want to contribute to get your PR processed, so...

Perhaps the question that I am trying to answer is: do we want to gradually become the support site of each vendor or do we want to focus on features, refactoring (one in the pipeline is to create a BSP and and HAL) and defect resolution? 
Personally I think that people choose Marlin for their features, the support that the code gets and the current unique boards it supports more than from being able to copy and paste a configuration/pin definition file from the repo. But again, I could be wrong.

Any way, I wanted to share these thoughts and ideas and see how they resonate.

thinkyhead, 2015-05-21:
> the UM board can be bought and used in other builds just fine

I think Marlin should include the pin definitions and configurations for all well-defined and individually-purchasable boards, since once their pinouts and basic configurations are defined they don't have to change (except keeping configs modernized). And they give us a good set of configurations to build against, making the Travis CI tests more meaningful. (In which light, we should also include configs that enable all existing options for the Travis test.)

Wackerbarth, 2015-05-21:
We need to recognize that "Arduino Pins" Is a Hardware Abstraction Layer, Rather than maintaining this abstraction layer, it is, perhaps, more appropriate to move directly to the HAL which has "Positioner_X" (or something similar) which is then mapped onto specific AVR registers.

thinkyhead, 2015-05-22:
@Wackerbarth Are you saying there is a more generalized system in the Arduino already mapping pins to MCU registers, but we're just not using it? (I'm not finding any good articles on the subject in my Googling so far.)

Wackerbarth, 2015-05-22:
No, if I understand your question, there is a mapping from bits in the MCU registers which we then remap into functional activities.

In particular, there is an Arduino standard that causes signals to appear on the standard Arduino Shield headers. However, "our boards" don't really implement those headers, but go directly to a functional I/O driver.

daid, 2015-05-22:
But there are the same arduino's used with different shields. (RAMPS versions, Ultimaker v1 board are just some examples)

Wackerbarth, 2015-05-22:
@daid -- Yes, but what is your point.  In both cases, we need to go from "Extrude Filament" to a particular bit(s) in some register on the microprocessor.  In your case, where you have drivers on an Arduino-compatable daughterboard, it may make sense to have the intermediate mapping onto the pins on the Arduino headers. However, I have no such headers on any of the three different controller boards that I am using. Each of those is a single board combining the processor and the drivers. Artificially introducing a virtual header is pure nonsense.

daid, 2015-05-22:
@Wackerbarth You don't need it, so it's nonsense, even tough other people need it?

Wackerbarth, 2015-05-22:
No, it is nonsense for you to expect me to pretend that I have those headers. Where YOU actually have them, it is reasonable to have the extra level of mapping so that you can "mix and match" processor boards and daughterboards.  But those details should all be hidden in a HAL.

daid, 2015-05-22:
pins.h is a HAL.

fmalpartida, 2015-05-22:
In parts the pins.h is part of the HAL but there are other levels of abstraction involved. 
One thing that we have to recognize is that there is not really a consistency in the nomenclature, pins.h define pins that are used by fastio and by the native Arduino io library (digital). I think it is worth opening one thread to start discussing and designing such abstraction.

daid, 2015-05-22:
Both fastio and the digital functions in Arduino serve different goals. They are just both interfaces using the same HAL definitions.

Drakelive, 2015-05-22:
Hi

If this abstraction is done well, and if Marlin will be based on chips instead
of Board, it will probably be easier to add 32bit Arm from different
manufacturers. This step is necessary today, sooner or later it will have to
be integrated.

Drk

fmalpartida, 2015-05-22:
@daid, not exactly there are some chips that there is a one-to-one mapping between the pin definition for the fastio interface, however, others have a mismatch. You can clearly see this when using functions that don't toggel IO and generate a PWM (for example).

I do fully agree though that the pin.h and its associated pin mapping is good part of the abstraction.

nophead, 2015-05-22:
You have a broken board support library if that is the case. There should only be one set of pin numbers.

nophead, 2015-05-22:
Well two actually if you include the physical pin numbers.

fmalpartida, 2015-05-22:
Some inconsistencies such as:
#define FAN_PIN            16  // Fan   -- from Teensyduino environment.
#define SDSS               20  // PB0 - 8 in marlin env.

Just to highlight a few examples.

fmalpartida, 2015-05-22:
As we are drifting a bit from the purpose of the thread:
What do we do with both vendor specific configurations and examples?
Drop them?
Encourrage them?

daid, 2015-05-22:
@Drakelive as I said before, adding 32bit support should not be done in the same fork.

nophead, 2015-05-22:
@fmalpartida,
   pins_arduino.h and fastio.h should agree with each other. If they don't when Teensyduino is selected then it is bug in the board support files. You shouldn't need two versions of the pin numbers in pins.h

Wackerbarth, 2015-05-22:
I STRONGLY suggest that configurations be kept. However, except for a LINT version (which has everything), they should be stripped of any and all entries that do not apply to that particular configuration.

Further, it would be helpful if they can be structured so that they "include" a generic version of the basic configuration and then override any unique parameters.

Given the present way that configuration is included, it might make sense maintain those configurations on a special Configuration Branch and handle it somewhat differently with respect to updates.

fmalpartida, 2015-05-22:
What I am more concerned about are specific configurations of the same thing with different mappings. I don't know, it looks very messy as such.

I do think, that they can be maintain by their owner is a separete branch. The day that maintenance ends, board owners will know that their boards are obsolete.

Wackerbarth, 2015-05-22:
@fmalpartida -
I'm not sure what you mean by "different mappings"  -- What are the domain and range of the functions to which you refer?

Here you seem to think that vendors will actually support their boards for the full functional life of the electronics.  It is my experience that such support usually ends well before the hardware is dead.

At that point, I consider the board "orphaned". Then we, as a community, need to have available the latest description of the board (preferably in the for of Marlin configuration files) so that someone might continue to support the boards.  If we have the configuration files within our repository, then there is a starting point should someone wish to "adopt" the board.

fmalpartida, 2015-05-22:
@Wackerbarth if you look at some of the boards that are currently being supported, they are a RAMPs with vendor/printer specific pin mappings for the additions they add to the boards. On the other hand you have "printer specific samples" which are deviating attention from the main core work.

Personally I would help in configuring the electronics through documentation. Most RepRap forums tend to help the noobs on how to setup their printers and electronics.

I am an electronics developer in my spare time with a good customer and distributor base, I am very sure I can collect over the coming weekend a good range of configurations and "printer samples" that could be pushed into this repo with the current status-quo. However, I think that I am accountable for helping them out, document and provide them with a sound configuration on a fork that I maintain specific to the boards. But again, I could be wronge.

thinkyhead, 2015-05-22:
> RAMPs with vendor/printer specific pin mappings

As we find them we try to [squish them](https://github.com/MarlinFirmware/Marlin/pull/2131) but with some boards it's not so easy.

ntoff, 2015-05-23:
>so which boards should be support in the mainline? Boards that can be used in reprap builds?
As the UM board can be bought and used in other builds just fine (and people do this)

Like @fmalpartida wants with the communications protocols - to introduce a standard, I think there should be a pinout standard. So you would have boards made by everyone using the exact same pins for the same functions. Every board would use pin 13 for an LED, pin 8-12 for an LCD, pin 13-40 for the steppers (just examples, I didn't count the number of pins actually needed).

That way there'd only be one pins.h and it'd work with every single board.

Like PC's, you can just plug a video card in from any manufacturer without having to worry about whether your motherboard has the correct pins going to the PCIe slot because some numpty thought it'd be funny to move stuff around.

Wackerbarth, 2015-05-23:
@ntoff - I think that you have exactly the WRONG attitude. Although it might be fine to SUGGEST some standardization, we really need to allow the board designers the freedom to implement their functionality as they see fit.  Carrying your logic to the extreme, the firmware would work on only one particular model of printer (and "knock-off clones thereof).

Instead, I feel that Marlin should be the control firmware of choice for ALL implementations of 3D printers that operate by positioning a nozzle and extruding the building material.

The purpose of a HAL is to hide the details of the design choices such as which "pin" controls the LED (actually the processor registers since there are no pins, as such, in many of the unified controller boards)

fmalpartida, 2015-05-23:
I also think it is a bit constraining too. While controllers, talk in terms of registers, pin.h talks in terms of pin numbers and the code in terms of symbolic constants that refer to those pins. 

I see two levels of abstraction here, first one covers the three statements above, while the last one talks in terms of full devices. 

I like to put the example of the new LiquidCrystal library that with just 2 lines drives LCDs 4 times faster than the stock Arduino LCD library. It uses classes, abstract classes to interfaces. Funny enough the speed is constrained by the LCD driver chip.

closed 2016-04-23 by jbrazio