MarlinFirmware/Marlin #2104: PWM fan with EEB

ruggb, 2015-05-17:
I was writing while u were celebrating - I need that glass of wine u posted - for medicinal purposes -- it doesn't "FULLY" work yet. I managed to get my reply in there so do u need to reopen it or continue here?
for the record here is my post
Sorry, I got excited prematurely......the chg made the fan work PWM but the heaters stopped working.
Come on, there has to be a simple solution for this. If it works for EFB on pin 9, why does changing to EEB and moving FAN_PIN to pin 11 break it. The only thing changing EEB, EFB,,,,, should do is redirect the i/o pins, right??? (aside from whatever has to be done with 2 extruders). So why does redirecting the fan pin with EEB set not work? EEB should automatically redirect the fan and since there is the extender option pin 11 or 6 would be a good choice without servos. AND what is the purpose of FAN_SOFT_PWM option - if it is a FAN option why does it break the heaters??

thinkyhead, 2015-05-17:
I wonder if pin 11 is being used for something else... Do you have any servos?

ruggb, 2015-05-17:
Scott -- let's keep 2046 open all the history is there.

ruggb, 2015-05-18:
If servos r defined 11 is used there. I have no servos and did not chg anything there.
If there is another definition for 11, why does it work when FAN_SOFT_PWM is defined?

the ONLY place I find a write command to fan_pin exists is here in planner.cpp(446):
#ifdef FAN_SOFT_PWM
fanSpeedSoftPwm = tail_fan_speed;
#else
analogWrite(FAN_PIN, tail_fan_speed);//<<<<<<<<<<
#endif //!FAN_SOFT_PWM
#endif // HAS_FAN

So if FAN_SOFT_PWM is defined where/how is it writing fan_pin? which it is since it runs.
there must be another definition for the fan pin and/or another place one of these variables is getting changed
fanspeed, fan, tail_fan_speed, fan_pin

thinkyhead, 2015-05-18:
> if FAN_SOFT_PWM is defined where/how is it writing fan_pin

It's a software routine. It does what PWM does. It set the pin to 255 or to 0, but in pulses.

thinkyhead, 2015-05-18:
> I have no servos

Good, that eliminates one possibility.

thinkyhead, 2015-05-18:
> why does it work when FAN_SOFT_PWM is defined?

Because if something else was - for example - turning the pin off, the soft PWM routine would just keep turning it back on. Not so with the hardware PWM, which (AFAIK) only sets the value once, when it changes.

thinkyhead, 2015-05-18:
> another place one of these variables is getting changed: fanspeed, fan, tail_fan_speed, fan_pin

`FAN_PIN` can't change, as it's not a variable. The `fanspeed` is whatever it says on your controller. There's no `fan` variable. The `tail_fan_speed` variable only exists in the logic of determining whether to apply the current fan speed or the fan speed from a printing session. No, something else is going on.

Are you able to attach an oscilloscope to your fan output?

ruggb, 2015-05-18:
I could - but whatever I would see isn't going to run a fan or turn on an LED. Is there a clue to be had? There is an LED on both outputs so I don't actually need a fan to see an output. If I were getting short pulses that wouldn't run the fan I can see them on the LED. My fan won't run under 20% but the LED blinks.
Let me reiterate the symptoms:
set to EFB the fan works fine on 9
changing fan_pin to 11 (or 6) and setting to EEB causes the fan to only run at 100%. 99% and below is dead except for the kickstart pulse.
defining FAN_SOFT_PWM causes the fan to work fine on 11 (or 6) but the heaters no longer work.
when I clk them on they sit for 5-10 sec then turn off. Not temp rise and I suspect the bed would do the same thing if I had it connected.
I am having a challenge reading this code (I'm not a programmer) but as far as I can tell there is no other 11 or 6 defined.

ruggb, 2015-05-18:
BTW - re fan_pin, I was referring to something else changing the pin# - I did not mean to imply that it was a "variable".

thinkyhead, 2015-05-18:
@nophead @daid Can you can make any sense of this issue? I'm not up on the quirks of overriding pins for the standard RAMPS electronics. Why should Pin 11 with EEB act differently from Pin 9 with EFB?

boelle, 2015-05-18:
no clue, but i have decided to replace my 644 based electronics to ramps again, but it will be 2 weeks befor that happens

but once i had ramps i did not have any issues, prob because i have a very basic printer. only 1 extruder

thinkyhead, 2015-05-18:
@boelle I love my RAMPS. P.S. I wonder, can you please come to my defense in [this topic](https://github.com/MarlinFirmware/Marlin/issues/2090)? It's really bumming me out to be dumped on for making an honest attempt.

daid, 2015-05-18:
@thinkyhead no clue. I've only worked with Ultimaker electronics. (but could be different PMW pins, or some pins not PWMable)

boelle, 2015-05-18:
all i did then was setting my boards as ramps and it worked... cant
remember the suffix but it was of course the one with bed extruder and fan.

if i should do any guessing a file have been changed that screw up what pin
do what

thinkyhead, 2015-05-18:
> could be different PMW pins, or some pins not PWMable

Yes, similar to my thinking. Perhaps the pin is used for a timer in some case, but not in others. Not sure which timer corresponds to Pin 10-11. I'll have to check and see if the source assigns different timers to different hardware.

boelle, 2015-05-18:
maybe the user could post config file on pastebin.com and give link here

and maybe the boards pin file

thinkyhead, 2015-05-18:
@boelle Well the board pins part is easy. It's just a `BOARD_RAMPS_EEB` with one single difference – he is overriding the `FAN_PIN` so that it is 11. So all we need to see is the `Configuration.h` (and, maybe `Configuration_adv.h`).

boelle, 2015-05-18:
yep, my reason for the pins file was that if he have changed that one and somehow made things cross

thinkyhead, 2015-05-18:
> somehow made things cross

That is my general thinking. EEB might be using Pin 11 for something already – but in disguise, either as a timer, or as a digital –rather than analog– pin. Software PWM works, and that is a clue, but it affects heating. The idea is to turn the 2nd "E" in EEB into an "F." Seems to me it would be better to use EFB and just wire the right pins.

boelle, 2015-05-18:
hmm both pins at least are PWM pins... ie i was looking at at mega2560 pin map so that far it should be ok... 

next thing is that we have a few flavours or ramps boards but to my knowledge there is only one board out there.... 

btw... pins 6 and 11 are normmaly for the servos.... yes you can connect a fan to them but they cant drive a fan directly... you have to use a transistor as middle man, ie the pin drives the transistor and the transistor drives the fan... a fan direct will take to much current

boelle, 2015-05-18:
btw... you dont need to change the fan pin either.... 

you just turn on the fan with G-Code... M42 P11 S255 for full on... off you just change S255 to S0

thinkyhead, 2015-05-18:
> a fan direct will take too much current

Ah, interesting. So, a setting of 100 on Pin 11 _could_ actually be more like 255 on Pin 9…

> pins 6 and 11 are normally for the servos

Yeah, but since no servos are enabled it's "free for use." But yeah, if the pins have a different voltage range that could be problematic. Meanwhile, the conflict with the heaters is still a little mysterious.

boelle, 2015-05-18:
not quite so.... PWM is a square wave.... so at the on period it will still take the same current

so S100 is the same on both pins and the same current will flow no matter what pin the fan is connected to.... 

the BIG difference is that D11 and D6 goes straight to the main chip on the mega board, and that one are not made to drive a fan.... 

some PC cpu fans have a 4 pin connector... 2 is for power, 1 is pwm output for sensing the current speed, the last one is a PWM input to tell how fast it should go... pwm out is just used to detect if the fan is blocked, bearings are dead etc...

and a servo pin drains almost no power (close to 0 mA) its just there to tell the electronics inside the servo what angle we want...

to drive a fan from the servo pins it needs to be wired like this: http://i.imgur.com/dYAa5.png

and then use the above M42 command to control the speed.... 

the signal pin on the servo connector would always be chip level voltage... either 3.3V or 5V

fmalpartida, 2015-05-18:
Just to bring some light into the topic. IN The Arduino library set when you do an analogWritre to a pin that is not associated to a timer, it will manage it a a normal digitalWrite. I can't remember the threshold but if memory serves me right, it is 1.
Therefore, no pwm will be synthesized on pins that don't use a timer or it will have to be all SW. 
Having said all this, for the pwm pin to work it has to be in the Arduino pin mapping space, not fastio.

ruggb, 2015-05-18:
this is being used with a fan extender bd. That info was in a previous thread. That bd uses pin 11 or 6 since it is mounted on the servo pins. (it also blocks usage of 4 & 5)

The pins 2-13 are the same type of I/O. The fan works on 11 or 6 in PWM mode if FAN_SOFT_PWM is enabled. But that breaks the heaters.

The fan will go on as long as I make it 100%. Does M42 do PWM? How would u make Sli3r use M42 instead of M106? 

Issuing M42 P11 S255 via Repetier host does nothing.

If EEB were already using 11, why would it work 100% and if FAN_SOFT_PWM is enabled work PWM? Any other use should break that also.

What is the purpose of FAN_SOFT_PWM? Without it, it works fine on pin 9. and EFB. So why is it needed for EEB and pin 11?

boelle, 2015-05-18:
You should not need SOFT pwm at all

Yes M42 P? S? tells what pin and how fast.... P? is the digital pin, it has to be a pwm pin or it will not work. S? is how fast and you can use any number between 0 and 255, where 0 is off and 255 is full on. in slic3r i would disable use of fan... ie in filament settings >>> cooling >>>> remove checkmarks from keep always on and auto cooling

then in printer settings >>> custom  G code 

in the start G code add:

M42 P? S?

and in stop G code:

M42 P? S0

ruggb, 2015-05-18:
@boelle to be clear, currently FAN_SOFT_PWM is not defined.
fan works @ 100% with m106, nothing less.
M42 does nothing - is that a clue?
that would give me constant full fan, that isn't a perfect solution.
I can;t believe I am the only one configuring a printer like this - in fact I know someone else who can't get his working.

boelle, 2015-05-18:
You have to type M42 P11 S155

M42 on its own does nothing

boelle, 2015-05-18:
M106 will be 100%... it just a fan on command

it would be the same as M42 P11 S255

replace 11 with the digital pin number that your fan connects to

ruggb, 2015-05-18:
No, I believe m106 has parameters. It is what slice uses. M107 has none. Just off. I entered the full m42 cmd multiple times with 11 & 6 & 100 & 255---nothing.

boelle, 2015-05-18:
well i use M42 P15 S255 with a fan connected to digital pin 15 on my board
and that one works fine

ruggb, 2015-05-18:
If u don't have an extender bd u can just connect an led &res to the pin to test - it will handle 40 ma. U will definitely see it flashing under maybe 80℅.

ruggb, 2015-05-18:
Try m42 on pins 11;6;5;4 --15 is not a pwm pin and the issue isn't 255 it is 100 or250 or 50

boelle, 2015-05-18:
well i can drive the fan lower than that

but something is not right with your setup... M42 P? S? should always work
as it directly tell what pin to use.

it does not even have to be defined in the pins file

ruggb, 2015-05-18:
As I indicated above even m42 p11 s255 does nothing.

ruggb, 2015-05-18:
R u conf for eeb

boelle, 2015-05-18:
and as i indicated above something is wrong with your setup

ruggb, 2015-05-18:
I will take a fresh copy of 1 0 3 and set
Corexy, eeb, Dir, fan pin, end stops, quickhome, ##the min needed to run. Maybe it is a corexy or quick home issue-which is what this started out as

ruggb, 2015-05-18:
Duh, why do u think I am discussing this with u
Do u have a corexy machine and quick home implemented with 2extruders

thinkyhead, 2015-05-18:
> CoreXY or quickhome

Those are a huge long shot. Those don't have any effect in the area of pins and PWMs, which happens at a lower level. It's best if you test and collect data about the behavior, see how you can affect it, and then the coders (well, me) will know what is most likely implicated.

ruggb, 2015-05-18:
I am trying to do that.........all that I have posted is all that I have RE tests & data. Working on a new clean build.

ruggb, 2015-05-18:
I dnld the latest dev
u changed this in ramps pins line ~88 to 4 which makes no sense since the extender uses 11 or 6.
  #define FAN_PIN            11 // IO pin. Buffer needed //** my change to 11

thinkyhead, 2015-05-18:
You're doing that hard-to-read thing again. Let me decipher, hold on…

Ok, so you're saying _someone_ (possibly "u") might have screwed up the RAMPS configuration for EEB because the FAN_PIN is 4? Well, I can't say. It's been that way for a long, long time, before I even came onto the project, and this is the first bug report about it that I know of.

ruggb, 2015-05-18:
sorry my mistake - but if a commercial bd exists why use 4 instead of 11 or 6?

ruggb, 2015-05-18:
the fan extender bd uses servo 11 & 6 not 5 or 4. I guess the other issue might be - what if it is an EEB using servos? mine is not.

thinkyhead, 2015-05-18:
@ruggb This thread may be of some interest to you on this subject: https://groups.google.com/forum/#!topic/kossel-clear/jlBOrnjHpeM

ruggb, 2015-05-18:
ok, this is a new build using a new dnld today and it works the same.
These r ALL my changes. line # is off by 2 b/c I entered a comment at the top of each file and a line.
fan runs full only. both extruders heat up.  //** is my marker for changed or uncommented line.
It is a Ramps 1.4, w/fan extender bd, and Arduino Mega 2560 R3

==================
Configuration.h(46): #define STRING_VERSION_CONFIG_H __DATE__ "05/18/15 " __TIME__ // build date and time //**
Configuration.h(47): #define STRING_CONFIG_H_AUTHOR "Bill Ruggirello" // Who made the changes. //**
Configuration.h(48): #define STRING_SPLASH_LINE1 "CC-BOT E2 " STRING_VERSION // will be shown during bootup in line 1 //**
Configuration.h(69):   #define MOTHERBOARD BOARD_RAMPS_13_EEB //**
Configuration.h(74): #define CUSTOM_MACHINE_NAME "CC-BOT 2E" //**
Configuration.h(78): #define MACHINE_UUID "277494b7-b7c8-4eb1-a653-c2e695c24911" //**
Configuration.h(82): #define EXTRUDERS 2 //**
#define EXTRUDER_OFFSET_X {0.0, -30.00} // (in mm) for each extruder, offset of the hotend on the X axis //**
Configuration.h(144): #define TEMP_SENSOR_0 5 //**
Configuration.h(145): #define TEMP_SENSOR_1 5 //**
Configuration.h(183): #define EXTRUDER_WATTS (12.0*12.0/3.5) //  P=I^2/R //**
Configuration.h(184): #define BED_WATTS (12.0*12.0/1.1)      // P=I^2/R //**
Configuration.h(301): #define COREXY //**
Configuration.h(323): const bool X_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop. //**
Configuration.h(324): const bool Y_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop. //**
Configuration.h(325): const bool Z_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop. //**
Configuration.h(326): const bool X_MAX_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop. //**
Configuration.h(327): const bool Y_MAX_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop. //**
Configuration.h(328): const bool Z_MAX_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop. //**
Configuration.h(329): const bool Z_PROBE_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop. //**
Configuration.h(330): #define DISABLE_MAX_ENDSTOPS //**
Configuration.h(362): #define INVERT_Y_DIR false //**
Configuration.h(363): #define INVERT_Z_DIR true //**
Configuration.h(391): #define X_MAX_POS 335 //**
Configuration.h(392): #define Y_MAX_POS 305 //**
Configuration.h(393): #define Z_MAX_POS 295 //**
Configuration.h(564): #define HOMING_FEEDRATE {50*60, 50*60, 8*60, 0}  // set the homing speeds (mm/min) //**
Configuration.h(568): #define DEFAULT_AXIS_STEPS_PER_UNIT   {200,200,800,317.48}  // default steps per unit for Ultimaker //**
Configuration.h(569): #define DEFAULT_MAX_FEEDRATE          {500, 500, 8, 25}    // (mm/sec) //**
Configuration.h(570): #define DEFAULT_MAX_ACCELERATION      {9000,9000,100,10000}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot. //**
Configuration.h(616): #define PLA_PREHEAT_HOTEND_TEMP 210 //**
Configuration.h(677): #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER //**

Configuration_adv.h(97): #define FAN_KICKSTART_TIME 100 //**
Configuration_adv.h(213): #define HOMING_BUMP_DIVISOR {8, 8, 4}  // Re-Bump Speed Divisor (Divides the Homing Feedrate) //**
Configuration_adv.h(214): #define QUICK_HOME  //if this is defined, if both x and y are to be homed, a diagonal move will be performed initially. //**

pins_RAMPS_13.h(89):   #define FAN_PIN            11 // IO pin. Buffer needed //**

ruggb, 2015-05-18:
I believe I saw that article a while ago. It appeared to me that anyone who got the fan working did so on D9 with mb 33 defined. I do not see any dual extruder mb 34 (EEB) running a fan on a servo pin - or any other pin for that matter. EXCEPT maybe extruder fans and controller fans, which r not what I need (yet). But I don't even think those r running with a dual extruders.
I want a fan that Slic3r program controls with M106 while running 2 extruders and a bed.
That is clearly what they sell this expander bd for. It says so in the literature.

thinkyhead, 2015-05-19:
What is the part number of the extender board? Is BD the brand name?

ruggb, 2015-05-19:
here is a link
http://www.geeetech.com/wiki/index.php/Reprap_Ramps1.4_RRD_Fan_Extender

ruggb, 2015-05-19:
no BD is short for board

thinkyhead, 2015-05-19:
What happens if you leave all pins as they are, but use `#define EXTRUDER_0_AUTO_FAN_PIN -1` and `#define EXTRUDER_AUTO_FAN_SPEED 255`?

ruggb, 2015-05-19:
AND THE ANSWER IS.........
I found this review:
works great. Needed to find some documentation on the web but did not have too many problems finding it. Control is on D6/D11 as indicated so Marlin board type 34 does not work directly, (Marlin Type 34 controls on D4). But it can be installed shifted two rows of pins towards Aux-1 and will then be on on D4/D5. 

So, I moved the FE over 2 connections so 4 & 5 would be connected instead of 6 & 11.
left everything default:
EEB, quickhome, 2E, temp, fan_pin,.............
WORKS PERFECTLY.
When I put everything back and chg fan_pin to 11 it does not work
chg to 6 - works
chg to 5 - moved fe bd over 2 slots - works
chg to 4 - works

so for some reason pin 11 does not work.
The question is: Is it Arduino or Marlin?
If it is Arduino, it is a defect. HOWEVER, why was Marlin set at 4, is there a conflict with 11 maybe with servo 0?
They appeared to have shifted it b/c Fan_Pin is set to 4 default instead of chg to 11 or 6.
I thought I set it to 6 at one time in the past, but maybe not.

thx

nophead, 2015-05-19:
It's probably because the different PWM capable pins use different timers
and Marlin needs some timers for its interrupts.

ruggb, 2015-05-19:
yeah, I read that but it did not involve pin 11 or pin 4
"The PWM outputs generated on pins 5 and 6 will have higher-than-expected duty cycles. This is because of interactions with the millis() and delay() functions, which share the same internal timer used to generate those PWM outputs. This will be noticed mostly on low duty-cycle settings (e.g 0 - 10) and may result in a value of 0 not fully turning off the output on pins 5 and 6. "

nophead, 2015-05-19:
Pin 4 uses timer 0
Pin 11 uses timer 1
Pin 5 uses timer 3
Pin 6 uses timer 4

Timer 1 is used for the stepper interrupt.

ruggb, 2015-05-19:
where can I find that info
thx

nophead, 2015-05-19:
http://www.arduino.cc/en/uploads/Hacking/PinMap2560big.png

ruggb, 2015-05-19:
perfect - I may have something like that but where is the info as to what the pins r used for. Like how to know which pins use which timers or is that part of the name of the pins?

nophead, 2015-05-19:
For full details see the datasheet for the chip but the simple answer is digital pin 11 is labelled OC1A and the 1 means timer 1.

[26 more replies]

closed 2015-06-01 by boelle