MarlinFirmware/Marlin #2109: Bed level probe not moving in X or Y axis.

marcopifferi, 2015-05-18:
Hi,

The printer is working fine (all axis movement are correct and I am able to print without issue).
I installed a bed probe and enabled the probe pin Z_PROBE_PIN.
Now everytime I issue a G29 command the printer head just move in Z and not moving X or Y.
If I execute G29 V4 I see the printer is trying to probe the correct X Y position but in reality the head is not moving.
That happens with both the GRID or 3Points options.

Any idea what could be wrong ? 

BTW: I am using 1.0.2 version from march 4th.

thinkyhead, 2015-05-18:
I'm not aware of what that could be. It's possibly fixed in the latest code. You should try the newest Development code and see if it has the same issue. But watch out, the configuration files have changed a lot since 1.0.2. It will be easiest if you compare your edited configs to the base 1.0.2 configs to determine the differences, then apply the equivalent changes to fresh configuration files from the Development branch.

If you still have the problem with the latest development code, then we can look into the bug.

I will keep a note on this too, however, so if it's an obvious bug in 1.0.2, maybe we can do a patch later.

marcopifferi, 2015-05-18:
I just tried the current developement version ... exactly the same problem.
It must be my configuration but I do not understand what could be wrong.
Any idea what I could check trying to debug the issue ?

This is the output of G29 V4 (please note the head is not moving among each probe.

Recv: G29 Auto Bed Leveling
Recv: Bed X: 60.000 Y: 60.000 Z: 0.003
Recv: Bed X: 140.000 Y: 60.000 Z: 0.000
Recv: Bed X: 60.000 Y: 140.000 Z: 0.000
Recv: Bed X: 140.000 Y: 140.000 Z: 0.003
Recv: Eqn coefficients: a: 0.00000000 b: 0.00000000 d: 0.00125000
Recv: Mean of sampled points: 0.00125000
Recv: 
Recv: Bed Height Topography:
Recv: +-----------+
Recv: |...Back....|
Recv: |Left..Right|
Recv: |...Front...|
Recv: +-----------+
Recv:  -0.00125 +0.00125
Recv:  +0.00125 -0.00125
Recv: 
Recv: planeNormal x: -0.000000 y: 0.000000 z: 1.000000
Recv: 
Recv: 
Recv: Bed Level Correction Matrix:
Recv: +1.000000 +0.000000 +0.000000
Recv: +0.000000 +1.000000 +0.000000
Recv: -0.000000 +0.000000 +1.000000
Recv: ok

thinkyhead, 2015-05-18:
> I just tried the current development version ... exactly the same problem.

That is strange. I'm trying to think of things that might affect movement in such a way. So, you can confirm that from your Pronterface you can do `G1 X100` and that works fine, and if you move the axis using the LCD controller (if you have one) then it also works. But, when you do a bed probe, the head doesn't move – only the Z axis…

The first thing that might affect XY movement is the `ENDSTOPS_ONLY_FOR_HOMING` setting, if it happened that end-stops were showing up as triggered when they should not. Then, there might be issues with the way that end-stops are being interpreted, which has to do with `ENDSTOPPULLUPS`. And, then there's a setting `[X/Y]_[MIN/MAX]_ENDSTOP_INVERTING` that also affects end-stops. If there are any problems with end-stops, you can figure that out using `M119` to see if they're electrically "triggered" when they shouldn't be.

Can you also post your `Configuration.h` either here or on Pastebin?

marcopifferi, 2015-05-18:
Hello,

Solved !

The problem was that M119 was reporting MAX endstops as activated while MAX endstops are not installed.
Strange thing is that printer is printing fine: apparently that values are just checked there.

thinkyhead, 2015-05-18:
> M119 was reporting MAX endstops as activated while MAX endstops are not installed

What were the settings that you changed to fix it? You're not the first to run into this issue, and I want to make sure that this doesn't keep happening going forward. We recently changed the configurations, and it sounds like some defaults need changing.

marcopifferi, 2015-05-18:
The problem was the default configuration was leaving the pullups and _MIN_ENDSTOP_INVERTING true. That automatically enable all the unconnected endstops.
In my case is not just as simple as disable all the max cause I am using the max for Y axis.
I also have ENDSTOPS_ONLY_FOR_HOMING enabled that is probably the reason why the printer was printing correctly.

BTW: not sure it is related to the same problem but now the printer is probing in correct locations but is not raising beetweeen probes. Any idea what could be the issue ?

thinkyhead, 2015-05-18:
> not sure it is related to the same problem but now the printer is probing in correct locations but is not raising between probes

Yes, we've been having some issues with `G29`. Are you still on Development? If so (and if you can hold out for a day or two) I hope to get it sorted out. Can I call on you to do some testing later on this week?

marcopifferi, 2015-05-18:
I am still on the development and the problem of not rising appear only when you define Z_PROBE_ENDSTOP.
I discover that by leaving the Z_PROBE_ENDSTOP disabled by mistake and I noticed it was rising correctly among probes.
Let me know when you have something to test.

thinkyhead, 2015-05-18:
> leaving the Z_PROBE_ENDSTOP disabled by mistake and I noticed it was rising correctly

@marcopifferi Your mistake might be just what we've been looking for. I will follow that up and I think I'll have a good idea of what's going on soon.

thinkyhead, 2015-05-18:
@marcopifferi I'm curious, when you had `Z_PROBE_ENDSTOP` disabled and the raise worked, was that before or after you fixed the endstop inverting problem? And, connected to that, does your z probe switch trigger properly – with the current inverting setting? The reason I ask this is because the only difference I can see from `Z_PROBE_ENDSTOP` being enabled is that the probe switch gets checked. So, if the switch was being triggered improperly, that might cause it not to raise. I know, it's a long-shot. But really, that's the only difference I can see so far.

marcopifferi, 2015-05-18:
after I fixed the endstop problem.
From the test i made the probe seems to switch correctly. Is there a Gcode command to check the probe status ?
The beahviour I see is that the head move to first probe point then start to move down. As soon as the head is near the bed I see the light turning on on the probe and the movement stop immediatelly so
it seems works correctly.
Then what happens is the head is not moving for a while (3-5 seconds) to move directly to other points without raising and very quickly.
I have a grid of four points and to do the last 3 points is taking less then 3 seconds; it basically move full speed to each point.
Please note the delay is just after the first probed point.

thinkyhead, 2015-05-18:
> Gcode command to check the probe status?

`M119` should now be reporting it, if all is configured rightly.

marcopifferi, 2015-05-18:
Of course...works fine 
Recv: Reporting endstop status
Recv: x_min: open
Recv: x_max: open
Recv: y_min: open
Recv: y_max: open
Recv: z_min: open
Recv: z_probe: TRIGGERED

I double checked and probe is working fine.

The probe and zmin endstop are quite close. Is there any possibility they could interfeer ? I mean what happens if during a probe the Z-MIN endstop is triggered ?

marcopifferi, 2015-05-18:
To be sure is not a zmin issue I move the bed to much higher position with the springs and tried again with G29.
Same issue but now I am 100% sure the Z-MIN is never activated during the probing.

thinkyhead, 2015-05-18:
> Then what happens is the head is not moving for a while (3-5 seconds)

That is very weird. It sounds like a very slow move over a very small distance.

I've got a branch here that is just like the latest code, except it prints a detailed log of all its actions. If you don't mind another flash, would you mind giving it a run using your configurations?

https://github.com/thinkyhead/Marlin/tree/debug_leveling

marcopifferi, 2015-05-18:
Here it is.
```
Send: G29 D
Recv: gcode_G29 >>>
Recv: G29 Auto Bed Leveling
Recv: Running in DRY-RUN mode
Recv: enable_endstops(true)
Recv: z_before = (before) 15.00
Recv: probe_pt >>>
Recv: > ProbeAction:1
Recv: > current_position: (-20.00, 220.00, 0.00)
Recv: Z Raise to z_before 15.00
Recv: > do_blocking_move_to: (-20.00, 220.00, 15.00)
Recv: do_blocking_move_to: (-20.00, 220.00, 15.00)
Recv: > do_blocking_move_to XY: (3.00, 63.00, 15.00)
Recv: do_blocking_move_to: (3.00, 63.00, 15.00)
Recv: > ProbeDeploy
Recv: deploy_z_probe > current_position: (3.00, 63.00, 15.00)
Recv: run_z_probe > current_position: (3.00, 63.00, 2.30)
Recv: <<< probe_pt
Recv: z_before = (between) 5.00
Recv: probe_pt >>>
Recv: > ProbeAction:0
Recv: > current_position: (3.00, 63.00, 2.30)
Recv: Z Raise to z_before 7.30
Recv: > do_blocking_move_to: (3.00, 63.00, 7.30)
Recv: do_blocking_move_to: (3.00, 63.00, 7.30)
Recv: > do_blocking_move_to XY: (83.00, 63.00, 7.30)
Recv: do_blocking_move_to: (83.00, 63.00, 7.30)
Recv: run_z_probe > current_position: (83.00, 63.00, 2.30)
Recv: Raise Z (between) by 5.00
Recv: > ProbeStay > do_blocking_move_to: (83.00, 63.00, 7.30)
Recv: do_blocking_move_to: (83.00, 63.00, 7.30)
Recv: <<< probe_pt
Recv: z_before = (between) 5.00
Recv: probe_pt >>>
Recv: > ProbeAction:0
Recv: > current_position: (83.00, 63.00, 7.30)
Recv: Z Raise to z_before 12.30
Recv: > do_blocking_move_to: (83.00, 63.00, 12.30)
Recv: do_blocking_move_to: (83.00, 63.00, 12.30)
Recv: > do_blocking_move_to XY: (83.00, 143.00, 12.30)
Recv: do_blocking_move_to: (83.00, 143.00, 12.30)
Recv: run_z_probe > current_position: (83.00, 143.00, 2.31)
Recv: Raise Z (between) by 5.00
Recv: > ProbeStay > do_blocking_move_to: (83.00, 143.00, 7.31)
Recv: do_blocking_move_to: (83.00, 143.00, 7.31)
Recv: <<< probe_pt
Recv: z_before = (between) 5.00
Recv: probe_pt >>>
Recv: > ProbeAction:2
Recv: > current_position: (83.00, 143.00, 7.31)
Recv: Z Raise to z_before 12.31
Recv: > do_blocking_move_to: (83.00, 143.00, 12.31)
Recv: do_blocking_move_to: (83.00, 143.00, 12.31)
Recv: > do_blocking_move_to XY: (3.00, 143.00, 12.31)
Recv: do_blocking_move_to: (3.00, 143.00, 12.31)
Recv: run_z_probe > current_position: (3.00, 143.00, 2.31)
Recv: > ProbeStow (stow_z_probe will do Z Raise)
Recv: stow_z_probe > current_position: (3.00, 143.00, 2.31)
Recv: <<< probe_pt
Recv: > probing complete > current_position: (3.00, 143.00, 2.31)
Recv: ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
Recv: Eqn coefficients: a: -0.00001563 b: 0.00004688 d: 2.30125045
```

marcopifferi, 2015-05-18:
It looks like is not rising.
The max Z I see is 7.30.

marcopifferi, 2015-05-18:
Just in case here is another run without the D option (just in case).
```
Send: G29
Recv: gcode_G29 >>>
Recv: G29 Auto Bed Leveling
Recv: enable_endstops(true)
Recv: z_before = (before) 15.00
Recv: probe_pt >>>
Recv: > ProbeAction:1
Recv: > current_position: (-20.00, 220.00, 0.00)
Recv: Z Raise to z_before 15.00
Recv: > do_blocking_move_to: (-20.00, 220.00, 15.00)
Recv: do_blocking_move_to: (-20.00, 220.00, 15.00)
Recv: > do_blocking_move_to XY: (3.00, 63.00, 15.00)
Recv: do_blocking_move_to: (3.00, 63.00, 15.00)
Recv: > ProbeDeploy
Recv: deploy_z_probe > current_position: (3.00, 63.00, 15.00)
Recv: run_z_probe > current_position: (3.00, 63.00, 2.53)
Recv: <<< probe_pt
Recv: z_before = (between) 5.00
Recv: probe_pt >>>
Recv: > ProbeAction:0
Recv: > current_position: (3.00, 63.00, 2.53)
Recv: Z Raise to z_before 7.53
Recv: > do_blocking_move_to: (3.00, 63.00, 7.53)
Recv: do_blocking_move_to: (3.00, 63.00, 7.53)
Recv: > do_blocking_move_to XY: (83.00, 63.00, 7.53)
Recv: do_blocking_move_to: (83.00, 63.00, 7.53)
Recv: run_z_probe > current_position: (83.00, 63.00, 2.53)
Recv: Raise Z (between) by 5.00
Recv: > ProbeStay > do_blocking_move_to: (83.00, 63.00, 7.53)
Recv: do_blocking_move_to: (83.00, 63.00, 7.53)
Recv: <<< probe_pt
Recv: z_before = (between) 5.00
Recv: probe_pt >>>
Recv: > ProbeAction:0
Recv: > current_position: (83.00, 63.00, 7.53)
Recv: Z Raise to z_before 12.53
Recv: > do_blocking_move_to: (83.00, 63.00, 12.53)
Recv: do_blocking_move_to: (83.00, 63.00, 12.53)
Recv: > do_blocking_move_to XY: (83.00, 143.00, 12.53)
Recv: do_blocking_move_to: (83.00, 143.00, 12.53)
Recv: run_z_probe > current_position: (83.00, 143.00, 2.54)
Recv: Raise Z (between) by 5.00
Recv: > ProbeStay > do_blocking_move_to: (83.00, 143.00, 7.53)
Recv: do_blocking_move_to: (83.00, 143.00, 7.53)
Recv: <<< probe_pt
Recv: z_before = (between) 5.00
Recv: probe_pt >>>
Recv: > ProbeAction:2
Recv: > current_position: (83.00, 143.00, 7.53)
Recv: Z Raise to z_before 12.53
Recv: > do_blocking_move_to: (83.00, 143.00, 12.53)
Recv: do_blocking_move_to: (83.00, 143.00, 12.53)
Recv: > do_blocking_move_to XY: (3.00, 143.00, 12.53)
Recv: do_blocking_move_to: (3.00, 143.00, 12.53)
Recv: run_z_probe > current_position: (3.00, 143.00, 2.54)
Recv: > ProbeStow (stow_z_probe will do Z Raise)
Recv: stow_z_probe > current_position: (3.00, 143.00, 2.54)
Recv: <<< probe_pt
Recv: > probing complete > current_position: (3.00, 143.00, 2.54)
Recv: ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
Recv: Eqn coefficients: a: -0.00001563 b: 0.00004688 d: 2.53125000
Recv: planeNormal x: 0.000016 y: -0.000047 z: 1.000000
Recv: set_bed_level_equation_lsq > current_position: (3.00, 143.00, 2.53)
```

marcopifferi, 2015-05-18:
To me it looks the issue is related with this 
  #define Z_RAISE_BETWEEN_PROBINGS 5 

that I have in the configuration file.
The difference beetween the Z_MIN and the probe could be that.
The bed is not very close to the Z_MIN now so just rising 5 from the Z_MIN endstop does not rise anything and that is the effect.
I think at line 2903 of Marlin_main.cpp the z_before should be set relative to the last probed poiint 
or at least make clear in conffiguration that has to be a large enough number to cover the difference.

I am building a new version and trying right now.

marcopifferi, 2015-05-18:
Unfortunatelly that is not solving the issue.
I increased the Z_RAISE_BETWEEN_PROBINGS to 20 and I can see that in the log but the 
probe is not rising.

Here is the log
```
Recv: gcode_G29 >>>
Recv: G29 Auto Bed Leveling
Recv: Running in DRY-RUN mode
Recv: enable_endstops(true)
Recv: z_before = (before) 20.00
Recv: probe_pt >>>
Recv: > ProbeAction:1
Recv: > current_position: (-20.00, 220.00, 0.00)
Recv: Z Raise to z_before 20.00
Recv: > do_blocking_move_to: (-20.00, 220.00, 20.00)
Recv: do_blocking_move_to: (-20.00, 220.00, 20.00)
Recv: > do_blocking_move_to XY: (3.00, 63.00, 20.00)
Recv: do_blocking_move_to: (3.00, 63.00, 20.00)
Recv: > ProbeDeploy
Recv: deploy_z_probe > current_position: (3.00, 63.00, 20.00)
Recv: run_z_probe > current_position: (3.00, 63.00, 2.31)
Recv: <<< probe_pt
Recv: z_before = (between) 20.00
Recv: probe_pt >>>
Recv: > ProbeAction:0
Recv: > current_position: (3.00, 63.00, 2.31)
Recv: Z Raise to z_before 22.31
Recv: > do_blocking_move_to: (3.00, 63.00, 22.31)
Recv: do_blocking_move_to: (3.00, 63.00, 22.31)
Recv: > do_blocking_move_to XY: (83.00, 63.00, 22.31)
Recv: do_blocking_move_to: (83.00, 63.00, 22.31)
Recv: run_z_probe > current_position: (83.00, 63.00, 2.31)
Recv: Raise Z (between) by 20.00
Recv: > ProbeStay > do_blocking_move_to: (83.00, 63.00, 22.31)
Recv: do_blocking_move_to: (83.00, 63.00, 22.31)
Recv: <<< probe_pt
Recv: z_before = (between) 20.00
Recv: probe_pt >>>
Recv: > ProbeAction:0
Recv: > current_position: (83.00, 63.00, 22.31)
Recv: Z Raise to z_before 42.31
Recv: > do_blocking_move_to: (83.00, 63.00, 42.31)
Recv: do_blocking_move_to: (83.00, 63.00, 42.31)
Recv: > do_blocking_move_to XY: (83.00, 143.00, 42.31)
Recv: do_blocking_move_to: (83.00, 143.00, 42.31)
Recv: run_z_probe > current_position: (83.00, 143.00, 2.31)
Recv: Raise Z (between) by 20.00
Recv: > ProbeStay > do_blocking_move_to: (83.00, 143.00, 22.31)
Recv: do_blocking_move_to: (83.00, 143.00, 22.31)
Recv: <<< probe_pt
Recv: z_before = (between) 20.00
Recv: probe_pt >>>
Recv: > ProbeAction:2
Recv: > current_position: (83.00, 143.00, 22.31)
Recv: Z Raise to z_before 42.31
Recv: > do_blocking_move_to: (83.00, 143.00, 42.31)
Recv: do_blocking_move_to: (83.00, 143.00, 42.31)
Recv: > do_blocking_move_to XY: (3.00, 143.00, 42.31)
Recv: do_blocking_move_to: (3.00, 143.00, 42.31)
Recv: run_z_probe > current_position: (3.00, 143.00, 2.32)
Recv: > ProbeStow (stow_z_probe will do Z Raise)
Recv: stow_z_probe > current_position: (3.00, 143.00, 2.32)
Recv: <<< probe_pt
Recv: > probing complete > current_position: (3.00, 143.00, 2.32)
Recv: ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
Recv: Eqn coefficients: a: -0.00001563 b: 0.00004688 d: 2.30875039
```

thinkyhead, 2015-05-18:
> Z_RAISE_BETWEEN_PROBINGS to 20

I assume that's greater than your z probe z offset…?

Looking at the logs, but my eyes are tired. I will pick this up later and scrutinize much closer. So, I assume that if you turn off the `Z_PROBE_ENDSTOP` it behaves as before too…?

marcopifferi, 2015-05-18:
Hi,
I removed a couple of comments that where irrelevants and just left the last one where the issue 
is clearly visible.
This is a part of previous post where the probe is deployed the first time 
then move to next probe location.
We can clearly see the first probe point Z goes from 20 to 2.31 correctly 
then print the z_before that was 20 (correctly) and try to move to 
22.31.
That happens only on the log : the printer never rise about the 2.31 after
the first probe.

Hope that is useful

```
Recv: deploy_z_probe > current_position: (3.00, 63.00, 20.00)
Recv: run_z_probe > current_position: (3.00, 63.00, 2.31)
Recv: <<< probe_pt
Recv: z_before = (between) 20.00
Recv: probe_pt >>>
Recv: > ProbeAction:0
Recv: > current_position: (3.00, 63.00, 2.31)
Recv: Z Raise to z_before 22.31
Recv: > do_blocking_move_to: (3.00, 63.00, 22.31)
Recv: do_blocking_move_to: (3.00, 63.00, 22.31)
Recv: > do_blocking_move_to XY: (83.00, 63.00, 22.31)
Recv: do_blocking_move_to: (83.00, 63.00, 22.31)
```

thinkyhead, 2015-05-18:
So, what happens if you put this line right in the start of the `probe_pt` function?

    st_synchronize();

thinkyhead, 2015-05-18:
So, what happens if you put this line right in the start of the `probe_pt` function?

    sync_plan_position();

marcopifferi, 2015-05-19:
No difference. Still not working.

thinkyhead, 2015-05-19:
Ok, actually we have another solution possibly. I'll patch up my debug branch and let you know when it is testable.

marcopifferi, 2015-05-19:
OK.
BTW I replaced the ZMIN with the probe, removed the probe pin and the calibration works even in debug build.

thinkyhead, 2015-05-19:
So, that's very interesting that it works when your z probe is your only z min. I will consider that as I look at this other solution. Of course that means `Z_PROBE_ENDSTOP` is disabled. Could you post a log with your new setup for comparison?

marcopifferi, 2015-05-19:
Here it is. BTW: Always worked without Z probe and using ZMIN.

```
Send: G28
Recv: gcode_G28 >>>
Recv: enable_endstops(true)
Recv: >>> home_axis(0
Send: G29
Recv: > TRIGGER ENDSTOP > current_position: (0.00, 83.93, 4.00)
Recv: axis_is_at_home 0 > (home_offset[axis]==-20.00) > current_position: (-20.00, 83.93, 4.00)
Recv: > AFTER axis_is_at_home > current_position: (-20.00, 83.93, 4.00)
Recv: axis_is_at_home 0 > (home_offset[axis]==-20.00) > current_position: (-20.00, 83.93, 4.00)
Recv: <<< home_axis(0
Recv: > homeX: (-20.00, 83.93, 4.00)
Recv: >>> home_axis(1
Recv: > TRIGGER ENDSTOP > current_position: (-20.00, 0.00, 4.00)
Recv: axis_is_at_home 1 > (home_offset[axis]==20.00) > current_position: (-20.00, 220.00, 4.00)
Recv: > AFTER axis_is_at_home > current_position: (-20.00, 220.00, 4.00)
Recv: axis_is_at_home 1 > (home_offset[axis]==20.00) > current_position: (-20.00, 220.00, 4.00)
Recv: <<< home_axis(1
Recv: > homeY: (-20.00, 220.00, 4.00)
Recv: > Z_SAFE_HOMING >>>
Recv: Raise Z (before homing) by 1.00
Recv: > home_all_axis > current_position: (-20.00, 220.00, 0.00)
Recv: > home_all_axis > destination: (43.00, 103.00, 1.00)
Recv: >>> home_axis(2
Recv: > TRIGGER ENDSTOP > current_position: (43.00, 103.00, 0.00)
Recv: axis_is_at_home 2 > (home_offset[axis]==0.00) > current_position: (43.00, 103.00, 0.00)
Recv: > AFTER axis_is_at_home > current_position: (43.00, 103.00, 0.00)
Recv: axis_is_at_home 2 > (home_offset[axis]==0.00) > current_position: (43.00, 103.00, 0.00)
Recv: <<< home_axis(2
Recv: <<< Z_SAFE_HOMING
Recv: > (home_all_axis || homeZ) > final: (43.00, 103.00, 0.00)
Recv: ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
Recv: <<< gcode_G28
Recv: ok
Recv: gcode_G29 >>>
Recv: G29 Auto Bed Leveling
Recv: enable_endstops(true)
Recv: z_before = (before) 1.00
Recv: probe_pt >>>
Recv: > ProbeAction:1
Recv: > current_position: (43.00, 103.00, 0.00)
Recv: Z Raise to z_before 1.00
Recv: > do_blocking_move_to: (43.00, 103.00, 1.00)
Recv: do_blocking_move_to: (43.00, 103.00, 1.00)
Recv: > do_blocking_move_to XY: (0.00, 60.00, 1.00)
Recv: do_blocking_move_to: (0.00, 60.00, 1.00)
Recv: > ProbeDeploy
Recv: deploy_z_probe > current_position: (0.00, 60.00, 1.00)
Recv: run_z_probe > current_position: (0.00, 60.00, -0.19)
Recv: <<< probe_pt
Recv: z_before = (between) 5.00
Recv: probe_pt >>>
Recv: > ProbeAction:0
Recv: > current_position: (0.00, 60.00, -0.19)
Recv: Z Raise to z_before 4.81
Recv: > do_blocking_move_to: (0.00, 60.00, 4.81)
Recv: do_blocking_move_to: (0.00, 60.00, 4.81)
[...]
Recv: > do_blocking_move_to XY: (61.00, 60.00, 4.81)
Recv: do_blocking_move_to: (61.00, 60.00, 4.81)
Recv: run_z_probe > current_position: (61.00, 60.00, -0.07)
Recv: Raise Z (between) by 5.00
Recv: > ProbeStay > do_blocking_move_to: (61.00, 60.00, 4.94)
Recv: do_blocking_move_to: (61.00, 60.00, 4.94)
Recv: <<< probe_pt
Recv: z_before = (between) 5.00
Recv: probe_pt >>>
Recv: > ProbeAction:0
[... 77 lines cut ...]
Recv: Raise Z (between) by 5.00
Recv: > ProbeStay > do_blocking_move_to: (61.00, 162.00, 5.25)
Recv: do_blocking_move_to: (61.00, 162.00, 5.25)
Recv: <<< probe_pt
Recv: z_before = (between) 5.00
Recv: probe_pt >>>
Recv: > ProbeAction:2
Recv: > current_position: (61.00, 162.00, 5.25)
Recv: Z Raise to z_before 10.25
Recv: > do_blocking_move_to: (61.00, 162.00, 10.25)
Recv: do_blocking_move_to: (61.00, 162.00, 10.25)
Recv: > do_blocking_move_to XY: (122.00, 162.00, 10.25)
Recv: do_blocking_move_to: (122.00, 162.00, 10.25)
Recv: run_z_probe > current_position: (122.00, 162.00, 0.22)
Recv: > ProbeStow (stow_z_probe will do Z Raise)
Recv: stow_z_probe > current_position: (122.00, 162.00, 0.22)
Recv: <<< probe_pt
Recv: > probing complete > current_position: (122.00, 162.00, 0.22)
Recv: ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
Recv: Eqn coefficients: a: 0.00094262 b: 0.00298203 d: -0.38106598
Recv: planeNormal x: -0.000943 y: -0.002982 z: 1.000000
Recv: set_bed_level_equation_lsq > current_position: (122.00, 162.00, -0.38)
Recv: 
Recv: 
```

thinkyhead, 2015-05-19:
The usual setup is a z max and a probe. Does it behave weird at all if you use your z min as z max, keeping the probe?

marcopifferi, 2015-05-19:
The probe on zmin sometimes get messed up becase it can not zero the z axis on the X Y endstops (it is outside the bed on my printer). It is ok to change the zero position in middle of the bed and I can live with this but at least it should be documented that is not going to work.

ipa64, 2015-05-19:
Hi, 
I got strange behavior too, at first I use only Z_min.

First issue:

If I send this:
G28 X0 Y0
G29
There is no movement from X and Y (They stay at 0,0), the probe tried to find the bed, but it is outside of the bed = crash

The workaround:
G28 X0 Y0
G90
G1 X50 Y50 F4800
G29
If I make an X and Y movement before G29 that is ok, X and Y will move to the first grid position to make measure. 

Second issue:
If before send a G29 probe Z is to high (more than 20mm) the sequence to find the bed give up before the sensor is activated, does we have to make an homeZ before G29?

Thanks for help
If you need I can make test to resolved autobed

marcopifferi, 2015-05-19:
The reason why is not moving is that probably you still have the MAX endstop enabled and the pullup cause them to trigger.
Could you try to do M119 and see the status of your endstops ?

AnHardt, 2015-05-19:
@ipa64 
Also Z has to be homed before G29 will work.
Tri G28 instead of G28 X0 Y0. If that will probe Z outside the bed - experiment with Z_SAFE_HOMING.

ipa64, 2015-05-19:
@marcopifferi Yes you are right, so I have disabled MAX end stops
@AnHardt I will try that
I will come back soon
Thanks

ipa64, 2015-05-19:
Now it's working with firmware 1.0.2 thanks!
What is your advice, stay on 1.0.2 or update to 1.0.3?
Is there some improvement for autobed leveling on 1.0.3 ?

AnHardt, 2015-05-19:
@ipa64 
Stay with 1.0.2. 1.0.3 is still a bit buggy - especially with auto bed leveling.

ipa64, 2015-05-20:
Ok thanks for your support
If you need some test for help to debug 1.0.3 , I can do it

ipa64, 2015-05-20:
Do you think to implement a repeating functions?
Make 3 measures on each points and keep the average.

thinkyhead, 2015-05-21:
> Stay with 1.0.2. 1.0.3 is still a bit buggy - especially with auto bed leveling.

I concur. The last remaining bug in auto-leveling is in our sites, but it will be a couple of days before we get it merged and tested.

thinkyhead, 2015-05-21:
> If you need some test for help to debug 1.0.3 , I can do it.

I'll let you know when the patch has been made, and then I'll hand it off to you and a few others.

thinkyhead, 2015-05-21:
> Do you think to implement a repeating functions?

There is an `M48` "repeatability test" that you can use to see how reliable your probe is.

ipa64, 2015-05-21:
Yes I know for M48 that why I think that can increase accurate by a an average of some measures (for not perfect probe).  in fact I have tested this feature in repetier firmware. 
Ok for patch test, I' m available.

thinkyhead, 2015-05-21:
> that's why I think that can increase accuracy by averaging 2 or more measurements

I'm sure you're correct. I'll have to try adding the option and see how it tests out.

ipa64, 2015-05-21:
Hi, 
I have try to modify Z offset from the display menu, but it does nothing. 
That is annoying because to adjust z_probe_offset_from_extruder you have to flash again by arduino.

thinkyhead, 2015-07-10:
Ok, this branch can be used to get debug output based on leveling code as of July 9…
https://github.com/thinkyhead/Marlin/tree/debug_G29

MarkJB, 2015-11-09:
I'm not sure if what I'm seeing is the same issue or related, but I recently added an IR height sensor to do bed levelling (https://miscsolutions.wordpress.com/mini-height-sensor-board/) . The sensor requires pullups enabled to switch it into 'digital' mode (fixed trigger height rather than giving a 4 level analogue output).

I enabled auto bed levelling in 1.0.2 and the IR sensor starts up in digital mode (so the pull-ups are working) but the z axis wasn't moving up between probes.

I switched to RC2 and unconnected endstops were triggered (and fixed by changing the inverting settings) but now the IR sensor won't start up in digital mode. On the plus side the z axis is now moving up between probes, but the probe isn't reliably triggering (I guess because it is in 4 level analogue mode).And I guess it is still in analogue mode because the endstop pullup resisitors aren't enabled? I've been trying to make sense of the macros...

conditionals.h has this:

 223    /**
  224:    * ENDSTOPPULLUPS
  225     */
  226:   #if ENABLED(ENDSTOPPULLUPS) //This condition would appear to be satisfied
  227      #if DISABLED(DISABLE_MAX_ENDSTOPS) //But this isn't and neither is the min endstop section so endstoppulls are never enabled?!?
  228:       #define ENDSTOPPULLUP_XMAX
  229:       #define ENDSTOPPULLUP_YMAX
  230:       #define ENDSTOPPULLUP_ZMAX
  231      #endif
  232      #if DISABLED(DISABLE_MIN_ENDSTOPS) //This looks like disabling the endstops enables pull ups?!?
  233:       #define ENDSTOPPULLUP_XMIN
  234:       #define ENDSTOPPULLUP_YMIN
  235:       #define ENDSTOPPULLUP_ZMIN
  236      #endif
  237      #if DISABLED(DISABLE_Z_MIN_PROBE_ENDSTOP)
  238:       #define ENDSTOPPULLUP_ZMIN_PROBE
  239      #endif
  240    #endif

Its late and I could be reading this wrong, but it looks like the endstop pullups are never enabled? Disabling the endstops complains about a sanity check... Do I disable the endstoppullups and then manually enable them as needed in configuration.h?

AnHardt, 2015-11-09:
@MarkJB 
The simplest way to get what __you__ want is to don't trust the automatics at all.
Switch off ENDSTOPPULLUPS, DISABLE_MAX_ENDSTOPS and DISABLE_MIN_ENDSTOPS.
Individually set the ENDSTOPPULLUP_? settings to your needs. If one or more of these are unused - don't care about them.
Test the endstops with M119. 
Set the `?_???_ENDSTOP_INVERTING` settings. Unused endstops should never show TRIGGERED. 
Test the used endstops. If you can't alter them, likely the ENDSTOPPULLUP_? setting has to be the opposite. If you can alter the output for that switch, but M119 shows the opposite value you expect alter ?_???_ENDSTOP_INVERTING.

For the probe it's the same.

If you can't move to z=0 because z-min or the probe is triggered before, be sure to have ENDSTOPS_ONLY_FOR_HOMING defined. Despite the 'homing' in the name this also works for 'probing'

Marlin is not prepared for a multi-distance-probe. We only test 1 digital bit.

MarkJB, 2015-11-09:
@AnHardt 
Disabling the end stops (#define DISABLE_MAX_ENDSTOPS, & #define DISABLE_MIN_ENDSTOPS) throws a sanity check error.

/SanityCheck.h:138:12: error: #error You must have a Z_MIN or Z_PROBE endstop to enable Z_MIN_PROBE_REPEATABILITY_TEST.

Commenting out //#define DISABLE_MIN_ENDSTOPS allows me to compile and upload the firmware. Endstoppullups appear to be working as per the fine endstoppullups settings. Endstops all work (including the IR height sensor, but the sensor is still starting up in 4 level analogue mode). 

I wanted to double check the endstop wiring and measure the digital output to see if the pullup resistor was being enabled and interestingly the digital input pin measures 4.4v at the RAMPS when running from USB power and 4.96v with the PSU. If I power up the PSU before connecting via USB the IR height sensor powers up in digital mode. I guess the voltage on USB power alone is just a little too low to set the right mode on the IR height sensor. If I disconnect and reconnect the sensor after powering on via USB, the sensor starts up in digital mode.

I guess I'm going to have to wire in a pull up to force the sensor into digital mode.

I'm not clear on why it worked reliably with 1.0.2 but doesn't work with 1.1.0 RC2... maybe some other external factor. I'll comment again if I figure it out.

AnHardt, 2015-11-10:
>Switch off ENDSTOPPULLUPS, DISABLE_MAX_ENDSTOPS and DISABLE_MIN_ENDSTOPS.

means comment in (for me).

MarkJB, 2015-11-10:
Now I'm confused :)

Commenting in DISABLE_MIN_ENDSTOPS (i.e. removing the comments) results in a sanity check regardless of the state of the other settings... disabling the disable (i.e. enabling the endstops by commenting out the disable) works :)

However, it looks like the problems I was seeing were down to the USB power issue rather than pullup resistors, endstops or settings.

AnHardt, 2015-11-10:
Double negations are not a simple thing. ;-(
```
#define DISABLE_MAX_ENDSTOPS
```
means, the option is on, the switches are off.

The transition from uncommented (above) to commented (below) is 'commenting in'.
The transition from commented (below) to uncommented (above) is 'commenting out'.

```
//#define DISABLE_MAX_ENDSTOPS
```
Now the option is off and the switches are on.

Roxy-3DPrintBoard, 2015-11-10:
> Double negations are not a simple thing. ;-(

Especially when you are not dealing with your native language!    But you speak English so well nobody even realizes that!

MarkJB, 2015-11-10:
I always thought commenting in meant to remove the comment token so it becomes an active part of the code. I.e. it is now 'in' the code.
And to comment out was to remove something from active code by putting a comment token in front of it to take it 'out' of the active code.

Maybe its just easier to say commented and uncommented :)

AnHardt, 2015-11-10:
@Roxy-3DPrintBoard 
```
	  /-_-\
	 /  /  \
	/  /    \
	\  \    /
	 \__\__/
	    \\
	    -\\    ____
	      \\  /   /
	____   \\/___/
	\   \ -//
	 \___\//-
	    -//
	     \\
	     //
	    //-
	  -//
	  //
	  \\
	   \\

```
Thanks.
But i know my limitations.

closed 2015-11-10 by AnHardt