MarlinFirmware/Marlin #2112: PID Autotune Last Line Number Error

clefranc, 2015-05-18:
Hey, just tried the PID autotune on my bed and hotends, get this error at the end:

```
09:27:39.957 : PID Autotune start
09:29:14.581 : bias: 185 d: 69 min: 99.17 max: 100.26
09:29:42.105 : bias: 185 d: 69 min: 99.19 max: 100.53
09:29:42.109 : PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
09:29:42.110 : #define  DEFAULT_bedKp 0.00
09:29:42.111 : #define  DEFAULT_bedKi 0.00
09:29:42.112 : #define  DEFAULT_bedKd 0.00
09:29:42.113 : ok N15 P3
09:29:43.137 : wait
09:29:44.139 : wait
09:29:45.141 : wait
09:29:45.171 : Error:Line Number is not Last Line Number+1, Last Line: 24
09:29:45.172 : Resend: 25
09:29:45.178 : ok N24 P4
```

Anyone noticed that? Normal?

I'm using #2077 and Repetier-Host.

AnHardt, 2015-05-18:
@clefranc 
Communication errors my occur randomly. If you get them regularly at the same time - g-code we have to investigate this.
An other thing here makes my more nervous. The Kp, Ki, Kd values look unhealthy to me.

MagoKimbra, 2015-05-18:
the error is normal it always after pid tuning. I tried to fix it but I could not ... I have even more in the writing to direct values in the variables, so only do m500 without recompiling the fw. I thought that, but it is not so ...

clefranc, 2015-05-18:
@AnHardt It seems pretty consistent to me, each M303 gave me a line number error.

@MagoKimbra Thanks for the feedback, just found this odd. Going directly to the EEPROM would be great.

thinkyhead, 2015-05-18:
> each M303 gave me a line number error

Who doesn't love a good puzzle? Who can figure it out first? Go!

thinkyhead, 2015-05-18:
> each M303 gave me a line number error

Now that I think about it, #2113 might actually fix that.

thinkyhead, 2015-05-18:
While I've got you all here, what's the consensus on the "wait" message? I have it now turned off by default in the configuration because the Pronterface console fills up. Do newer versions of Pronterface suppress "wait" – or should they, I wonder? It's very useful, if only it wasn't so visible.

clefranc, 2015-05-18:
@thinkyhead Honestly, I don't know how I got the 0.00, I was just playing with PID autotune, so it was a bad example to post.

thinkyhead, 2015-05-18:
> I don't know how I got the 0.00

Whew :exclamation:

MagoKimbra, 2015-05-18:
@clefranc and @thinkyhead 
I add in void PID_autotune(.....) this:

if (cycles > ncycles) {
      ECHO_LM(DB, MSG_PID_AUTOTUNE_FINISHED);
      if (hotend >= 0) {
        PID_PARAM(Kp, hotend) = Kp;
        PID_PARAM(Ki, hotend) = scalePID_i(Ki);
        PID_PARAM(Kd, hotend) = scalePID_d(Kd);
        updatePID();

In this way it upgrades directly the values in memory and message MSG_PID_AUTOTUNE_FINISHED I added that you can give the command to save the current parameters M500 ...

clefranc, 2015-05-18:
Sorry for giving you cold sweats guys.

I'll try #2113 tonight, before my night shift, and also try @MagoKimbra code if I'm awake enough.

Perhaps there will be a new PR when I woke up...

MagoKimbra, 2015-05-18:
This is old code:
if (cycles > ncycles) {
      SERIAL_PROTOCOLLNPGM(MSG_PID_AUTOTUNE_FINISHED);
      const char *estring = extruder < 0 ? "bed" : "";
      SERIAL_PROTOCOLPGM("#define  DEFAULT_"); SERIAL_PROTOCOL(estring); SERIAL_PROTOCOLPGM("Kp "); SERIAL_PROTOCOLLN(Kp);
      SERIAL_PROTOCOLPGM("#define  DEFAULT_"); SERIAL_PROTOCOL(estring); SERIAL_PROTOCOLPGM("Ki "); SERIAL_PROTOCOLLN(Ki);
      SERIAL_PROTOCOLPGM("#define  DEFAULT_"); SERIAL_PROTOCOL(estring); SERIAL_PROTOCOLPGM("Kd "); SERIAL_PROTOCOLLN(Kd);
      return;

change in:
if (cycles > ncycles) {
      SERIAL_PROTOCOLLNPGM(MSG_PID_AUTOTUNE_FINISHED);
    if (extruder >= 0) {
       PID_PARAM(Kp, extruder ) = Kp;
       PID_PARAM(Ki, extruder ) = scalePID_i(Ki);
       PID_PARAM(Kd, extruder ) = scalePID_d(Kd);
       updatePID();
   }
   return;

thinkyhead, 2015-05-18:
@MagoKimbra I like it very much, but I think it deserves an extra parameter on `M303`. It's possible that someone may want to run the test and see the results, but not store them. And it's possible someone may still want to copy-paste the code into their configuration too.

MagoKimbra, 2015-05-18:
It puts them in the volatile values, but unless it switches M500 store them in EEPROM.

thinkyhead, 2015-05-18:
> It puts them in the volatile values

Well, yes. They _become active_ is what I meant to say.

MagoKimbra, 2015-05-18:
A ok...

thinkyhead, 2015-05-18:
I wonder if the line number error is just because it takes so long to do `M303`. Hosts don't use `M303` at all and it shouldn't be used in a gcode file, so perhaps `M303` can send back its `ok` before it starts and that will fix the error…

MagoKimbra, 2015-05-18:
No same problem... 
18:33:15.140 : <MK4>: PID Autotune start
18:33:15.140 : <MK4>: Hotend: 0 Temp: 170.00 Cycles: 5
18:33:15.140 : ok                                                                  <----------------- send ok
18:34:22.862 : bias: 137 d: 117 min: 165.71 max: 171.37
18:34:32.944 : bias: 137 d: 117 min: 166.87 max: 172.15
18:34:45.423 : bias: 115 d: 115 min: 168.84 max: 174.32 Ku: 53.44 Tu: 12.47
18:34:45.423 : Classic PID
18:34:45.423 : Kp: 32.07 Ki: 5.14 Kd: 49.98
18:34:57.002 : bias: 101 d: 101 min: 169.24 max: 174.20 Ku: 51.82 Tu: 11.59
18:34:57.002 : Classic PID
18:34:57.012 : Kp: 31.09 Ki: 5.37 Kd: 45.04
18:35:07.714 : bias: 96 d: 96 min: 169.06 max: 173.22 Ku: 58.76 Tu: 10.71
18:35:07.714 : Classic PID
18:35:07.714 : Kp: 35.26 Ki: 6.59 Kd: 47.18
18:35:07.724 : <MK4>: PID Autotune finished! Put the last Kp, Ki and Kd constants from above into Configuration.h or send command M500 for save in EEPROM the new value!
18:35:07.724 : <MK4>:  Kp: 35.26 Ki: 6.59 Kd: 47.18 <------------------ finish
18:35:07.724 : ok N13 P15 B3                                 <------ 
18:35:08.763 : wait                                                  <-----
18:35:09.754 : wait
18:35:10.770 : wait
18:35:11.010 : Error:Line Number is not Last Line Number+1, Last Line: 22 <-------------- problem
18:35:11.010 : Resend:23
18:35:11.030 : ok N22 P15 B4
18:35:12.170 : wait

thinkyhead, 2015-05-18:
@MagoKimbra Maybe calling `ok_to_send()` instead of just echoing "ok" makes a difference?

MagoKimbra, 2015-05-18:
I think I found the problem ...
I hope to explain myself with my bad English.
Host (repetier) makes a request for M105 every few seconds depending on how it was set. The fw actually can not answer because engaged in PID autotune.
But every two seconds in the routine simulates the response:

// Every 2 seconds...
    if (ms > temp_ms + 2000) {
      int p;
      if (hotend < 0) {
        p = soft_pwm_bed;
        ECHO_SMV(OK, MSG_B, input);
        ECHO_EMV(MSG_AT, p);
      }
      else {
        p = soft_pwm[hotend];
        ECHO_SMV(OK, MSG_T, input, 1);
        ECHO_EMV(MSG_AT, p);
      }

      temp_ms = ms;
    } // every 2 seconds

This means that the host can draw the graph of temperature. But in fact it is not a real answer to the request M105, but only a simulation.
If i block the sending of that response all works well, because the host after a while he stops and waits for answers M105, of course you do not see any change on the chart, but all is well.
But with the answer by mistake because it can happen, who eventually is that the Host and fw simultaneously send, one request M105 and the other the virtual response .. This creates the command error ...

I increased the time to 5 seconds the error shall be the same, but on a line more distant from the start autotune ...
Decreasing to 1 second instead of the error almost immediately, but it all depends on the time set in repetier for control M105 ...
We should do so, the answer is only a result of the command M105 and not random ...
I hope I was clear.

MagoKimbra, 2015-05-18:
No, the problem is in the counter command ... That of course fw has blocked at command M303 while that of the host went ahead because he received his answer ...

MagoKimbra, 2015-05-18:
Only solution i found is this:
case 303: // M303 PID autotune
        gcode_M303();
        gcode_LastN += 1;
        FlushSerialRequestResend();
        break;

Now only see Resend: xx
but not see error....

thinkyhead, 2015-05-19:
If we made Marlin into a proper state machine, TUNING could be just another state, and we could still handle host requests while it is ongoing. Maybe in the future Marlin will be a little more amenable to that kind of behavior.

thinkyhead, 2015-05-21:
#2136

AnHardt, 2015-07-18:
Resolved by #2136

closed 2015-07-18 by AnHardt