MarlinFirmware/Marlin #2114: Defaults for these settings backward?

thinkyhead, 2015-05-18:
Recent changes to simplify the configurations introduced a new quirk: Non-existent MAX end-stops show up as triggered by default. So, we need to fix that one.

And, these are now the defaults for axis directions:
```
#define INVERT_X_DIR false
#define INVERT_Y_DIR true
#define INVERT_Z_DIR false
```
…but on my Prusa i3 the axis directions are the exact opposite. So, I assume that to be more proper for most RepRap machines they need to be changed too. What is the consensus?

nophead, 2015-05-18:
Hasn't it always been the case that RAMPS defaults to six endstops and you have to modify it or add jumpers if you have three? Certainly that is the case with my fork from 3 years ago.

The invert dir just depends on how you orient your motors, which side of the belt you take the drive and how you wire them. I don't think there is any real standard. All machines are different.

thinkyhead, 2015-05-18:
@nophead Well, the issue seems to be that some users migrating from previous versions are getting triggered max endstops and are just seeing this as a failure for axes to move. It's not obvious from their pov that the endstop settings need to be changed.

nophead, 2015-05-18:
Yes but I have seen these questions for years so I am questioning whether anything has actually changed recently. If it did it must have changed to being disabled and then reverted.

thinkyhead, 2015-05-18:
@nophead Actually (whoops!) I was looking at my own config, where I have them set to `true` while leaving the active ones `false`. I've updated the OP to fix my error.

So, anyway, the defaults are to have all the __inverting as `false`__ and then both `DISABLE_MAX_ENDSTOPS` and `DISABLE_MIN_ENDSTOPS` are commented out. So all 6 end-stops are active in the default config. If that seems fine in your view, I'm satisfied to leave it alone.

The axis directions are less important. For all I know my motor connectors are all backward.

nophead, 2015-05-18:
It does cause a lot of confusion to RAMPS users and I don't think there are very many 3D printers that have six endstops as there is little point. However changing the default might break machines that have XMIN, YMIN and ZMAX.

thinkyhead, 2015-05-19:
@nophead True. Well, we have to write up a document about configuring Marlin anyway. We'll just include an extra note about end-stops in the troubleshooting tips.

nophead, 2015-05-19:
You could also error or if min and max are triggered at the same time.

thinkyhead, 2015-05-20:
@nophead That's an idea. It wouldn't happen until the first HOME, but that would be a good way to catch a problem!

nophead, 2015-05-20:
Or you could do it as a special case during initialisation as they should never be triggered together.

thinkyhead, 2015-05-21:
Food for thought…

closed 2015-05-25 by thinkyhead