MarlinFirmware/Marlin #2121: Marlin branching scheme - proposal

fmalpartida, 2015-05-18:
This is the first entry regarding the branching strategy that can be implemented in the RepRap repo now that it has been reactivated and which has generated a bit of confusion.

The main intention of this entry is to is to set alignment, collect ideas and see how to manage corner cases before fully rolling it out in the main Marlin repo and avoid misunderstanding with what goes where.

I will try to edit the full text here before publishing it in the main Marlin page.

For today I will try to set the intent and see how it goes.

————————8< ———————— 

This repository contains the code base for Marlin’s firmware. Its a place for users to download the latest release that the team has put together, where the next release is cooking and new developments happen. 

If you are a Marlin user and you are after the latest stable release version, please download it using the “Download ZIP” button in this page. We will be making periodic updates to the main codebase as they are released to the general public. You will also find releases in the release tab section.

If you are looking to contribute to the project, there are a few places that you can do so but before you have to understand how changes are processed before they are released. Please continue reading to understand the uses, and meaning, that each branch has.

## Branches
Branches have been organised in promotional levels (or maturity level); from simple unit testing of certain feature, bug fixing or refactoring to a fully tested release by the community. There are 3 main branches following this scheme.

**Release branch**
This is the latest and greatest code base available to the public. The code is reasonably stable a fit for general use and downloads. The code is not bug free, but it is the most reliable and feature pack version that is available to the public. Only critical bugs will be resolved in this branch which shouldn’t find their way here.

**Integration branch**
This is the latest beta quality version available that is being built to become a future public release. The code base in this branch is still under tests by the community where each release is being constructed for general public use. Their are several users to this branch: 
- Marlin testers who are taken on the latest developments and trying them out on their setups.
- Early adopters who don’t mind finding a defect on their way and are willing to tradeoff reliability with feature content.
- Just want to use the latest in the works.
The main guide is to kindly report any defect you find in this branch so that it can be address prior to the public release.

**Development branch**
This is where developments from contributors are pulled in from their own forks. The main conditions to accept a pull request is that the codebase compiles with the default Marlin configuration and that the code has been tested in the developer’s setup. The code base here is very lightly to be unreliable. So if you are a developer, want to contribute and share your code or you are feeling adventurous, this is the branch for you. The main guide is to kindly report any defect you find in this branch so that it can be fixed prior to going into the integration branch.

## Working on branches
TO DO

## Promoting code on branches
TO DO

## Versioning

The version can be found in the configuration.h file under "#define STRING_VERSION". If the configuration file does not have a version, you are using a pre or 1.0.2 release.

- *Release branch*: `x.y.z Rel`
- *Integration branch*: `x’.y+1.z’` Integ, where `x’.y+1.z’` is the future release that will replace `x.y.z`
- *Development*: `x’.y+1.z’` Dev, where `x’.y+1.z’` is the future release that will replace `x.y.z`

and
- `x` - mayor number, denotes a mayor change in the entire codebase.
- `y` - mid number, denotes a new release that may include new features, code refactoring and bug fixing that has gone through testing.
- `z` - minor number, denotes a minor release that can be a bug fix over the actual release branch.

For example:
- *Current release branch*: `1.0.2 Rel` (perhaps we can do without the Rel qualifier).
- *Integration branch*: `1.1.0 Integ`
- *Development Branch*: `1.1.0 Dev`

Indicating that development and testing taking place contributing to the new 1.1.0 release.

AnHardt, 2015-05-19:
To limit the confusion of the first days we already have a "//#define STRING_VERSION 1.0.2" in the main `Configuration.h` of `Release`. We also changed to "#define STRING_VERSION 1.0.3 __int__" in all `Configuration.h`'s in the `Integration` branch.
@fmalpartida :thumbsup: :clap::clap::clap:

AnHardt, 2015-05-19:
May i suggest:
* Labels for `PR Release` and `PR Integration`.
* At least 2 Persons needed to merge to `Integration`. One for the PR - an other for the merge.
* At least 3 Persons for `Release`. PR - agreement (signoff) - merge.

thinkyhead, 2015-05-19:
How do you propose to merge into Integration from a single Development branch? It's not easy to bundle commits out of order.

CONSULitAS, 2015-05-19:
:+1:

If we look on the current dev branch we are not on 1.0.3 dev anymore, but at least on 1.1.0 dev or  2.0.0 dev.

If I follow @fmalpartida s proposal it should be 1.1.0. E.g. ABl, thermal protection, long filename host support, language support.

After we have resolved the issues with the communication protocol (see #2014, #2000 and so on), it will be 2.0.

What do you think: Should we move to 1.1.0?

thinkyhead, 2015-05-19:
I agree, 1.1 is appropriate because we have changed the configs a lot.

fmalpartida, 2015-05-19:
Hi folks,

thanks for the feedback. I will continue to elaborate the entry in these lines. Here is where some of the gritty stuff gets on its way.

With respect to labels on the integration branch, I am not entirely sure. Do you know if the get published in the main repo in the release tab? If they do, I would be reluctant to have them published as people may download a version that in reality is under construction. Can labels be removed or better, can they be removed from the download section? 

I do agree with all of you, the version we are cooking should be 1.1.0, by definition is a large update where changes apply all over the shop.

@AnHardt, that is the idea. To transition from the development branch to the integration branch, we will need to PR and 3 for the release. Each of which should happen when certain conditions happen (need to work on that section of the entry - Promoting code to branches -).

I will try to get an update by tomorrow.

Cheers,

fm

ghpicard, 2015-05-19:
So... if I were to make a port to Energia on MSP432+RAMPS 1.4 I should take the code from the development branch, right? Or the release branch?

AnHardt, 2015-05-20:
@ghpicard 
Correct. New features - Development.

thinkyhead, 2015-05-21:
? labels on the integration branch, I am not entirely sure. Do you know if they get published in the main repo in the release tab?

Do you mean "tags" here?

fmalpartida, 2015-05-21:
Yes, that was the idea. I don't think anything else than releases should by published in the release tags. O think if we label integ branches too, they will end up there as they are global.

AnHardt, 2015-05-21:
@fmalpartida 
Indeed i tried to talk about labels not tags. The colored markers at the right of the title of an PR what could tell you: "Releasebranch! Be super careful here."

fmalpartida, 2015-05-21:
Excellent! I think we are on the same page with this one.

Wackerbarth, 2015-05-21:
Quoting Paul Harvey, "Page 2".

May I suggest that we offer "github" as vendor resource.

1) Each vendor would be offered a collection of branches that would be archived by github.
2) Each branch must represent a distinct physical printer offering. (only the "latest" revision to a particular model deserves a branch)
3) (after initial setup) Changes to branches will be accepted only if they derive from the most recent release (or a candidate for the next release).
4) Vendors are encouraged to maintain their own repository (and support) which (utilizing automated procedures) will refer their users to their vendor website.

The advantages to us are that it gives vendors an incentive to "stay close" to the master branches.
It also provides us with the access to devices that may, otherwise, become "orphaned".

Since accepting such updates could be automated there is no real burden on the Marlin developers to allow this "service".

thinkyhead, 2015-05-22:
Some questions to think about:
- How often do vendors "hack" Marlin?
- How often do vendors post a ZIP on their website and never update it?
- How can we facilitate the kinds of customizations vendors need so they don't "hack" Marlin?
- How can we encourage vendors to keep their fork, .hex, and source ZIPs in sync with the latest Marlin?

Wackerbarth, 2015-05-22:
From my perspective:
1) Frequently
2) (Too) Often
3 & 4 ) VERY good questions (for discussion)

CONSULitAS, 2015-05-23:
Perhaps we should objectify the discussion about vendors.

I would start with a list, because if we have data, we can decide what to do:

Vendor | Model | distribution | latest Marlin ver | comment
---------- | -------- | --------------| ------------------ | --------------
Vellemann | K8200 | ZIP | 2013-06-05 | popular printer at least in germany

What do you think? I would update this list if you give me data. Thanks in advance.

clefranc, 2015-05-23:
@fmalpartida Here are my suggestions; is it possible to keep a version of README.md for each branches, indicating as intro, clearly which branch we are currently seeing, and a link to switch to the desired branch. You can't know for sure how a first user that wish to update its firmware got the link for Marlin. We all once were newbies, keep this in mind for the README.md. Back when I first uploaded Marlin to my Prusa i3, there was indications on how to flash the firmware; now a wiki is in place, so indicate it clearly with a link.

As I said (and I hear @thinkyhead ), the newbies of today are the developpers of tomorrow, so put informations in each README.md as if you're speaking to total new comers, this will incite them to be part of the Marlin community.

Just my 2¢, cheer!

thinkyhead, 2015-05-24:
> put informations in each README.md as if you're speaking to total new comers

Good idea, will do!

fmalpartida, 2015-05-24:
Can you add me to the development team. I am going to start updating some documentation and getting to do a bit of bug fixing too. Also in the main page to get contacted.

Suggestion, people my be put off, how about if we also put the language they can use next to the developer: My case: [@fmalpartida] English, Spanish

thinkyhead, 2015-10-12:
Now that branches have become "whatever anyone feels like" I will close this issue. If developers feel like discussing branches again, start a fresh topic.

closed 2015-10-12 by thinkyhead