MarlinFirmware/Marlin #2122: Duemilanove w / ATMega328P sketch too big ;(

ksusha-me, 2015-05-19:
Hello. A few days ago held an attempt to create a very low-cost 3D printer. The file configuration.h pins.h and describes the support board 4 = Duemilanove w / ATMega328P pin assignment but this board simply can not work as well, and compatible UNO ATMega328P. The minimum size of the firmware which I managed to get 45900 bytes but these boards Duemilanove 30720 bytes maximum and UNO 32256 bytes as possible which makes it impossible to upload the firmware. In this case, it is my mistake or this board (ATMega328) is no longer supported?
I apologize in advance, I really do not know much English =)

AnHardt, 2015-05-19:
@ksusha-me 
Sorry. Indeed the minimal size for a processor is now 64K of progmem. We are fighting hard to not break this limit. So the chances to get it once small enough to fit again in 32K is close to zero.

closed 2015-05-19 by thinkyhead

ksusha-me, 2015-05-19:
Thanks for the answer. It is a pity of course deystvietelno would collect quite a budget printer. Then I think it would be better to remove the general line with an indication of the cost of the firmware that does not mislead users

PR #2123 (2015-05-19): Remove Duemilanove
No way to fit Marlin into 32K.
Closes #2122