MarlinFirmware/Marlin #2124: Combine Witbox/Hephestos into one board

thinkyhead, 2015-05-19:
Is there a name for these electronics? It's RAMPS with 1 heater and a fan on Pin 9. We should have only one option, perhaps BOARD_RAMPS_BQ, or just add a Configuration option to allow overriding the fan pin on a standard RAMPS and add it to the example configs for these machines.

Before release, also we will be removing most example configs except for examples of each distinct geometry.

ntoff, 2015-05-19:
Why not just a line in pins_ramps_13.h if all other pins are the same?

thinkyhead, 2015-05-19:
The override of a single pin, or in this case fan and heater, isn't facilitated by Marlin, except through pins_MYBOARD.h.

ntoff, 2015-05-19:
isn't that what the #if MB(motherboard_variant) lines do? Why not just call it Ramps_13_BQ and have
#if MB(RAMPS_13_BQ)
#define FAN_PIN            9

or am I missing something?

fmalpartida, 2015-05-19:
I am very much inclined to remove all "configurations" of unique boards. We are making a clear invitations to things that can be maintained elsewhere.
In addition the original idea was to have users customize their own boards for their printers. If they want to help their users with a specific configuration, the can fork off the release branch, synch there and maintain their configuration there.

fmalpartida, 2015-05-19:
See my other thread to this respect.

thinkyhead, 2015-05-19:
@fmalpartida I'm half with you. Thing is, as I mention, there's no facility other than boards.h + pins.h + pins_MYBOARD.h to override a pin, so we would need to make that possible via Configurations. But we can discuss more on the other thread…

fmalpartida, 2015-05-19:
Yes, I think we can touch basis on this topic in the other thread,

In my mind there are several ways to go around it which do allow full control to people of what they want to release to thier customer. I don't want to boil the discussion to these two boards, there are others that need to be addressed too. But we do need to figure out an overall solution and guidance.

Just imagine that someone postedin any forum something like:
"I folks, if you want to have your special 3D printer electronics configuration into the reprap firmware repo".
or
"I have seen that the marlin repo is accepting configurations and examples for customized printer variations".

We will have to give a lot of explenation to questions like: "why is XYZ printer/electronics specific configuration in the repo and not mine?"

I think that it is wise to set the ground rules now before we release to the public.

paclema, 2015-05-19:
Witbox and hephestos use Ramps 1.4, so in my opinion, they might to use pins_RAMPS_13.h config. As they use its own config.h and config_adv on config folder, I think they can create its own pins_RAMPS_13.h if they change any pin,and replace for compiling the original.

Summary:
* pins_"your_electronic".h for new electronics
*new file on your configs folder for configuring the electronic that now exists

Wackerbarth, 2015-05-20:
@fmalpartida - "In addition the original idea was to have users customize their own boards for their printers. If they want to help their users with a specific configuration, the can fork off the release branch, synch there and maintain their configuration there."

IMHO, this is a VERY BAD approach. If you wish to have a growing community or any contributing back to your infrastructure, you need to DISCOURAGE forking.  Every time a project forks, a portion of the audience goes off in a different direction. Pretty soon, your software becomes a smaller and smaller niche in the overall marketplace. Hardware is continually changing (and improving). Marlin needs to embrace those changes to avoid becoming a "has-been".

fmalpartida, 2015-05-20:
Hi folks,

let me bounce these ideas and thoughts over and see how they resonate. 

First up we have to recognise that Marlin is a fully configurable firmware that can be customised and tailored to a good range of electronics, printer geometries and setups. 
Based on this fact, the number of possible configurations can grow exponentially and the growth of examples "vendor specific" configurations too. At some point, the core development Marlin team will start dedicating more time on updating vendor specific files than necessary and will end up giving up on them. Just in one weekend, with a simple well targeted email, we could end up with over a dacade "new custom boards" and example configurations. I can foresee too that any broken vendor specific configuration will get a gun report and have the community fixing it.

There are several ways around this that can help vendors help their customers and it very much depends on how they want to implement it:
1. Fork, merger the latest release branch (or any of the others, depending on how suicidel they feel), push changes, their web page link should always be up to date as soon as they push their changes.
2. Get the release branch once published, merge their changes and publish on their website.
3. Any other method that they see fit to support their products.

Following the current branching scheme that is being rolled out it should be very simple.

I can see a good range of advantages and very little disadvantages. To start with:
1. End-less support/maintenance of a deprecated configuration is not maintained forever. How is the marlin team notified that a particular board is no longer in use?
2. Vendors can decide when to pull in the latest changes and what to publish to their customers. They are in full control of the firmware update to their machines or electronics.
3. The Marlin team is not accountable for vendor's specific support. I do think that vendors are accountable for supporting their own products and what they sell more than relying on the Marlin community to do so.

I can think of a good range of additional reasons that do discourage the current practice, that has been recently adopted, and I see very little benefit for the community other than having a vendor specify printer configurations "showcased" in the Marlin repo. Also think that most "new hardware" that has gone into the repo is just a RAMPs with a similar pin mappings or a remix of a different board.

Most people that buy electronics board over internet will have to do a customisation of the firmware at some point  anyway(calibration could be one example) or they will have their board shipped from their reliable vendor with its base configuration already in place for their setup.

Contributions, well here you will have to fork anyway if you want to contribute to get your PR processed, so...

Perhaps the question that I am trying to answer is: do we want to gradually become the support site of each vendor or do we want to focus on features, refactoring (one in the pipeline is to create a BSP and and HAL) and defect resolution? 
Personally I think that people choose Marlin for their features, the support that the code gets and the current unique boards it supports more than from being able to copy and paste a configuration/pin definition file from the repo. But again, I could be wrong.

Any way, I wanted to share these thoughts and ideas and see how they resonate.

thinkyhead, 2015-05-21:
> Witbox and Hephestos use Ramps 1.4

Indeed. They _look like_ they should just be `RAMPS_13_EFB` (33) but for some reason they're defined as a "fallback RAMPS_13 board" (meaning, they get the last `#else` value from all the conditionals in `pins_RAMPS_13.h`) and then the `FAN_PIN` and `HEATER_1_PIN` are redefined. I will have to see if they are indeed identical to `RAMPS_13_EFB` and make the necessary changes so the configs both use that board.

thinkyhead, 2015-05-21:
#2131

ntoff, 2015-05-21:
>Hardware is continually changing (and improving). Marlin needs to embrace those changes to avoid becoming a "has-been".

It's not like Marlin is shunning newer CPU's and boards with onboard RAM and flash storage. It's still running on the atmega hardware.

>Perhaps the question that I am trying to answer is: do we want to gradually become the support site of each vendor or do we want to focus on features, refactoring (one in the pipeline is to create a BSP and and HAL) and defect resolution? 
Personally I think that people choose Marlin for their features, the support that the code gets and the current unique boards it supports more than from being able to copy and paste a configuration/pin definition file from the repo. But again, I could be wrong.

I look at it like a computer's motherboard. They all run AMIBIOS or Award BIOS but each PC has different features based on what the vendor wants to integrate. Marlin should provide the interface between a host / sd card and the hardware and it should be up to vendors to customize the rest if they want certain features to use different pinouts.

IMHO each board should come from the seller with a pinout or a schematic that you use when configuring the firmware. Since that doesn't happen it seems to have fallen to the devs here to constantly try and keep up with all the different boards.

fmalpartida, 2015-05-21:
@ntoff I fully agree, we cannot expect that the community fills the gaps.

When I talk about features, they obviously include new boards as opposed to configuration.

fmalpartida, 2015-05-21:
Parallel thread discussion going on #2102, which is a bit broader in scope.

Wackerbarth, 2015-05-21:
I guess that I look at it from a non-vendor perspective.  As one who "mixes and matches" hardware components and develops algorithmic improvements, not only for myself, but to share with others,
I find it desirable to have as many users as possible running on, or at least near, the current mainline of development. This is so that (1) it is easier to integrate enhancements and (2) avoid having users encounter situations that have already been addressed.

There is also the very real situation of orphaned hardware. Far too many designs are no longer supported by their original vendor (either because the vendor has no incentive to have "upgradable" hardware, or because the vendor is out of business or otherwise no longer providing support)

Now, I understand that there are some vendors who wish to take the best of the community work and split off into their own fiefdom and "lock in" their customer base  (Microsoft certainly operated in that manner, and we still suffer from it), but I have no interest in supporting those vendors in any manner.

On the other hand, if we wish to have users remain close to the mother project, we must make it easy for all vendors to provide their users with the necessary branch of a common repository so that those who wish to proceed at their own pace, rather than having the pace dictated by the vendor, can do so.
In addition, we need to encouraging sharing. It is not good if each user, individually, must discover and implement the same enhancements.

fmalpartida, 2015-05-21:
I think so too but not at the expense of the community. 

What you are describing is option 1 self-paced upgrades through git. But vendors can choose to move to option 2 using a simple zip file,or why not both.

Wackerbarth, 2015-05-21:
Perhaps I am misconstruing your intention WRT "option 2".  Since the GPL requires certain source distribution and some vendors may not wish to maintain a PERMANENT repository, providing an archive file (.zip, etc.) is a reasonable approach to meeting their obligation.

However, I feel that it is important to attempt to have the files available in a git repository that is derived from the master Marlin repository. To the extent that we can make it easy for the vendors to do so, I feel that everyone will profit. If you will look at the "Automatic Versioning" work that I am doing, you will see that it is an effort to automate the production of firmware .hex files and source archives based on a git repository. Hopefully, this will encourage vendors to "stay close" to the main development and allow their users an easy path to experimenting with our main development branch.

What I really want to avoid, whenever possible, is having vendor development being performed outside of the git repository.

When I received by Kossel Pro, I had immediate firmware issues. Unfortunately, I had to expend significant effort to figure out how the vendor had modified some version of the Marlin code so that I could make corresponding changes to a much-more-recent version and have some that I can reasonably share with others. I certainly want my efforts to be "preserved" so that someone else doesn't have to repeat the portion of the work that I have successfully completed.

closed 2015-05-25 by thinkyhead