MarlinFirmware/Marlin #2125: Synchronized vs unsynchronized commands

alexrj, 2015-05-19:
Hello!
Is there any documentation about what commands are synchronized/buffered and which ones are executed immediately?

I see a lot of confusion about that point. The [documentation for M400](http://www.marlinfirmware.org/index.php/M400) states:

    This command should rarely be needed since non-movement commands should already wait, 
    but M400 can be useful as a workaround for badly-behaved commands.

But looking at the code most of the non-movement commands such as `M104`, `M106`, `M42`, `M280` don't call `st_synchronize()` thus appear to be executed immediately (well, as soon as the command buffer is processed but not waiting the motion queue to be finished).

In this GitHub issue tracker I found several conflicting statements about `M106` being synchronized and not.

What's the situation? Can this be documented clearly? Thank you! :)

daid, 2015-05-19:
All commands are executed "synchronized". You get the OK back when the command is done.

Then you just need to think G0/G1 as "plan move" instead of "do move", and you are pretty much golden. (And M400 is just "finish planned moves")

M106 put in the planner in Marlin as well, so it's also queued up and executed in order with G0/G1 moves. But really old versions did not do this, and other firmwares might do it differently.

Those are about the only exceptions.

(st_sync is generally bad, and should be avoided)

alexrj, 2015-05-19:
Okay, then I think the [documentation page for M400](http://www.marlinfirmware.org/index.php/M400) should be updated since it's not true. It should be rephrased something like this:

   All non-movement commands except `M106` do *not* wait for planned moves to finish
   and M400 is needed in all cases when user wants the action to be synchronized with 
   moves.

The current documentation leads user to think that commands like `M104`, `M42`, `M280` *are* synchronized with the execution of planned moves, and that `M400` is not needed for that.

nophead, 2015-05-19:
Or Marlin should be fixed so all non movement codes wait for the movement to finish to avoid having to put M400 before every code that follows a movement code.

daid, 2015-05-20:
... Wouldn't that negatively effect print quality? Why would LED commands for example wait for the planner to be empty? The whole idea of the planner is to always have the head moving while you are doing other things.

bkubicek, 2015-05-20:
yes, that  would have negative influence on the print quality. Any Buffer
emptying causes a wart on many printers.
E.g. calling a m115 should in no way cause a wart. Neither should changing
the flow, or the feedraate, or asking for temperatures ...
I think the compromise is currently alright. If you need to do something
_really_ in sync with movement, use m400. For many commands, where this
makes sense always, its already implemented in sourcecode, e.g. suicide or
turn of power supply.
I even preferred the old status quo where the fan commands were not in sync
with the movement, because it takes a time to spin the fan up, so it makes
sense to trigger fans earlier than the actual fan-requiring moves.

Bernhard

bkubicek, 2015-05-20:
he primary goal of a 3d printer firmware is to create the best possible
prints. Anything that decreases print quality is hence bad and needs a
sufficient tradeoff of to be in any default config.

Bernhard

nophead, 2015-05-20:
Why would a slicer send none movement commands while it was extruding
unless it wanted them to be executed then? It would put them between
filament runs where a milliseconds pause doesn't matter. It wouldn't expect
them to be executed earlier while it was still moving, which is what
happens now.

daid, 2015-05-20:
Because it want's them to be executed "around that time". For example, M42 to turn LEDs on near the end of a print to signal the user?
Photo camera trigger to take a snapshot when a layer is done? This command also does not take a millisecond, it has multiple delays in it. Pretty sure you would effect the print quality if you sync this one.

Don't make assumptions about what slicers will generate. You're just making my life, and the life of @foosel harder.

@alexrj if you need some more info about what is synced and not, let me know. Most things are not synced to the movement of the printerhead. Except for the fan commands and movement commands.

Wurstnase, 2015-05-20:
As far as I understand this part of the code.

We have two queues. The command-queue and the planner.
The command comes from the host and is put in the command-queue. Every command will be execute in the right order.

G1 X10 Y10 -> command-queue -> planner
M104 S200 -> command-queue
G1 X20 Y30 -> command-queue -> planner

this will be executed from G1 -> M104 -> G1. The move in the planner needs an extra calculation (jerk etc.). So everything is in time.

To solve this, we could have an extra flag when a move was execute and only after that the non-movement-commands will be executed. Just a naive solution.

alexrj, 2015-05-20:
Waiting for the planner to be empty does affect print quality, but that's just a suboptimal implementation...
Synchronizing all actions doesn't imply you need to empty the planner.
If you put everything in the planner queue you won't break look-ahead and won't need to wait for the planner to be empty. That's the ideal solution IMHO.

In the mean time I really think the current behavior should be documented in order to prevent users from assuming wrong things. For example, since `G4` calls `st_synchronize();`, a sequence of `M42` and `G4` commands (for blinking a LED or triggering a buzzer) would produce unexpected results.

nophead, 2015-05-20:
>this will be executed from G1 -> M104 -> G1.
No it will be executed M104 G1 G1 because moves are simply transferred to the planner queue, not executed.

Wurstnase, 2015-05-20:
> No it will be executed M104 G1 G1 because moves are simply transferred to the planner queue, not executed.

right,
the command-queue will execute G1 M104 G1? the planner has his own queue?

nophead, 2015-05-20:
Yes they get pulled from the command queue G1 M104 G1 but the G1s are transferred to the planner queue whereas the M104 is executed immediately instead of waiting for the first G1 to finish (or even start if the planner wasn't empty). If the slicer wanted that to happen it would have sent M104 G1 G1.

AnHardt, 2015-05-20:
Maybe M104 is not the perfect example. Until the temperature will begin to rise an average G1 is history anyway.
More interesting, to me, is the question how a change affecting movements is handled. Changing the acceleration for example. Will this alter the behavior of already planed moves?

alexrj, 2015-05-20:
By the way, often there's a (small) delay before `M104` is executed as well; I guess that's because `manage_heater()` is not called immediately and it's postponed by blocking operations if any (motion? reading from SD?). Not a huge issue, but might be worth documenting since some people try to use the heaters for controlling lasers and delays matter.

daid, 2015-05-20:
Acceleration changes are changes to the settings. These settings are used in the planner. So as soon as you do an acceleration value change, you could be effecting some of the previous planned moves. But it does not have to be (the planner is a complex piece of code). It will however, for sure, be applied on all the G1 after the acceleration setting change.

Wurstnase, 2015-05-20:
@alexrj ```manage_heater()``` and other periodical parts need a rework with a ticktimer or something like that. Just read an issues in the reprap-forum for long moves on deltas.

M104 is a really good example for that. If I print my first layer, my temp is at maybe 250°. If I have a long last move and the temp will rise or drop to early or i shut of the temp after my print i will run into trouble.

AFAIK, already planned moves shouldn't use new acceleration setting.

nophead, 2015-05-20:
The planner only uses the acceleration and jerk values from the settings when it puts the trapezoid into the queue and stores the block acceleration in the queue. When new moves are added it r-evaluates all the trapezoids in the queue potentially adjusting their speeds, but it only uses the stored acceleration value, not the one from the settings.

foosel, 2015-05-20:
Is it only me or does anyone else have a strong dejavu regarding that discussion?

Just for the record (sorry if that was already clear to everyone, just wanted to make sure we are all on the same page), lifecycle of a planned move command: rx buffer, command buffer, "ok", planner queue, executed vs lifecycle of an unplanned non-move-command: rx buffer, command buffer, executed, "ok". Execution of non planned commands takes place within the command queue whereas execution of planned moved happens outside of that (in the planner? Or separate from that too?). At least that was the flow inside marlin the last time I checked (around new years). Note that the lifecycle for planned commands used to be rx buffer, "ok", command buffer, planner queue, executed, but that caused commands to not be acknowledged in the same order as they were received (since they were ack'ed basically in the rx buffer handling, with command buffer handling taking place after that, so a G0 could "overtake" an M114 in acknowledgement order, causing problems for command lifecycle tracking by the host).

Afaik the only planned commands are G0 through G3, G28 blocks iirc but is synchronized with the planner queue. Am I right in assuming that it would make sense to document the expected blocking/non-blocking behaviour in the standardization attempt?

nophead, 2015-05-20:
Yes that is my understanding.

>in the planner? Or separate from that too?

The current trapezoid taken from the head of the planner queue is executed the interrupt routine in stepper.cpp. While the trapezoids are waiting in the planner queue they get adjusted when a new one arrives to take into account they might no longer need to stop so they can potentially go faster.

nophead, 2015-05-20:
As for documenting a broken scheme and making it a standard, I think not. Other firmwares should prefer to execute commands in the correct order.

daid, 2015-05-20:
@foosel G4 is also synced.

alexrj, 2015-05-20:
I'm totally with @nophead about how commands should be executed. But I think that documentation should always reflect the current software behavior, which is not the case now. If you mark something as "known issue" or "experimental" or "may change in the future" or "may differ from other implementations" you're not making it a standard.

thinkyhead, 2015-05-21:
> M400 current software behavior

yeah, sorry about that. This is the early revision, and I need to rephrase: Commands that *should* wait do wait.

thinkyhead, 2015-05-21:
> If you mark something as "known issue" or "experimental" or "may change in the future" or "may differ from other implementations" you're not making it a standard.

True. The "standard" is one thing. The specific implementation and its particulars is all we're after here. I would hate for any weird behavior of Marlin get pulled into a "standard" just because it becomes "expeted behavior." Marlin should strive to be as adherent as it can, and to label any disparities as quirks of the current version – some of which may be hard to fix without changes to internal methodologies.

thinkyhead, 2015-05-21:
> Commands that should wait, do wait.

Let me catch myself here. Yes, ideally, commands like `M106` should sync up with the move that it has in sequence. There was a very brief discussion of extending the "planner queue" to be able to call back to things like this that would be executed immediately, but that's a complicated solution.

daid, 2015-05-21:
M106 does sync up. (It's a bit tricky to follow, but it happens here: https://github.com/MarlinFirmware/Marlin/blob/Release/Marlin/planner.cpp#L456 )

The acceleration change also syncs up (as someone pointed out, and what I remember from the code, that's true, as long as you just change the general acceleration value, messing with the max acceleration values might effect previous moves)

thinkyhead, 2015-05-21:
@daid Aha, I should have noticed that, as I was just looking at the `tail_fan_speed` code the other day. So of course, we don't care if (or really want) temperature changes to sync up the same way (if they for some reason are in the G-Code). As part of the docs, I'll try to make a definitive list of which codes "matter" in terms of sync, and which codes don't. If there are some that _should wait_ but don't, we'll just have to patch that.

nophead, 2015-05-21:
Temperature codes should be synced up. It makes no sense they get executed
early. For example the hot end can get turned off at the end before it has
finished extruding. And second layer temperature changes happen early.
Basically people and slicers put the code where they want it to change and
that is not generally while extruding, so a sync would have no effect on
quality.

If a slicer did put a code in the middle of extruding then it would be for
a specific reason, like a bridge. I doubt it would react fast enough but it
should still be done at the correct time, even if that does cause a brief
stop. It is only as bad as turning a sharp corner during an outline or
infill, so if your machine makes a blob then it can't print decent
rectangles anyway.

daid, 2015-05-21:
@nophead on ARM, yes. On AVR, we need to think about what makes sense and what is effective in CPU cycles.

nophead, 2015-05-21:
I don't see how making most of the M codes wait for the planner queue to be empty is a CPU cycle issue.

alexrj, 2015-05-21:
Maybe it's more a memory problem (larger block_t objects in the planner?)...

nophead, 2015-05-21:
No, just some more calls to st_synchronise().

alexrj, 2015-05-21:
Doesn't that break look-ahead? Shouldn't the proper implementation work as `M106` does, putting the action inside a block in the planner?

daid, 2015-05-21:
It isn't, unless you do it properly and not call st_sync, as st_sync is evil. Making M104 synced will cause quality loss.

nophead, 2015-05-21:
If you don't use M104 while it is extruding how does it affect quality? It would generally be at a layer boundary. The lack of a look-ahead simply means it will make a full stop there, but it does anyway.

daid, 2015-05-21:
But I am using M104 while extruding.

nophead, 2015-05-21:
Well if you are then don't you want it to occur at the right time instead of potentially much too early if the planner queue has some long moves in it?

daid, 2015-05-21:
I want it to occur at about that time, not exactly. I know the head takes time to warm up, I have an estimate about how long that takes, so I put the M104 in the spot where the other hotend has enough time to heat up so it is ready to be used when it needs to be used.
(Dual extrusion, heatup nozzle before it's needed)

thinkyhead, 2015-05-22:
Ultimately it would be best to be able to say: be at this temperature at this time. But that would require pre-reading the whole file and doing a heatup far enough in advance of the actual g-code line that it will be settled at temperature by that point.

thinkyhead, 2015-05-25:
> so I put the M104 in the spot where the other hotend has enough time to heat up

In the Hangout discussion yesterday we talked about how the "smart thinking" is better left to slicers. When generating G-code for a machine, software has to be cognizant that the machine has physical limitations and that moves will always lag behind command processing. Taking a "purist" view of the firmware, it should be a perfect slave and it shouldn't try to infer too much. The firmware should faithfully carry out its orders in a predictable fashion and as smoothly as it can. We know that `M104` will most likely start before the G1 that preceded it even does, but that is standard and expected behavior, and G-code generators can anticipate it.

nophead, 2015-05-25:
No it is a bug / memory limitation of Marlin and should not be made standard. There is no logical reason it can't put line numbers in the planner queue and then hold other commands in a queue with their line numbers and only execute them when all the preceding moves have been executed. LinuxCNC executes Gcode in the correct order. I don't know what Smoothie does but it should have enough memory to do things properly.

daid, 2015-05-25:
I don't mind M104 being executed "in order", but adding a st_sync isn't the
proper way to achieve that.

nophead, 2015-05-25:
Yes it would be fine with a single extruder but as you pointed out it could cause a problem with dual extruder pre-warmup. However, don't you get a problem when you set the active extruder to a lower temperature when it is switching to the new one? That could happen early while it was still extruding.

Also when you do st_sync it will simple drop to jerk speed rather than carrying higher speed through the next junction. How is that worse than turning a sharp corner? Some of my old machines that run my own firmware always do full stop and have no command buffer, let alone lookahead and they work fine.

daid, 2015-05-25:
st_sync will drop the speed to 0, then delay till the next command is received. Talked with ErikVdZalm about this on friday, and he agreed that there should not be a sync in the M104, as it will negatively effect quality.

nophead, 2015-05-25:
My understanding is the speed drops to jerk / 2 and then to zero but when the next G1 is seen it will start again at jerk / 2. So if there are commands in the command queue it will be at zero for negligible time, it simply decelerates to jerk / 2 and then accelerates from there. I.e. a few milliseconds at zero is simply of no consequence. The reason it uses jerk / 2 is that it could potentially change direction as it doesn't know the next move. That would give a total change of speed equal to jerk, worst case. This means the description of Marlin jerk you find over the web as the starting speed is wrong, it is halved.

Without the flush it may decelerate less and carry a higher speed through the junction if the angle is shallow, but if it is an acute angle (where an axis needs to reverse) it will decelerate to something approaching jerk / 2 anyway. 

In my own firmware I have no jerk setting and always stop between segments but I don't stop the extruder, it runs at constant speed. The results I get are not much different. This is why I don't understand how it has such a negative effect on quality unless you also get poor quality at sharp corners anyway.

Regardless, the correct solution is to queue the non-moves that change machine state and execute them when all the moves that came before have left the planner and stepper. It does need a bit more memory but won't detract from quality. Then temperatures will change when you expect them to and things like triggering cameras and lasers, controlling fans that are using M42 and showing messages will actually happen at the correct instant. Currently they can be minutes early if you are infilling a large base layer slowly unless you use M400.

thinkyhead, 2015-05-26:
Currently the fan speed is changed as part of the planner queue. Perhaps this could be generalized to change other pin states too. Then any commands that only trigger a camera or laser can use the same methods we use for changing the fan speed.

As for doing lookahead on whole print jobs and pre-planning temperature changes, it's an interesting idea. But it opens up some questions. Would hosts then be expected to send the entire G-Code file once to the firmware at the start for pre-scan, and then again for the actual print? The host could pre-process the code and simply move temperature commands to earlier lines a lot faster than handing it off to the firmware. And, the firmware doesn't necessarily know during pre-scan if it will be running at full speed, slower, or faster, so… when hosts are controlling the printing, they have more leeway about where to move those temperature commands.

In the meantime, I suppose we could look at adding to the planner queue any commands that trigger pins or which can be implemented with simple callback pointers.

jbrazio, 2016-04-23:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

closed 2016-04-23 by jbrazio

amrithmmh, 2019-02-15:
If i give a G0 X100 i do not get any feedback once completed
also if i send servo M280 Px Sx then it executr\es it in between movement anyway to know if movement is completed or not?

nophead, 2019-02-15:
Use M400 before the M280.

amrithmmh, 2019-02-15:
Thank you @nophead

amrithmmh, 2019-02-20:
I am sorry actually when i tried directly in arduino serial monitor it shows "ok" after command completion but i was using only pronterface to test earlier