MarlinFirmware/Marlin #2140: Merge #2113 breaks command parsing

Wackerbarth, 2015-05-22:
Scott,
Thanks for merging my Kossel Pro infrastructure.

Unfortunately, when I tried to test it this morning, the printer command parser is broken.
It will not recognize commands from Repeater-Host Mac.
I have isolated it to the problem to one of the merges that you did last night.

If I start from 71ff9ad, everything is fine. But based on 4d11b29 things fail.

You can find my build at https://github.com/Wackerbarth/Marlin/tree/BrokenCommandParser

thinkyhead, 2015-05-22:
#2141 should do the trick, don't know how I forgot that line…

Wackerbarth, 2015-05-22:
It appears to be fixed at #2142

thinkyhead, 2015-05-23:
@Wackerbarth Excellent. This should show some performance benefit, save energy, and cool the electronics by an amount that no one will ever notice!

closed 2015-05-23 by thinkyhead