MarlinFirmware/Marlin #2150: Filament sensor not work rambo

galeots, 2015-05-23:
filament sensor does not work in my rambo.
It always remains fixed at 5 in the display, perhaps the pin is wrong to put it.
I have a geeetech matherbord.

clefranc, 2015-05-24:
@galeots Which pin did you use in which pins_XXXX.h?

thinkyhead, 2015-05-24:
Probably in `pins_RAMBO.h`…
```
#ifdef FILAMENT_SENSOR
  //Filip added pin for Filament sensor analog input 
  #define FILWIDTH_PIN        3
#endif
```
Are you using a pin other than 3? If so, try changing it here.

galeots, 2015-05-24:
pin 3 does not work does not read any real live always reads 5 volts without anything.
the pin used to think it is wrong pinout.

clefranc, 2015-05-24:
@galeots There is many flavour of the RAMBO board:

1. Which version of the RAMBO board are you using?
2. What kind of filament sensor are you using?
3. On which connector and pins the filament sensor is connected (photo)?
4. Did you test the filament sensor output with a voltmeter?

galeots, 2015-05-25:
my version is 1.2a.
the card sees me without anything on display from 5 volts attacked and if I put the sensor does not detect it.
the sensor is this :http://www.thingiverse.com/thing:454584
The sensor works by its pinout

clefranc, 2015-05-25:
@galeots It's connected there:

[image]

thinkyhead, 2015-05-25:
That looks like a 1.3L board. The 1.2a board looks like this:
[image]

galeots, 2015-05-25:
@thinkyhead @clefranc The card looks like thinkyhead says, the sensor as antrebbe connected!?
but those are not the doors mx !?

clefranc, 2015-05-25:
@galeots Yes, @thinkyhead shows the right board. Here is another pic from the RAMBO 1.2 wiki showing where is the ADC3 pin on the X40 connector.

http://reprap.org/wiki/Rambo_v1.2

[image]

@galeots Can you confirm?

galeots, 2015-05-25:
@thinkyhead @clefranc 
It does not work in the pin 3 to pin 8 as from image
[image]

clefranc, 2015-05-25:
@galeots I see a little confusion in your pictures. The top image shows you're connected to analog 8, but the lower one shows analog 5. I also see that the table from the wiki is wrong, some pins don't match the schematic. *So ignore the table and concentrate on the schematic.*

**Can you tell us exactly where the filament sensor pins are connected on the X40 connector.**

Filament Sensor      RAMBO x40
**GND**                             **?**
**OUT**                             **?**
**+5V**                              **?**

galeots, 2015-05-25:
@thinkyhead @clefranc  pin assigned by default in rambo.h
GND 2
OUT 8
+5    1

clefranc, 2015-05-25:
@galeots Then you must use analog 5 in pins ```RAMBO.h``` :

```
#ifdef FILAMENT_SENSOR
  //Filip added pin for Filament sensor analog input 
  #define FILWIDTH_PIN        5
#endif
```

galeots, 2015-05-25:
@galeots Yes, @thinkyhead not!  #define FILWIDTH_PIN  3 pinout pcb  8

clefranc, 2015-05-25:
@galeots Is there a reason you didn't follow the installation instructions given in Thingiverse: https://thingiverse-production.s3.amazonaws.com/assets/99/86/c1/1b/8e/3D_printer_hookup.pdf

galeots, 2015-05-25:
@galeots Yes, @thinkyhead not! correct is
[image]

thinkyhead, 2015-05-27:
@clefranc Am I reading it right, that the PDF wants Analog Pin 4 for the sensor input? So, what do you think should be the default setting in `pins_RAMBO.h`? I'm not sure where the current value of 3 comes from (except of course, from "Filip"). But apparently @galeots has found Pin 5 to be more suitable for the RAMBo 1.2a. If the pins differ for different versions of RAMBo, maybe we should label the pin with the best numbers for each version.

galeots, 2015-05-27:
I tried all the pins of the card, the allocation is correct this card geetech
[image]

clefranc, 2015-05-27:
@thinkyhead The pin for RAMBO is correct, the instructions from Filip are correct. I've a hard time to understand what @galeots is trying to explain.

@galeots Is this issue resolve? Please answer with YES or NO. Thanks!

galeots, 2015-05-27:
@thinkyhead @cajun-rat It is correct my photo!
https://cloud.githubusercontent.com/assets/6709995/7835339/6e3ced0e-0478-11e5-8d22-cf34ecc95a96.jpg

galeots, 2015-05-27:
@thinkyhead  @clefranc  It is correct my photo!
https://cloud.githubusercontent.com/assets/6709995/7835339/6e3ced0e-0478-11e5-8d22-cf34ecc95a96.jpg

clefranc, 2015-05-27:
@galeots No, your photo is not correct!

galeots, 2015-05-27:
@clefranc yes! I tried pin for pin

thinkyhead, 2015-05-28:
@galeots posts a photo showing the pin order as G 5 4 3 | V 6 7 8 and he confirms that probing the pins proves this order is correct on his RAMBo. @clefranc is that not the usual pin ordering on a RAMBo 1.2a? Pin 5 is next to GND which seems correct from other pix I have seen.

clefranc, 2015-05-28:
@thinkyhead I'm looking at the [reference schematic](http://reprap.org/mediawiki/images/9/91/Rambo1-2-schematic.png) for the 1.2 version, what I see for X40 is this:

**PIN#2 - GND** | **PIN#4 - A3** | **PIN#6 - A4** | **PIN#8 - A5**
--- | --- |--- | ---
**PIN#1 - 5V** | **PIN#3 - A6** | **PIN#5 - A7** | **PIN#7 - A8**

thinkyhead, 2015-05-28:
> Did you find other schemas?

@clefranc I thought I had, but now of course I can't find it. Maybe just a false memory.

Hmm, so anyway, it's an interesting discrepancy.

@galeots Maybe you can find a demo sketch —or just write one— to test the analog pins on your board, just to double-confirm that the order is different from the table @clefranc just posted, which seems to match up with the PDF linked above as well. Maybe you just have a weirdly-wired RAMBo.

galeots, 2015-05-28:
@clefranc @tgiphil i have tested pin to pin, work pin as photo

thinkyhead, 2015-05-28:
Interesting that the "exit" and "PIN#" numbers in the two images match up.

Info from Geeetech and testing by @galeots confirms this order for his RAMBo 1.2a:
**exit 2 - GND** | **exit 4 A5** | **exit 6 - A4** | **exit 8 - A3**
--- | --- |--- | ---
**exit 1 - 5V** | **exit 3 - A6** | **exit 5 - A7** | **exit 7 - A8**

While the schematic for RAMBo 1.2g shows this order:
**PIN#2 - GND** | **PIN#4 - A3** | **PIN#6 - A4** | **PIN#8 - A5**
--- | --- |--- | ---
**PIN#1 - 5V** | **PIN#3 - A6** | **PIN#5 - A7** | **PIN#7 - A8**

Has this changed between 1.2a and 1.2g ?

boelle, 2015-06-01:
i will close this one.... there is so much confusion that it can just go on forever

@galeots take a picture yourself and let us see how YOU have connected it. Post your config to a site like pastebin.com so we can see your config... 

Then we will tell you if you have connected it right

closed 2015-06-01 by boelle