MarlinFirmware/Marlin #2151: Arduino 1.6.4 does not compile

fizcris, 2015-05-24:
I am facing the next error:

Cannot run program "C:\Program Files (x86)\Arduino\{build.toolchain}{build.command.g++}" (in directory "."): CreateProcess error=2, The system cannot find the file specified

How can I fix it?

fizcris, 2015-05-24:
This is the same error with verbose ON:

C:\Program Files (x86)\Arduino/{build.toolchain}{build.command.g++} -c {build.flags.optimize} {build.flags.common} {build.flags.cpp} {build.flags.cpu} {build.flags.defs} -DARDUINO=10604 -DF_CPU={build.fcpu} -DARDUINO_ARCH_AVR -D{build.usbtype} -DLAYOUT_{build.keylayout} -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy C:\Users\Cris\AppData\Local\Temp\build518418193100349581.tmp\sketch_may24a.cpp -o C:\Users\Cris\AppData\Local\Temp\build518418193100349581.tmp\sketch_may24a.cpp.o 

Cannot run program "C:\Program Files (x86)\Arduino\{build.toolchain}{build.command.g++}" (in directory "."): CreateProcess error=2, The system cannot find the file specified

eboston, 2015-05-24:
I have seen thia problem and 1.6.4 doesn't always show a compiler error and appears to just hang. I have gone back to an earlier release and all is well.

fizcris, 2015-05-24:
Could you say which release of Teensyduino and Arduino IDLE worked for you?

fizcris, 2015-05-24:
Well... here my advances...:
*This version of Marlin does not compile for me in any 1.6.x version in Arduino IDLE , it does in 1.0.6.
*With Arduino IDLE 1.0.6 there are problems with the libraries, "Arduino.h" and "Wire.h" 
*Teensyduino does not recognize Arduino IDLE 1.0.5

How can I proceed?

fmalpartida, 2015-05-24:
What board are you compiling for?

Wackerbarth, 2015-05-24:
What mpu do you have? And what entry are you selecting?

I have builds working based on my 1286 board, a fairly recent Marlin source, using the very latest Arduino IDE, but with the core from 1.0.5), I think.

eboston, 2015-05-24:
I have done completely clean installs of 1.6.4 with latest versions of teensy installed over it and have had it failed when compiling for a Printrboard.  I have also done a clean install without teensy and had it failed when compiling for a RAMPS board (Mega).

fizcris, 2015-05-24:
I am compiling for SAV MKI with an AT90USB1286.

I was able to do a success compilation with this 1.0.5 Arduino IDLE https://github.com/fizcris/arduino-1.0.5-r2-at90usb1286

But now I am facing problems with the fmalpartida branch of the code for this board.

Wackerbarth, 2015-05-24:
I'm fairly certain that the issue with the Mega has to do with the hack that Marlin does to the serial I/O driver. Those routines have changed in the Arduino code base between 1.0.5 and the present.

fizcris, 2015-05-24:
With fmalpartida branch I am facing this error:

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
Arduino: 1.0.5-r2 (Windows NT (unknown)), Board: "[BootloaderCDC]AT90USB1286"
Marlin_main.cpp: In function 'void gcode_M81()':
Marlin_main.cpp:3513: error: 'WRITE' was not declared in this scope
Marlin_main.cpp: In function 'void gcode_M18_M84()':
Marlin_main.cpp:3557: error: 'WRITE' was not declared in this scope
Marlin_main.cpp:3565: error: 'WRITE' was not declared in this scope
Marlin_main.cpp:3566: error: 'WRITE' was not declared in this scope
Marlin_main.cpp:3567: error: 'WRITE' was not declared in this scope
Marlin_main.cpp:3570: error: 'WRITE' was not declared in this scope
Marlin_main.cpp: In function 'void gcode_M119()':
Marlin_main.cpp:3678: error: 'READ' was not declared in this scope
Marlin_main.cpp:3686: error: 'READ' was not declared in this scope
Marlin_main.cpp:3694: error: 'READ' was not declared in this scope
Marlin_main.cpp: In function 'void enable_all_steppers()':
Marlin_main.cpp:6017: error: 'WRITE' was not declared in this scope
Marlin_main.cpp: In function 'void disable_all_steppers()':
Marlin_main.cpp:6027: error: 'WRITE' was not declared in this scope
Marlin_main.cpp:6028: error: 'WRITE' was not declared in this scope
Marlin_main.cpp:6029: error: 'WRITE' was not declared in this scope
Marlin_main.cpp:6030: error: 'WRITE' was not declared in this scope

Wackerbarth, 2015-05-24:
Can you join me at the hangout that we had going yesterday?

fizcris, 2015-05-24:
Wackerbarth I succed using that modified Arduino IDLE, I could upload software to the board, but I can't find a functional Marlin version...

fmalpartida, 2015-05-24:
It is a bit odd, I am using Arduino 1.0.5 which is working fine on my PC. 
I do need to find some time to migrate to the new 1.6 IDE to see how it fairs.
For the SAV, you should use the latest build that is available on it wiki. I also have a 1.0.2 in my release branch for the board: 
https://github.com/fmalpartida/Marlin/tree/SAV-MkI-config
or
https://github.com/fmalpartida/Marlin/releases
1.0.2.3 is fully operational.

Several questions:
- Do you have the LCD activated? Did you download and install the New LiquidCrystal library?
READ/WRITE are macros declared in fastio.h, so its a bit strange that you are not compiling.

Did I read that you got it going on the latests Arduino IDE too?

Wackerbarth, 2015-05-24:
I just pushed a build for the SAV Mk-I to https://github.com/Wackerbarth/Marlin/tree/Trial_for_Cristian

Please try the following:

Download and install a very recent copy of the Arduino IDE (preferably 1.6.4 or later)
Clone my repository (you can do so as an additional remote)
Check out "Trial_for_Cristian"
Install the Arduino 1.5 hardware support folder (I do it with a symbolic link)
Select the [BootloaderCDC]Printerboard and try to build.
If that succeeds, try testing it.

fizcris, 2015-05-24:
Thank you very much Wacerbarth!!

I did exactly that you told me but, I couldn't get the  [BootloaderCDC]Printerboard, I think it is not in the files, I found others like Gen7,RAMBo... but the compiler still ask for Teensy2++

Wackerbarth, 2015-05-24:
Did you install the hardware support files from ArduinoAddons/Arduino_1.5.x/hardware/at90usb1286/ ?
On my system that folder is located at ~/Documents/Arduino/hardware/at90usb1286/

And any time you change the support files, you will need to restart the IDE in order to get it to recognize the changes.

fizcris, 2015-05-24:
Wackerbarth you can check that the folder at90usb1286 does not exist in the repo you passed me...

Wackerbarth, 2015-05-24:
Oops! Try again.

fizcris, 2015-05-24:
Thank you again!

It woprked! I can find the board but the compilation problem still continues...

Arduino: 1.6.4 (Windows 8.1), Board: "[BootloaderCDC]Printrboard"

/usr/local/bin/generate_version_header_for_marlin C:\Users\Cris\Desktop\3d_printer\Marlin-Trial_for_Cristian\Marlin-Trial_for_Cristian\Marlin C:\Users\Cris\AppData\Local\Temp\build5305325282213429987.tmp/_Version.h 

Cannot run program "\usr\local\bin\generate_version_header_for_marlin" (in directory "."): CreateProcess error=2, The system cannot find the file specified

Arduino Version: 1.6.4

"language_kana.h" contains unrecognised characters.If this code was created with an older version of Arduino,you may need to use Tools -> Fix Encoding & Reload to updatethe sketch to use UTF-8 encoding. If not, you may need todelete the bad characters to get rid of this warning.

"language_kana_utf8.h" contains unrecognised characters.If this code was created with an older version of Arduino,you may need to use Tools -> Fix Encoding & Reload to updatethe sketch to use UTF-8 encoding. If not, you may need todelete the bad characters to get rid of this warning.

"language_ru.h" contains unrecognised characters.If this code was created with an older version of Arduino,you may need to use Tools -> Fix Encoding & Reload to updatethe sketch to use UTF-8 encoding. If not, you may need todelete the bad characters to get rid of this warning.

"language_test.h" contains unrecognised characters.If this code was created with an older version of Arduino,you may need to use Tools -> Fix Encoding & Reload to updatethe sketch to use UTF-8 encoding. If not, you may need todelete the bad characters to get rid of this warning.

"utf_mapper.h" contains unrecognised characters.If this code was created with an older version of Arduino,you may need to use Tools -> Fix Encoding & Reload to updatethe sketch to use UTF-8 encoding. If not, you may need todelete the bad characters to get rid of this warning.

fizcris, 2015-05-24:
Arduino: 1.6.4 (Windows 8.1), Board: "[BootloaderCDC]Printrboard"

Build options changed, rebuilding all

/usr/local/bin/generate_version_header_for_marlin C:\Users\Cris\Desktop\3d_printer\Marlin-Trial_for_Cristian\Marlin-Trial_for_Cristian\Marlin C:\Users\Cris\AppData\Local\Temp\build5305325282213429987.tmp/_Version.h 

Cannot run program "\usr\local\bin\generate_version_header_for_marlin" (in directory "."): CreateProcess error=2, The system cannot find the file specified

Wackerbarth, 2015-05-24:
Remove the platform.local.txt to turn off automatic Version header generation (You may also then need to tweak the Default_Version.h file.

We will eventually replace that bash script with something that works on Windows.

Wackerbarth, 2015-05-24:
Let's take this over to https://github.com/Wackerbarth/Marlin/issues/1 while we get things smoothed out.

fizcris, 2015-05-24:
Thank you very very much Wackerbarth!!!

It finally compiled! I also needed the SPI library but that was easy to find.

But I still wasn't able to deploy the code to the card, the error using [BootloaderCDC]Printrboard as target board is:
Error while uploading: missing 'upload.tool' configuration parameter

And, if I try to use [BootloaderCDC]AT90USB1286 as target board, the Arduino IDLE gets stucked while compiling...

Wackerbarth, 2015-05-25:
This is more than a documentation issue.

We will need to make some changes to the serial handlers and properly define boards.txt for the mcc's that we support.

boelle, 2015-06-01:
well i just tested marlin with the standard config (ramps) and no issue with ide 1.6.4

Wackerbarth, 2015-06-02:
@boelle --

I've figured out what seems to be happening.

I think that the following explains the differences that you and I are observing:

There are 3 basic configurations for Serial I/O.
1) Serial I/O based on the ATmega16U2 (or similar) interface chip as found on the Arduino Mega 2560.
2) A Bluetooth Serial I/O using an alternate UART.
3) Integrated USB Serial I/O as found on AT90USB1286 chips.

In case (1), the Marlin code REPLACES the Serial I/O with its own implementation.
Thus, when you switch to the IDE 1.6.4, in reality, you are getting a mostly new "core", but with very old Serial drivers. This, as you observe, appears to function acceptably.

However, for the other cases, there are issues. In case (3), it "compiles", but does not function.

thinkyhead, 2016-03-26:
Do we call this issue solved and closed? We've got much newer code and newer tools. If things are building well now, let's close this, as I am also looking to deprecate older build environments. If for any reason this should be re-opened, please feel free.

closed 2016-03-26 by thinkyhead

fmalpartida, 2016-03-26:
The code base has gone up so much from RC3 to RC4 that it doesn't fit any more in smaller devices. 
Configuration:
- coreXY
- thinned down  MMI with graphical LCD
- auto leveling (without repetitivity test or debug)

Will not fit on 128K devices, memory usage goes beyond that value.

thinkyhead, 2016-03-27:
@fmalpartida Do you have `DEBUG_LEVELING_FEATURE` disabled?

thinkyhead, 2016-03-27:
@fmalpartida I'm surprised it has gone up so much. For my configuration the size has gone down quite a bit. We'll have to see where we can trim some bytes for you.

fmalpartida, 2016-03-27:
I will try to post the configuration later on. Baby stepping is not enabled. 

I have already trimmed down a wide range of LCD features on the build. I went back to rc3 for the time being and need to merge changes back into an rc4 integ branch and share.

fmalpartida, 2016-03-27:
What has gone into the LCD code lately?

thinkyhead, 2016-03-27:
> What has gone into the LCD code lately?

"PID Autotune" is now in the menus, disabled by default. I'm wondering if new fonts or graphics or something else might have been added to support graphical LCDs. If you have any luck narrowing down which are the most expensive changes since RC3, maybe we can figure a way to squash the code down to a more reasonable size.

Have you tried different versions of Arduino and/or different optimization flags to see if you can get smaller code?

fmalpartida, 2016-03-27:
I am currently using version 1.0.5/6 for compiling. Its an old version but in general it works fairly well.

thinkyhead, 2016-03-27:
Some of those old versions of Arduino have quirky bugs in the compiler. Arduino 1.0.6 in particuar has been getting flagged a lot. You should try 1.6.7 or 1.6.8 to ensure that the code compiles as well as it can, and perhaps you will get a slightly smaller binary.

Incidentally, how much over your 128K limit is Marlin now? Do we need to shave off a few hundred bytes, just a few K, or more like 5K?

fmalpartida, 2016-03-27:
The main problem here is that AT90USB procesadores no lo ver work well on 1.6.x. Interruptor Management has changed in the platforms in Arduino env. Making all compilations unreliable. To make things worse, cdc bootloader support on Avrdude on those versions only support up to 64K!

We are talking about thinking tmdown about 3-4K. I have cut a good range of menu options and I can now get it to fit. I feel the main problem is around the MMI: no user interface: 72K, with MMI only 121K! Almost double.

thinkyhead, 2016-03-28:
MMI being the "manual movement interface"?

fmalpartida, 2016-03-28:
:-) The LCD interface. It looks also as the selection menus for temperature, movement do add a fair bit. Things that require a selector value.

Wackerbarth, 2016-03-28:
"To make things worse, cdc bootloader support on Avrdude on those versions only support up to 64K!"

Please be clear. The CDC bootloader uses a protocol that is limited to 64K. Period. It should never have worked with any version of avrdude.  However, some earlier versions of that program contained a error which allowed it to work much of the time.  As other changes to avrdude were made, that error was corrected and the current versions of avrdude no longer contain that loophole.

You can still use the modern IDE by installing the obsolete version of avrdude which was packaged with the 1.0.5 IDE.

Alternately, you can install a different bootloader which is designed for devices which exceed 64K. But reflashing the bootloader is "beyond the capability" of the typical user.

Wackerbarth, 2016-03-28:
"The main problem here is that AT90USB procesadores no lo ver work well on 1.6.x. Interruptor Management has changed in the platforms in Arduino env. Making all compilations unreliable."

The issue here is with the Arduino core routines and the way Marlin has altered the serial I/O routines.
The build environment that I set up addresses this by using the up-to-date compiler from the 1.6 iDE, but retains the Arduino core code from the 1.0 era.

Fortunately, the 1.6 IDE allows me to set up a "mix and match" environment which can be automatically downloaded. Thus I am able to have the improved features of the more recent IDE and compilers while maintaining the older code base upon which the Marlin interrupt code was written.

andreiciocanvlad, 2016-04-22:
maybe i`m posting in wrong place, how to make latest Arduino IDE work with marlin and a Gen7 Board ? i can't seem to make it see the board in IDE

bobc, 2016-04-22:
@andreiciocanvlad  Do you mean Arduino release 1.6.8 or a nightly build? 
I suggest raising a new issue, and also let us know what version of Gen7 board you have.

andreiciocanvlad, 2016-04-22:
I tryed on 1.6.5 with the ArduinoAddons, and in tools-> boards i did not find anything about gen7. It is a custom gen7 with 1284p uc. I have an older setup with ide 1.0.4 that worked ( 0.9 marlin ) but could not compile the newer marlin 1.1

jbrazio, 2016-04-22:
@andreiciocanvlad open a new ticket please.

Jordivanes2, 2018-05-28:
just clear the temp folder, worked for me on arduino 1.6.4, same issue as you guys
C:\Users\esvanj\AppData\Local\Temp\