MarlinFirmware/Marlin #2152: How change pulse width in Marlin 1.0 firmware for AM882 stepper motor?

newb3d, 2015-05-24:
Hello guys
I use Rumba with Marlin 1.0 firmware with AM882 stepper motor driver to power 2 stepper motors on Z axis.
1, 7,8, 10 pins connected to stepper motor according to this schematic
http://reprap.org/mediawiki/images/2/24/RRD-RUMBA_SCHEMATICS.png
If ENA unconnected motor "lock" shaft, if connected I can spin shaft by my hand freely.
Oscilloscope showed me small pulses when I tried to run z axis but they seems pretty short to reaction of AM882.
How to make pulse wider?
Thank you in advance.

boelle, 2015-06-01:
i think the people who code on marlin would also need a datasheet for the stepper driver so that they can figure how wide the pulse should be

thinkyhead, 2015-06-10:
See issue #1900 for discussion about making the pulse wider.

closed 2016-04-23 by jbrazio