MarlinFirmware/Marlin #2155: Better LCD/SD Configuration

thinkyhead, 2015-05-25:
Currently to configure an LCD controller, SD reader, etc., you must un-comment one or more of the following defines in `Configuration.h`:
```
//#define ULTRA_LCD
//#define DOGLCD
//#define SDSUPPORT
//#define SDSLOW
//#define SD_CHECK_AND_RETRY
//#define ENCODER_PULSES_PER_STEP 1
//#define ENCODER_STEPS_PER_MENU_ITEM 5
//#define ULTIMAKERCONTROLLER
//#define ULTIPANEL
//#define LCD_FEEDBACK_FREQUENCY_DURATION_MS 100
//#define LCD_FEEDBACK_FREQUENCY_HZ 1000

//#define PANEL_ONE

//#define MAKRPANEL

//#define VIKI2
//#define miniVIKI

#define ELB_FULL_GRAPHIC_CONTROLLER

//#define REPRAP_DISCOUNT_SMART_CONTROLLER

//#define G3D_PANEL

//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

//#define REPRAPWORLD_KEYPAD
//#define REPRAPWORLD_KEYPAD_MOVE_STEP 10.0

//#define RA_CONTROL_PANEL

//#define LCD_I2C_SAINSMART_YWROBOT
//#define LCD_I2C_PANELOLU2
//#define LCD_I2C_VIKI

//#define SAV_3DLCD
```
Some of the granular options are set explicitly in `Conditionals.h` with certain controllers. For example:
```
#if defined(MAKRPANEL)
  #define DOGLCD
  #define SDSUPPORT
  #define DEFAULT_LCD_CONTRAST 17
  #define ULTIPANEL
  #define NEWPANEL
#endif
```
The option `SDCARDDETECTINVERTED` probably should be included in the first group of options, but instead, it's defined in `Configuration_adv.h` in such a way that it will always be _enabled_ by default. However, if `ULTIPANEL` is enabled (as it is with various controllers) then it is forced to be _disabled_ (and this cannot be overridden from a configuration file).

It might be better if this option were set based on the controller. Currently that means it will be _disabled_ for all `ULTIPANEL` controllers (except for `ELB_FULL_GRAPHIC_CONTROLLER` which needs it enabled) and _enabled_ for all other controllers. (Another exception is Makibox which has a "slow" SD slot but no display).

Can we determine what all the LCD/SD-related settings ought to be for each display? And, can we enable/disable them appropriately in the LCD/SD section of `Conditionals.h` so that they don't need to be set separately in most cases? How easy is this to accomplish without breaking configs?

boelle, 2015-06-01:
to me is sounds strange that ultipanel blocks the disable.... maybe they did it just to make sure it was on?

anyway my thoughts is that you should not need to override anything but only have to set what display you have and then all the settings that are tied to that choice are effective

@daid & @ErikZalm can you cast any light on this?

daid, 2015-06-01:
ULTIPANEL is a specific hardware implementation:
http://www.thingiverse.com/thing:15081

So it makes sense that it forces the SDDetect in a certain way.

boelle, 2015-06-01:
ok, but why force it? why not just set it if the ultipanel is used?

if the user then changes the firmware and screws up then its the user to blame and not marlin

thinkyhead, 2015-06-10:
@boelle @daid The main reason I opened this issue is that `ELB_FULL_GRAPHIC_CONTROLLER` introduced an exception. Although it specifies that it is an `ULTIPANEL` display, it wants to be able to have `SDCARDDETECTINVERTED` on or off according to user preference (in case the user implements the display differently, according to the designer). So, is the author just incorrect in calling his display an `ULTIPANEL`, or is he incorrect in wanting to allow `SDCARDDETECTINVERTED` to be editable? So far, this ELB display is the _only exception_ to the "rule" that `SDCARDDETECTINVERTED` must be disabled for `ULTIPANEL`.

eboston, 2015-06-10:
The defining of ULTIPANEL was needed because that enables other aspects of the panel, like the encoder if I remember right.  The whole problem with the LCD support is separate parts of the panel, like the encoder and sd, are not defined as independent items.

Look at Marlin_main.cpp where powersupply is set.  It requires ULTIPANEL be set even though other panels might support the power supply option.

Gcode M0, M1, M145, M117, and other, are only defined if ULTIPANEL is defined.  These are all LCD functions, not specific to just one panel.

ULTIPANEL has become a define that there is an LCD panel, not a specific one anymore.

daid, 2015-06-10:
@eboston Then that behaviour is wrong and should be fixed. Instead of abusing the ULTIPANEL define.

Wackerbarth, 2015-06-10:
@daid - I agree. It is inappropriate to write code that uses an instance rather than the property to select the proper variant of the code.
#ifdef ULTIPANEL
  #define HAS_LCD_PANEL
 ....
#endif

eboston, 2015-06-10:
Yes, the defines need to be fixed.  But my display is not the only one abusing the ULTIPANEL define.  Pretty much every graphic LCD panel uses the define, even though they are not an ULTIPANEL.  It's just that the one I am making had one difference to the other panels.

Maybe as a simple fix to this, we can just rename ULTIPANEL to HAS_LCD_PANEL.  But even if we did that, we still have the original problem I had to work around.

Wackerbarth, 2015-06-10:
I'm suggesting that we rename ULTIPANEL to HAS_LCD_PANEL in the code that is compiled for any panel. for example the part that enables M0, M1, etc.

SDCARDDETECTINVERTED, logically, is a separate characteristic.
Since you need it, your configuration would have

#define HAS_LCD_PANEL
#define SDCARDDETECTINVERTED

An ULTIPANEL would not.

If Scott wants to ASSERT that SDCARDDETECTINVERTED is not true for an ULTIPANEL, he can do so.
However, it would be incorrect to ASSERT that it is false just because HAS_LCD_PANEL is TRUE

thinkyhead, 2015-06-12:
> If Scott wants to ASSERT that `SDCARDDETECTINVERTED` is not true for an `ULTIPANEL`, he can do so.

Alas, I cannot take credit for the code that makes this assertion. It was there before I arrived on the scene.

Wackerbarth, 2015-06-12:
We know that ULTIPANEL is NOT the same as HAS_LCD_PANEL. There are some, LCD panels, at least one that we can name, which do not have the same SD card detectors.

Therefore we should say that
ULTIPANEL implies HAS_LCD_PANEL

The issue is in the code. At the present time, there are conditional compilations based on the ULTIPANEL token. However, in most cases, the code also works for a more general LCD panel.  In those cases, the HAS_LCD_PANEL token would be more appropriate.

Fundamentally, it build down to examining EVERY use of the ULTIPANEL token and deciding if that code instance is specific to an ULTIPANEL, or if it applied to the larger class of HAS_LCD_PANEL configurations.

Making those changes will not "break" any configuration which actually has an ULTIPANEL.
Nor will it affect configurations which have not LCD panel.
That leaves the configurations which have non-ULTIPANEL LCD displays.

This type of change will fix some issues. However, it might break some things.
In particular, if some non-ULTIPANEL was "riding the coattails" of an ULTIPANEL and has a property that is like the ULTIPANEL, but not universal, then it will fail because we have not split it off as an an additional feature.

fmalpartida, 2015-06-12:
I see this more broken into several blocks. First are things that are normally included in panels:
HAS_KEYS, HAS_ENCODER, HAS_LCD, HAS_CARD_READER, HAS_SOUND, ...

Panels are defined in a panel definition file that includes their building blocks.
The main configuration file defines available supported panels straight of the bat.
If HAS_LCD, it should have one of the supported drivers available and their supported properties: FEEDBACK, CONTRAST and BACKLIGHT.

If you look at how they are being used, it is more or less what the ULTIPANEL and NEWPANEL convey.

closed 2016-05-04 by jbrazio