MarlinFirmware/Marlin #2158: zprobe_zoffset sign on LCD

chris-bo, 2015-05-25:
In Configuration.h and M851 `zprobe_zoffset` has to be applied negative (probe below extruder), but internally `zprobe_zoffset` is saved as a positive number. So on LCDs its shown positive. 
If you now limit the allowed values to negative numbers to make shure the extruder doesn't crash into the bed, you can't adjust zprobe_offset via LCD.

thinkyhead, 2015-05-26:
The default limits for editing `zprobe_zoffset` on the LCD are:

    #define Z_PROBE_OFFSET_RANGE_MIN -20
    #define Z_PROBE_OFFSET_RANGE_MAX 20

So, you should be able to select a value from -20 (probe 20mm higher than the nozzle) up to +20 (probe 20mm lower than the nozzle). Are you unable to select values in this range?

chris-bo, 2015-05-26:
My problem is the confusion with signs.
I think , you described the way the sign is interpreted by the code and LCD.

But here it is the exact opposite (Configuration.h):
```
#define Z_PROBE_OFFSET_FROM_EXTRUDER -12.35 // -below (always!)
```
And for `M851` you also need to send a negative value for a probe lower than the nozzle.
This sign will then be changed by the gcode handling (Marlin_main.cpp):
```      
if (Z_PROBE_OFFSET_RANGE_MIN <= value && value <= Z_PROBE_OFFSET_RANGE_MAX) {
        zprobe_zoffset = -value;
        SERIAL_ECHO_START;
        SERIAL_ECHOLNPGM(MSG_ZPROBE_ZOFFSET " " MSG_OK);
        SERIAL_EOL;
      }
```

thinkyhead, 2015-05-27:
@chris-bo I'm not sure why it was done this way. To make everything consistent we would only have to change the on-screen value so that it has the opposite sign. That's the kind of UI improvement we should make, because most people will appreciate it.

thinkyhead, 2015-05-27:
#2167

boelle, 2015-06-01:
since there is a PR for this i will close this one... the PR will prob not be merged before all bugs are fixed

closed 2015-06-01 by boelle