MarlinFirmware/Marlin #2160: Not so autoleveling

LazyPrinter, 2015-05-25:
[image]
[image]

I've checked the servo mounting and Z switch, and neither are loose. The probing is always on the glass. 

The Z offset is perfect, but the autoberd leveling seems to be failing, I'm not sure why.

 #ifdef AUTO_BED_LEVELING_GRID

    // set the rectangle in which to probe
    #define LEFT_PROBE_BED_POSITION 40
    #define RIGHT_PROBE_BED_POSITION 190
    #define BACK_PROBE_BED_POSITION 150 //Rear Left to Front Left
    #define FRONT_PROBE_BED_POSITION 40

     // set the number of grid points per dimension
     // I wouldn't see a reason to go above 3 (=9 probing points on the bed)
    #define AUTO_BED_LEVELING_GRID_POINTS 2

  #else  // not AUTO_BED_LEVELING_GRID
    // with no grid, just probe 3 arbitrary points.  A simple cross-product
    // is used to esimate the plane of the print bed

      #define ABL_PROBE_PT_1_X 15
      #define ABL_PROBE_PT_1_Y 180
      #define ABL_PROBE_PT_2_X 15
      #define ABL_PROBE_PT_2_Y 20
      #define ABL_PROBE_PT_3_X 170
      #define ABL_PROBE_PT_3_Y 20

  #endif // AUTO_BED_LEVELING_GRID

  // these are the offsets to the probe relative to the extruder tip (Hotend - Probe)
  // X and Y offsets must be integers
  #define X_PROBE_OFFSET_FROM_EXTRUDER -26 
  #define Y_PROBE_OFFSET_FROM_EXTRUDER 16
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -8.58 //11.70 //13.86

  #define Z_RAISE_BEFORE_HOMING 4       // (in mm) Raise Z before homing (G28) for Probe Clearance.
                                        // Be sure you have this distance over your Z_MAX_POS in case

  #define XY_TRAVEL_SPEED 4000         // X and Y axis travel speed between probes, in mm/min

  #define Z_RAISE_BEFORE_PROBING 30    //How much the extruder will be raised before traveling to the first probing point.
  #define Z_RAISE_BETWEEN_PROBINGS 5  //How much the extruder will be raised when traveling from between next probing points

  //#define Z_PROBE_SLED // turn on if you have a z-probe mounted on a sled like those designed by Charles Bell
  //#define SLED_DOCKING_OFFSET 5 // the extra distance the X axis must travel to pickup the sled. 0 should be fine but you can push it further if you'd like.

  //If defined, the Probe servo will be turned on only during movement and then turned off to avoid jerk
  //The value is the delay to turn the servo off after powered on - depends on the servo speed; 300ms is good value, but you can try lower it.
  // You MUST HAVE the SERVO_ENDSTOPS defined to use here a value higher than zero otherwise your code will not compile.

  #define PROBE_SERVO_DEACTIVATION_DELAY 300

//If you have enabled the Bed Auto Leveling and are using the same Z Probe for Z Homing,
//it is highly recommended you let this Z_SAFE_HOMING enabled!!!

  #define Z_SAFE_HOMING   // This feature is meant to avoid Z homing with probe outside the bed area.
                          // When defined, it will:
                          // - Allow Z homing only after X and Y homing AND stepper drivers still enabled
                          // - If stepper drivers timeout, it will need X and Y homing again before Z homing
                          // - Position the probe in a defined XY point before Z Homing when homing all axis (G28)
                          // - Block Z homing only when the probe is outside bed area.

  #ifdef Z_SAFE_HOMING

    #define Z_SAFE_HOMING_X_POINT (X_MAX_LENGTH/2)    // X point for Z homing when homing all axis (G28)
    #define Z_SAFE_HOMING_Y_POINT (Y_MAX_LENGTH/2)    // Y point for Z homing when homing all axis (G28)

  #endif

Wackerbarth, 2015-05-25:
I need for you to explain in more detail just what it does, and how that differs from what you expect.

Does it compile?  What codes do you issue to attempt leveling? Does it go through the motions? If so, what values are reported? And, finally, what do you expect?

LazyPrinter, 2015-05-25:
I'll see if I can get data from multiple autobed (G29) later on, as the printer is going to be busy for the next 8 hours. 

I first use G28, when homes X and Y, then centers for the Z. G29 runs on the furthest I can goto to each corner.

The issue I'm having, the autobed is having different values everytime It's ran. The numbers are different everytime it probes a corner, but the autobed function should be compensating for the uneaveness based on the Z axis offset, which it seems It's not.

I can further give the details of the probes later.

Wackerbarth, 2015-05-25:
Thanks for the additional explanation. I now understand the nature of your issue.

It would be instructive to know just how much variance, from pass to pass, you are seeing.

LazyPrinter, 2015-05-26:
Ran 4 G29's and a single G28, here are the results

12:26:55.335 : Bed x: 40.00 y: 40.00 z: 8.44
12:27:00.399 : Bed x: 190.00 y: 40.00 z: 9.08
12:27:04.965 : Bed x: 190.00 y: 150.00 z: 8.89
12:27:10.158 : Bed x: 40.00 y: 150.00 z: 8.38
12:27:10.165 : Eqn coefficients: a: 0.00 b: -0.00 d: 8.37
12:27:10.167 : planeNormal x: -0.00 y: 0.00 z: 1.00
12:27:10.170 : echo:endstops hit:  Z:8.38
12:27:30.728 : Bed x: 40.00 y: 40.00 z: 8.32
12:27:35.904 : Bed x: 190.00 y: 40.00 z: 8.56
12:27:40.414 : Bed x: 190.00 y: 150.00 z: 8.70
12:27:45.596 : Bed x: 40.00 y: 150.00 z: 8.14
12:27:45.602 : Eqn coefficients: a: 0.00 b: -0.00 d: 8.15
12:27:45.604 : planeNormal x: -0.00 y: 0.00 z: 1.00
12:27:45.607 : echo:endstops hit:  Z:8.14
12:27:56.746 : Bed x: 40.00 y: 40.00 z: 8.85
12:28:01.837 : Bed x: 190.00 y: 40.00 z: 9.47
12:28:06.369 : Bed x: 190.00 y: 150.00 z: 9.40
12:28:11.553 : Bed x: 40.00 y: 150.00 z: 8.83
12:28:11.560 : Eqn coefficients: a: 0.00 b: -0.00 d: 8.72
12:28:11.562 : planeNormal x: -0.00 y: 0.00 z: 1.00
12:28:11.565 : echo:endstops hit:  Z:8.83
12:28:21.417 : Bed x: 40.00 y: 40.00 z: 8.82
12:28:26.512 : Bed x: 190.00 y: 40.00 z: 9.40
12:28:31.051 : Bed x: 190.00 y: 150.00 z: 9.28
12:28:36.235 : Bed x: 40.00 y: 150.00 z: 8.68
12:28:36.241 : Eqn coefficients: a: 0.00 b: -0.00 d: 8.71
12:28:36.243 : planeNormal x: -0.00 y: 0.00 z: 1.00
12:28:36.246 : echo:endstops hit:  Z:8.68

brainscan, 2015-05-26:
Have you tested the repeatability of your probe? Set it to probe a single point 10 times and see how consistent it is. It's the best place to start.

LazyPrinter, 2015-05-26:
I'm not sure how to do that. Under another print (still in progress) I've found that my entire bed is "twisted" by around 0.2-0.6mm from the front right corner to the rear left

brainscan, 2015-05-26:
I've been using an old modified version of marlin so I haven't used the official version for a while but I'm sure that the repeatability code has been added since. Roxy came up with the idea to prove the same point over and over and average the results to show how accurate the probe works. I'm at work at the moment so can't look at my machine unfortunately but if you have a search there should be the instructions for running it. If not I'll check it out later and get back to you. There are a lot of places that can cause problems, such as your twisted bed. You can use G29 to output the bed topology to help manually flatten/level the bed which helps as the auto level then has very little to do.

Roxy-3DPrintBoard, 2015-05-26:
M48 help you measure the 'repeatability' of your probe and positioning hardware.

brainscan, 2015-05-26:
Thanks Roxy!! I couldn't remember which code it was, I've not had to touch it since you helped me last year!! I haven't plucked up the courage to try the latest marlin yet, worried I won't get things working again without more help!

Roxy-3DPrintBoard, 2015-05-26:
> Thanks Roxy!! I couldn't remember which code it was, I've not had to touch it since you helped me last year!! I haven't plucked up the courage to try the latest marlin yet, worried I won't get things working again without more help!

My advice would be to just keep using the version you have for a while.   When things settle out and get stable I think we will start another thread over at 3DPrintBoard and encourage/Help everybody get caught up and current.

LazyPrinter, 2015-05-26:
I ran M84 P10 X100 Y100, results were

23:52:13.353 : M48 Z-Probe Repeatability test.   Version 2.00
23:52:13.355 : Full support at: http://3dprintboard.com/forum.php
23:53:37.335 : Mean: 8.990875
23:53:37.337 : Standard Deviation: 0.028274
23:53:37.338 : echo:endstops hit:  Z:9.01

LazyPrinter, 2015-05-28:
Anyone able to help with my inconsistant autobed leveling?

Roxy-3DPrintBoard, 2015-05-28:
Well...   You seem to have some 'slop' in your machine and/or probe.   .03 mm (rounded up) is a lot of error.   With that said, you still should be able to get decent prints if you can get the first layer down OK.

LazyPrinter, 2015-05-28:
I'll have to play around with the axis, especially where the carrige is to see if their's any play. This printer is brand new and is from CTC. Unsupprisingly, I think CTC is the cause of this.

LazyPrinter, 2015-05-29:
I can confirm their's play on my X and Y axis, only in the Z positions. I'll do fixes then post the new Z-Probe repeatability test results here.

LazyPrinter, 2015-05-29:
Printing new parts, so I'll report back here within a couple of days with my results.

New parts:
http://www.thingiverse.com/thing:726567
http://www.thingiverse.com/thing:535793

boelle, 2015-06-01:
cant wait to see if this was just slop :-)

LazyPrinter, 2015-06-01:
Ok, results are here!

I ran M48 P10 X100 Y100, same as the previous test.

18:36:02.799 : Mean: 8.784626
18:36:02.800 : Standard Deviation: 0.004907
18:36:02.802 : echo:endstops hit:  Z:8.79

boelle, 2015-06-01:
not bad

brainscan, 2015-06-02:
That's definitely more like the accuracy you need. Reducing the z homing speed can help a little sometimes but it does slow down the whole process so it's a bit of a balancing act.

LazyPrinter, 2015-06-03:
Still not fixed, but it is marginally off, so I installed the good old wing nuts with springs so I can fine tune when printing.

It is off  by +/- 0.1mm. Sometimes a single corner can be too far, or too close and sometimes it's 2-3 corners.

Roxy-3DPrintBoard, 2015-06-03:
> It is off by +/- 0.3mm. Sometimes a single corner can be too far, or too close and sometimes it's 2-3 corners.

You might want to do a G29 P 5 T  command to see the bed topography.  Depending upon what that shows, different approaches may make sense.   If you post your Bed Topography, it will be more clear what you are fighting.

LazyPrinter, 2015-06-03:
Printer doing a print, I'll do the command later tonight (or early in the morning)

LazyPrinter, 2015-06-03:
I ran a G29 P 5 T, results:

18:24:49.368 : Bed x: 40.00 y: 40.00 z: 10.31
18:24:54.665 : Bed x: 190.00 y: 40.00 z: 9.14
18:24:59.355 : Bed x: 190.00 y: 150.00 z: 7.99
18:25:04.483 : Bed x: 40.00 y: 150.00 z: 8.89
18:25:04.490 : Eqn coefficients: a: -0.01 b: -0.01 d: 10.99
18:25:04.491 : planeNormal x: 0.01 y: 0.01 z: 1.00
18:25:04.494 : echo:endstops hit:  Z:8.89

Roxy-3DPrintBoard, 2015-06-03:
The G29 T should have given you a topographical map.

LazyPrinter, 2015-06-03:
Should I do G28, G29, then G29 P 5 T? I'm not sure with the commands

LazyPrinter, 2015-06-03:
I've tried G29 T, their is no different info than what's above.

Roxy-3DPrintBoard, 2015-06-03:
Apparently that code is not in the version of Merlin you are using.   You might need to load a newer branch (temporarily!!!) to get that information.

LazyPrinter, 2015-06-03:
The settings I used are at the top of this issue. Using the latest Dev release.

In file included from E:\Temp\build5667710504165244660.tmp\Configuration_adv.h:588:0,
                 from E:\Temp\build5667710504165244660.tmp\Configuration.h:812,
                 from E:\Temp\build5667710504165244660.tmp\Marlin.h:22,
                 from E:\Temp\build5667710504165244660.tmp\blinkm.cpp:5:
E:\Temp\build5667710504165244660.tmp\SanityCheck.h:189:10: error: #error The Y axis probing range is too small to fit all the points defined in AUTO_BED_LEVELING_GRID_POINTS
         #error The Y axis probing range is too small to fit all the points defined in AUTO_BED_LEVELING_GRID_POINTS
          ^
Error compiling.

thinkyhead, 2015-06-10:
@LazyPrinter Are you sure those are the settings you used? The test that produces that error works thus:
```
#define PROBE_SIZE_Y (Y_PROBE_OFFSET_FROM_EXTRUDER * (AUTO_BED_LEVELING_GRID_POINTS-1)) // 16 * (2-1) = 16
#define PROBE_AREA_DEPTH (BACK_PROBE_BED_POSITION - FRONT_PROBE_BED_POSITION) // 150 - 40 = 110
#if PROBE_SIZE_Y >= PROBE_AREA_DEPTH // 16 >= 110 ?
  #define Y_PROBE_ERROR
#endif
```
The settings you have shown above should not invoke this error according to my reading.

LazyPrinter, 2015-06-10:
The settings I used on the Dev release:

//===========================================================================
//============================ Bed Auto Leveling ============================
//===========================================================================

// @section bedlevel

#define ENABLE_AUTO_BED_LEVELING // Delete the comment to enable (remove // at the start of the line)
#define Z_PROBE_REPEATABILITY_TEST  // If not commented out, Z-Probe Repeatability test will be included if Auto Bed Leveling is Enabled.

#ifdef ENABLE_AUTO_BED_LEVELING

  // There are 2 different ways to specify probing locations
  //
  // - "grid" mode
  //   Probe several points in a rectangular grid.
  //   You specify the rectangle and the density of sample points.
  //   This mode is preferred because there are more measurements.
  //
  // - "3-point" mode
  //   Probe 3 arbitrary points on the bed (that aren't colinear)
  //   You specify the XY coordinates of all 3 points.

  // Enable this to sample the bed in a grid (least squares solution)
  // Note: this feature generates 10KB extra code size
  #define AUTO_BED_LEVELING_GRID

  #ifdef AUTO_BED_LEVELING_GRID

    #define LEFT_PROBE_BED_POSITION 40
    #define RIGHT_PROBE_BED_POSITION 190
    #define FRONT_PROBE_BED_POSITION 150
    #define BACK_PROBE_BED_POSITION 40

    #define MIN_PROBE_EDGE 10 // The probe square sides can be no smaller than this

    // Set the number of grid points per dimension
    // You probably don't need more than 3 (squared=9)
    #define AUTO_BED_LEVELING_GRID_POINTS 3

  #else  // !AUTO_BED_LEVELING_GRID

      // Arbitrary points to probe. A simple cross-product
      // is used to estimate the plane of the bed.
      #define ABL_PROBE_PT_1_X 15
      #define ABL_PROBE_PT_1_Y 180
      #define ABL_PROBE_PT_2_X 15
      #define ABL_PROBE_PT_2_Y 20
      #define ABL_PROBE_PT_3_X 170
      #define ABL_PROBE_PT_3_Y 20

  #endif // AUTO_BED_LEVELING_GRID

  // Offsets to the probe relative to the extruder tip (Hotend - Probe)
  // X and Y offsets must be integers
  #define X_PROBE_OFFSET_FROM_EXTRUDER -26     // Probe on: -left  +right
  #define Y_PROBE_OFFSET_FROM_EXTRUDER 11   // Probe on: -front +behind
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -12.65  // -below (always!)

  #define Z_RAISE_BEFORE_HOMING 4       // (in mm) Raise Z before homing (G28) for Probe Clearance.
                                        // Be sure you have this distance over your Z_MAX_POS in case

  #define XY_TRAVEL_SPEED 8000         // X and Y axis travel speed between probes, in mm/min

  #define Z_RAISE_BEFORE_PROBING 20   //How much the extruder will be raised before traveling to the first probing point.
  #define Z_RAISE_BETWEEN_PROBINGS 5  //How much the extruder will be raised when traveling from between next probing points
  #define Z_RAISE_AFTER_PROBING 15    //How much the extruder will be raised after the last probing point.

//   #define Z_PROBE_END_SCRIPT "G1 Z10 F12000\nG1 X15 Y330\nG1 Z0.5\nG1 Z10" //These commands will be executed in the end of G29 routine.
[... 24 lines cut ...]
    #define Z_SAFE_HOMING_Y_POINT (Y_MAX_LENGTH/2)    // Y point for Z homing when homing all axis (G28)

  #endif

  // Support for a dedicated Z PROBE endstop separate from the Z MIN endstop.
  // If you would like to use both a Z PROBE and a Z MIN endstop together or just a Z PROBE with a custom pin, uncomment #define Z_PROBE_ENDSTOP and read the instructions below.
  // If you want to still use the Z min endstop for homing, disable Z_SAFE_HOMING above. Eg; to park the head outside the bed area when homing with G28.
  // WARNING: The Z MIN endstop will need to set properly as it would without a Z PROBE to prevent head crashes and premature stopping during a print.
  // To use a separate Z PROBE endstop, you must have a Z_PROBE_PIN defined in the pins.h file for your control board.
  // If you are using a servo based Z PROBE, you will need to enable NUM_SERVOS, SERVO_ENDSTOPS and SERVO_ENDSTOPS_ANGLES in the R/C Servo below.
  // RAMPS 1.3/1.4 boards may be able to use the 5V, Ground and the D32 pin in the Aux 4 section of the RAMPS board. Use 5V for powered sensors, otherwise connect to ground and D32
  // for normally closed configuration and 5V and D32 for normally open configurations. Normally closed configuration is advised and assumed.
  // The D32 pin in Aux 4 on RAMPS maps to the Arduino D32 pin. Z_PROBE_PIN is setting the pin to use on the Arduino. Since the D32 pin on the RAMPS maps to D32 on Arduino, this works.
  // D32 is currently selected in the RAMPS 1.3/1.4 pin file. All other boards will need changes to the respective pins_XXXXX.h file.
  // WARNING: Setting the wrong pin may have unexpected and potentially disastrous outcomes. Use with caution and do your homework.

  //#define Z_PROBE_ENDSTOP

#endif // ENABLE_AUTO_BED_LEVELING

Triggers 
E:\Temp\build4645354219055017273.tmp\SanityCheck.h:189:10: error: #error The Y axis probing range is too small to fit all the points defined in AUTO_BED_LEVELING_GRID_POINTS
         #error The Y axis probing range is too small to fit all the points defined in AUTO_BED_LEVELING_GRID_POINTS
          ^
Error compiling.

thinkyhead, 2015-06-12:
Try swapping your `BACK_PROBE_POSITION` and `FRONT_PROBE_POSITION` values. The back position should always be larger than the front.

LazyPrinter, 2015-06-15:
That does compile. I'm very busy at the moment, but I'll post the results asap. It seems as though the autobed "might" be fixed since I replaced my extruder to a dual hobbed bolt, which seems to still push  filament even when the layer height is less than 0.02mm..

thinkyhead, 2015-06-15:
Wow. Hobbed bolts always impress me with their grippy-ness. I use an MK7 and it really needs a tight idler to be reliable sometimes.

KerwoodDerby, 2015-06-19:
@thinkyhead @LazyPrinter I had horrible auto bed levelling consistency, but I only had occasional times when I could investigate it, so it'd had been slow going. One investigation involved me running an M48 test with and without the "E" parameter, which forces a probe stow and extend operation between each test -- there was a substantial difference between the two. When run with the "E" parameter the standard deviation was 10 times the run done without that parameter.

At my next session, I acted on my suspicion that my servo was terrible at repeatability but excellent at holding position once settled. I commented out the line which would have stowed the z probe after the initial, central z probe. Five prints later I can state with fair certainty that I fixed the inconsistency; each print delivered a first layer whose thickness of deposition was identical and uniform.

Roxy-3DPrintBoard, 2015-06-19:
>  I acted on my suspicion that my servo was terrible at repeatability but excellent at holding position once settled. I commented out the line which would have stowed the z probe after the initial, central z probe. 

On the G29 command you have the option of holding the probe down during the entire session or having the probe retract and re-engage for each point.   You should be able to specify which way you want it to behave on the command line.

KerwoodDerby, 2015-06-19:
@Roxy-3DPrintBoard Yes, however the preceding G28 command does the initial, central probe, extending and then retracting the probe. G28 sets the Z origin which will be used by G29. My servo's bad repeatability was introducing a random error between the end of the G28 and the beginning of the G29 operation. So I commented out the final stow in the G28 code and my errors magically vanished.

Roxy-3DPrintBoard, 2015-06-19:
OK...   I understand what you did and why.    But I'm not convinced you need to do that.   The G28 gets an initial Z position.   But the G29 gets much more detailed information and at the end of the G29 it compensates for the Z_PROBE_OFFSET_FROM_EXTRUDER.   There is some debate on how that should be done, but what ever happens in the G28 gets over ruled by the end of the G29.

KerwoodDerby, 2015-06-19:
It's not clear to me that "what ever happens in the G28 gets over ruled by the end of the G29" is true. Why else would G28 be mandatory before G29? At any rate, all I can say is that commenting out that stow operation made a breathtaking difference.

Roxy-3DPrintBoard, 2015-06-20:
OK....   I believe you!   But what happens is both of those routines set the current position for Z based on where the nozzle is at that instant in time.    G29 is 'over ruling' what ever G28 set.

thinkyhead, 2015-06-28:
> Why else would `G28` be mandatory before `G29`?

`G28` establishes a known (uncorrected) position before attempting to probe, so the probes don't occur someplace way off the bed. After probing is completed, the current XYZ values are immediately corrected to where they "really are" based on the leveling plane. For example, if the bed is angled at 45° with the right side higher than the left, the X position at the center with the nozzle touching the bed is the same as the uncorrected X position. But at 45° when you move straight Z upwards, the X value increases to the right. Note that this is just for illustration purposes. Plane correction doesn't actually move XY based on Z movement. It only moves Z based on XY movement, but the same principle applies.

We can debate whether XY or Z need to be corrected at all at the end of probing. Theoretically, if you moved Z down (to 0) at the end of probing so the nozzle touches the bed, the XY and Z shouldn't _need_ to be corrected. It matters in this case that the plane correction doesn't alter XY in response to Z movement.

closed 2016-04-23 by jbrazio