MarlinFirmware/Marlin #2161: GCODE buffer overflow hangs Marlin?

MaxOLydian, 2015-05-25:
I recently added the auto bed leveling code to my printer, and have found that repeated M105 queries to the printer during long-running command (e.g. G29) cause the Marlin FW to hang, from what it seems is an incoming command buffer overflow.

Specifically, I control my printer with OctoPrint.  If I leave OctoPrint connected after issuing a G29 (autolevel) command to my printer, it continues to send M105 (temp update) commands.  By the time G29 completes, so many M105 commands have been sent that it seems the command buffer has been saturated- "M105" appears on the status line of the LCD display, it no longer sends any response to any command from OctoPrint, and needs a reboot to behave again.

If I shut down the connection from OctoPrint to the printer immediately after issuing the G29, and don't reconnect until after G29 has finished, everything seems to be fine (except I don't get to see the results of my autolevel).

I am using Marlin 1.0.2 on a RAMPS 1.4 driving a Prusa iv3.

thinkyhead, 2015-05-26:
@MaxOLydian This might also happen with `M303` and any other commands that take a while to run and which don't allow the buffer to be processed in the mean-time. Definitely needs to be looked at…

thinkyhead, 2015-05-26:
#2166 should allow communication to resume after `G29` or `M48`. Can you confirm that this fixes the problem?

MaxOLydian, 2015-05-30:
According to foosel, it's already been fixed in the devel branch.  I haven't had time to confirm myself, but check it out...

MaxOLydian, 2015-05-30:
Oops- sorry, I got my projects confused!  I also submitted this as a possible problem in Octoprint, and was told a workaround had already been applied.

I'll check out https://github.com/MarlinFirmware/Marlin/pull/2166 next weekend.

LazyPrinter, 2015-06-02:
I can cause Marlin to hang by simply sending too many commands (like move axis). I'm not sure if it's related.

daid, 2015-06-02:
@LazyPrinter with with host? As I never had this problem with Cura when I was developing the USB sender (which did an intentional overflow at the start to get as much stuff filled in the buffer to start with)

(Closing this otherwise, as @foosel confirmed that it was a bug in octoprint)

LazyPrinter, 2015-06-02:
Not with the host, but with the LCD control

AnHardt, 2015-06-02:
@LazyPrinter 
For the LCD this is a known bug (#1166) but now for all easiy to reproduce. When time comes we'll have a look on this. At the moment we fight bugs with higher impact.

foosel, 2015-06-02:
I'm not so sure the LCD control thing is the same issue as described in #1166. Quoting the OP with emphasis by me:

> By the time G29 completes, so many M105 commands have been sent that it seems the command buffer has been saturated- **"M105" appears on the status line of the LCD display**,

This sounds like an actual buffer overflow, as in, memory getting overwritten. Not like "LCD becomes unresponsive due to too many impulses from the rotary encoder", but more like "data is written where it shouldn't be written".

I'm not sure this issue should be closed just yet due to that reason, @daid - Yes, the problem can be triggered with a particular buggy behaviour in OctoPrint's stable branch, but M105 should **never** show up on the LCD unless I send `M117 M105` via the serial.

AnHardt, 2015-06-02:
Dear @foosel 
I specifically talked to @LazyPrinter who apparently has a known problem with the LCD control.
Yes this "M105" looks scaring. Hopefully we only misused a PSTR() again, and put out a random piece of memory instead of the wanted message. I fixed several of this in the last month.

foosel, 2015-06-02:
@AnHardt I'm getting my nicks mixed up again... Thought LazyPrinter was the OP, hence the confusion, sorry. Glad you agree that the LCD overflow should be investigated.

thinkyhead, 2015-06-09:
#2166 obviously isn't the solution. The parser code has been reworked a bit recently and has been tested, debugged, and patched up. I encourage host developers to try overflowing the input buffers or to try long operations and see what happens to commands sent during those processes. Please provide feedback as to whether responses seem proper, such as whether "resend" is working properly. Please suggest best remedy to deal with an unresponsive machine.

AnHardt, 2015-09-25:
Has anyone seen this problem recently?

thinkyhead, 2015-10-12:
Re-open if this issue crops up again.

closed 2016-04-23 by jbrazio