MarlinFirmware/Marlin #2162: LCD change filament command not working?

redsqrly, 2015-05-25:
Is there anything i need to define in Marlin to make the M600 command work? Currently LCD seems to freeze up after extruder moving in position and retracting. Clicking the button does nothing and i have to kill the print. Am i missing something? Thanks!

thinkyhead, 2015-05-26:
Note: `M600` won't work if the hot end is too cold. I don't think it gives an error in that case except "prevented dangerous extrude." That's something we need to fix. It probably helps to do a `G28` first also.

By default the command will do `M600 E-2 Z10 X3 Y3 L-100` which is supposed to result in these actions:
- Retract 2mm of filament
- Lift Z by 10mm
- Move to position 3, 3
- Retract 100mm of filament
- Display "Change filament" on the screen and wait for click
- Move back to starting X, Y
- Move back to starting Z
- Move back to starting E (Un-retract 2mm of filament)

redsqrly, 2015-05-27:
Hot end is definitely up to temp as I'm attempting to change filament mid print, everything works correctly up to "Display "Change filament" on the screen and wait for click", however it doesnt respond to the click and seems to freeze. I know it worked for me before on an older version of marlin, so not sure whats going wrong. Using the integration branch just now.

thinkyhead, 2015-05-27:
@redsqrly Does the lcd make any clicking or beeping sounds after it displays the message? Apparently it's supposed to make a click "every 256th frame."

redsqrly, 2015-05-27:
I have the buzzer pin disabled as it bugs the life out of me when i scroll... I shall try again with it on and report back. But i get the same result if i send M600 from Repetier mid print, nozzle moves in position, extruder retracts and everything freezes.

redsqrly, 2015-05-27:
Does it work fine on your machine @thinkyhead? My friend running release version on his Hephestos says his machine freezes up on M600 also.

thinkyhead, 2015-05-27:
The only thing unusual about `M600` is that it pauses to wait for a controller click. You might try this: comment out the wait loop in `gcode_M600` and see if it proceeds normally. If so, then we can poke around and see if there's some issue with constantly polling the controller button.

Wurstnase, 2015-05-28:
On my last try to implement this without sd-card printing i've got the same issue. Thought i made something wrong and stopped working on it.

thinkyhead, 2015-05-28:
@redsqrly @Wurstnase

I have been playing with the `FILAMENTCHANGEENABLE` option on my machine, and yes `M600` was in need of some fixing. #2191 addresses the obvious issues that stand out:
- `M600` would proceed even if the hot end was cold. I have added a patch so `M600` won't even start in this case.
- Rapid and continuous calls to `lcd_quick_feedback` (possibly because `lcd_update` was recently optimized). On my machine this made a continuous rapid and messy clicking sound. I patched the code to use the millisecond clock and to make a tick sound every 2.5 seconds.

redsqrly, 2015-05-28:
Thanks for looking into it @thinkyhead , much appreciated!!

boelle, 2015-06-01:
so this one can be closed as it is fixed?

AnHardt, 2015-07-18:
No response after 6 weeks. Closing.

closed 2015-07-18 by AnHardt