MarlinFirmware/Marlin #2165: Feature suggestion: Autoexecute file on SD card?

kefir-, 2015-05-26:
Would it make sense to be add some functionality to automatically read a file with a predefined name (eg /autoprnt.g) from the SD-card on power-on, and immediately execute the gcode there, or is this too error prone?

I see several potential issues if this isn't done eight, like unintentionally starting an old print when powering up a printer, or a power cycle on the grid causing a print to interrupt and then the nozzle to crash into the half printed part when the power returns. But issues like this can be fixed, I think. For example if Marlin renamed the file before executing it, it wouldn't rerun the same file accidentally. I'm sure there are other checks necessary to keep this safe. But this could be an optional feature, and when enabled it could be a very simple way to print files from SD without connecting a computer or a screen and interface.

nophead, 2015-05-26:
Marlin already has this feature. It runs AUTO0.G on power-on.

thinkyhead, 2015-05-27:
The SD card that's always in my printer includes an __auto0.g__ file to play a couple of beeps and print a message on the screen. There's no option for the file to be renamed before execution, but that's an interesting idea. For example, a file named __once0.g__ might be renamed to __once0.g~__ before being automatically processed.

whosawhatsis, 2015-05-27:
I do all of my printing from an auto0.g file. I don't need an LCD and only plug in a host for maintenance functions. I like the idea of a file that will be renamed before running to prevent accidental re-running (for instance, in case there's a power failure, so that the print doesn't restart with the previous one still on the platform when the power comes back).

It would be nice to have the option to use a special comment at the beginning of the file to choose the new name of the file, so that it would be identifiable on the card after it's been renamed.

daid, 2015-05-27:
A simple added "rename" command could be used for that which you put at the end of your AUTO0.g ?

Some people are using the AUTO0.G for setting up their printer with proper settings/auto leveling. So then you don't want the renaming.

kefir-, 2015-05-27:
I guess I should use the source next time. :) I see the point of AUTO0.g to configure the printer, and you don't want to rename that, But I like the suggestions, either a comment in that file that causes a rename, or a ONCE0.g that would be renamed. In either case, the file should be renamed to something else as the first step.

A third alternative could be something like ONCE0.cmd that simply contains the filename to execute, and then ONCE0.cmd could be deleted after it was read. That way we could auto-print any file on an SD-card without losing the file name.

I work in a multi-user setup, and if I put a full print in AUTO0.g on the SD-card, it's inevitable that someone would unknowingly print my part next time they power on the printer.

whosawhatsis, 2015-05-27:
I do run full (long) prints directly from auto0.g, but I've long thought that what I should really be doing is using a one-line auto0.g file that uses M23 to tell the printer to open another file. An auto0.g (or auto1.g, if you're already using auto0.g for something else) file like this could safely be deleted (M30) after the other file starts printing (you could put a line in your start code that deletes it once the handoff is made). The reason I have not done this yet is because it's much more difficult than just duplicating the file and renaming it. It's particularly difficult on the Mac, as I haven't found any way to get the OS to reveal the short name of a file. Only the long names are displayed, and M23 requires the short name, so my attempts to write a script to achieve this have failed. You can usually guess the short name based on the long name, but if there were multiple files starting with the same string an some have been deleted, there may be no way to tell.

kefir-, 2015-05-27:
Wow, I wasn't even aware of M23, M24 and M30. Nice, too bad there's no command to rename. :) It's as simple as putting this at the end of auto0.g, then? I assume it'll fail gracefully if autorun.g isn't there.
```
M23 /autorun.g
M24
M30 /autorun.g
```
Am I correct in assuming that "execution" returns to the original file after M23 and M24, so that the M30 is run here? I could also create autorun.g to point to another file again with M23/M24, so that I can structure the file names as I like, and simply consider autorun.g a configuration file. At that point I have what I need.

daid, 2015-05-27:
M23 will open the new file, but will close the old file. So the M24 is never seen.

kefir-, 2015-05-27:
Do you mean M24 does this, and M30 is never seen? From http://www.marlinfirmware.org/index.php/M23:

"M23 only selects a file for reading, but doesn't start to print it. Use M24 to begin printing."

In any case, the M23/M24/M30 wasn't ideal anyway, because the same file would be reprinted if the print was interrupted. But I can use it for now. :)

kefir-, 2015-05-27:
By the way, I guess I could hack my way around this by sticking:

```
M30 /autorun.g
```
in my default start.gcode. That would seem to do the trick like I want.

daid, 2015-05-27:
M23 will open a new file, and thus close the auto0.g file (as there can only be 1 file open at a time)
https://github.com/MarlinFirmware/Marlin/blob/Release/Marlin/Marlin_main.cpp#L1966

kefir-, 2015-05-27:
OK, so in my case SD printing has already been started, so M24 is not necessary and an M23 immediately switches file? I guess I misunderstood the webpage, perhaps it could be rephrased. Anyway, I think I've managed what I need. Thanks for all the helpful comments!

closed 2015-05-27 by kefir-

daid, 2015-05-27:
No, M23 also stops SD printing (as closing a file stops SD printing)

M32 should be used to select&start a file.

thinkyhead, 2015-05-27:
`M32 !/path/to/file.g#`

The `!` and `#` are required.

thinkyhead, 2018-11-22:
@Yahyaoui94 — You're free to implement that, but it's not really something we would add as a general feature. You should take a look at `M32` as an alternative. http://marlinfw.org/docs/gcode/M032.html

AnHardt, 2018-11-22:
Maybe in combination with AUTOSTART.