MarlinFirmware/Marlin #2168: unknown command and comm failures

ruggb, 2015-05-27:
I get the following when bd resets:

21:11:35.375 : echo:SD init fail
21:11:35.459 : echo:Unknown command: "M110 "
21:11:35.460 : echo:Unknown command: "M110 "
21:11:35.470 : FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:CC-BOT 2E EXTRUDER_COUNT:2 UUID:00000000-0000-0000-0000-000000000000

this is Arduino 1.6.4 and today's download.
Do I need to have SD card inserted for it to init?
why is it saying ""Unknown command: "M110 " ''"?

I am also getting errors RE last line number not being what it expected. When I try to print is sometimes just stops mid print and I must reset. It is like comm failed. When I change comm speed from 250000 to something else (19200 or 115200) nothing works.

Any ideas?

thinkyhead, 2015-05-27:
@ruggb M110 is getting passed through to the parser, apparently. It used to get filtered, but recent optimizations to the parser have introduced that kink. I will add a patch to suppress the error shortly.

thinkyhead, 2015-05-27:
> When I change comm speed from 250000 to something else (19200 or 115200) nothing works.

I'm not sure why that would be. If your controller board and your host software both support those speeds, they should communicate as usual once they have established a connection. How bad is the failure? Just no connection at all?

ruggb, 2015-05-27:
I get a total dead link. Sometimes it shudders before it goes completely dead. I put my laptop on it and it ran OK - so it may be an issue with my desktop. more tests needed. thx

thinkyhead, 2015-05-27:
Communication voodoo makes me shudder too.

ruggb, 2015-05-27:
Here is another clue.
With the ramps bd unmounted Repetier show the following sequence on connection:
E2 shows 183° immediately
E1 takes 2 sec to show and both show @ 129
about 2 sec later both jump to 183

With the Ramps bd mounted - no other connections made- IE therms NOT connected
E2 shows 183 immediately
E1 never shows.

With therms connected the same thing shows.

I have verified there is no issue on the Ramps bd.

So, IF it is a problem in your f/w then it has to do with setting the I/O pins
If u can confirm that this is NOT an issue, then I must have a bad mega bd.

LazyPrinter, 2015-05-28:
You have a bad Mega board. I recently replaced mine as the Z endstop pins and thermistors failed on my Mega without warning.

ruggb, 2015-05-28:
I found a link that described how to reflash the mega bd - to reportedly solve issues like that.
It might work for u too - I have not tried it yet
https://www.virtuabotix.com/how-to-cure-sudden-uno-death-syndrome/

closed 2015-05-28 by ruggb