MarlinFirmware/Marlin #2173: Splash Screen shows Marlin 1.0.0 RC on "Release" download

LeChuck71, 2015-05-27:
Hi there,

I triple checked that I downloaded the "release" and "1.0.2" branch of the firmware still I get a splash screen telling me I have Marlin 1.0.0 RC xx installed.

I am completely lost. I had the 1.0.2 release flashed but had to reflash because of a new ABL sensor position - now I cant get the latest release on my board.

Any help would be greatly appreciated. What I notice is that i  have a 

#define MOTHERBOARD "boardname"

instead of the "older"

#define MOTHERBOARD "number"

and some "limit checks" in the auto bed level section. Are those parts now included in the release? Last year I had issues because I always got the developer version from github and those statements were included in this version.

How can i make sure from the downloaded files that i have "latest release"?

Kind regards
Axel

AnHardt, 2015-05-27:
@LeChuck71 
Take a look into your Configuration.h Line 28.
If you can read:
//#define STRING_VERSION "1.0.2"
you have the latest Release version - regardless of what you can read on the splash-screen or serial console.

LeChuck71, 2015-05-27:
@AnHardt 
Thanks for the info, unfortunately line 24-28 show this code:

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "LeChuck71" // Who made the changes.

If i take a look into configuration.h here on Github I can see your line 28. The zip I downloaded and which is named Marlin-1.0.2.zip does contain my version .... strange lol

So how can I download the right version, atm I feel like never having used Github before ;)

AnHardt, 2015-05-27:
https://github.com/MarlinFirmware/Marlin/archive/Release.zip

LeChuck71, 2015-05-27:
@AnHardt 
Hey thx, that did the trick ;)

But still "strange" that the explicit 1.0.2 release ist not what it pretends to be 

Greatly appreciate your help

cheers

LeChuck71, 2015-05-28:
Sad news,

compiled the scource with the //#define STRING_VERSION "1.0.2" statement from the release.zip
Still get the 1.0.0 RC 2 Splashscreen and Pronterface tells me on connect Marlin Version 1.0.0

What is going wrong with my Arduino IDE?

Any suggestions 

Cheers
Axel

AnHardt, 2015-05-28:
@LeChuck71 
As i sayed:

>If you can read:
//#define STRING_VERSION "1.0.2"
you have the latest Release version - regardless of what you can read on the splash-screen or serial console.

'Release' simply has some bugs. Two of them are the different version numbers.

closed 2015-05-28 by AnHardt

thinkyhead, 2015-05-28:
@AnHardt Any chance we could fix the version number issue in Release and move the tag to the new commit? It would be nice for 1.0.1 and 1.0.2 to build in such a way that they show the right version.

Wackerbarth, 2015-05-28:
May I suggest that you include my pending versioning pull and fix it within that context.

thinkyhead, 2015-05-29:
@Wackerbarth I'd rather not mess with 1.0.2 except to fix this. There's no point in back-porting new features to older versions, and it messes with the history. With the release of 1.1 it would be nice to retire all previous versions and recommend that everyone upgrade.

Wackerbarth, 2015-05-29:
I don't consider this so much a "feature" as a "style" since, in the default case, it doesn't change anything except the location of the file containing the version number.

Wurstnase, 2015-05-29:
The 'problem' is there -> https://github.com/MarlinFirmware/Marlin/blob/Release/Marlin/Marlin_main.cpp#L64

AnHardt, 2015-05-29:
@thinkyhead 
For 1.0.2 i do not see real problems to patch in the right strings or to centralize like it's done now. @Wackerbarth 's method is more of a change than i'd like to do here.
For 1.0.1 i don't see a way without a further branch on github.

To remove a tag:

>git tag -d 1.0.2
git push origin :refs/tags/1.0.2

Maybe 'owner' rights are required - have never tested without. I suggest not to delete the tags but making only new ones. '1.0.2.a' for example.

Creating new tags for the "current state" of a branch is no problem on the web surface.
Else a:

>git tag -a 1.0.1 9fceb02
git push origin :refs/tags/1.0.1

will do the trick.
The 'releases' can be edited/remover/created on the web-surface. I suggest to only let them point to the new tag.

Thank you @Wurstnase. and here: https://github.com/MarlinFirmware/Marlin/blob/Release/Marlin/dogm_lcd_implementation.h#L129

thinkyhead, 2015-05-30:
@AnHardt We could go with `1.0.2-1` or `1.0.2.1` as the format to avoid confusion with (a)lpha and (b)eta. I like keeping the old tag. It is tempting to fix one or two glaring bugs in 1.0.2, but I think it is better to move forward and not look back. The 1.1 (née 1.0.3) change log on the wiki is a good place to look to see what bugs existed up to 1.0.2.

AnHardt, 2015-05-30:
Taged as `1.0.2-1`. Edited 'Relese notes' for 1.0.2.

What to do with 1.0.1? I think leaving as it is hurts less than a rebase of `Release`.

Wackerbarth, 2015-05-30:
Actually, I doubt that it will make much difference in the long run. If this project goes the way of many successful ones, you will abandon the concept of a single Release branch and, instead have a branch for each of the minor versions. That way, after 1.1 is released, you can still go back and patch 1.0.

In particular, this will become important when you start releasing 2.0 which will impact the host programs, etc.

thinkyhead, 2015-06-07:
Well, strictly speaking, there is no 1.0.0 because Marlin simply had that version number for a long time across many changes. Then, 1.0.1 was only around for a short time before 1.0.2 was tagged (as a bug-fixed release, more or less). In terms of any real usefulness, only the 1.0.2 tag marks a meaningful point in development, and 1.1 will be the first version that has been thoroughly vetted and tracked through its development.