MarlinFirmware/Marlin #2178: need a troubleshooting suggestion re heaters

ruggb, 2015-05-27:
This issue is intermittent - sometimes it seems permanent like now.
Print quit with this log:
16:02:21.613 : echo:  M301 P21.96 I1.86 D64.76 C1.00
16:02:21.613 : echo:Filament settings: Disabled
16:02:22.013 : Error:MINTEMP triggered, system stopped! Heater_ID: 1
16:02:22.013 : Error:Printer halted. kill() called!
16:02:22.033 : N1 M110 *2
16:02:22.033 : N1 M110 *2
16:02:22.033 : N2 M115 *4
16:02:22.033 : N3 M111 S6 *68
16:02:22.043 : N4 T0 *30
16:02:22.043 : N5 M20 *52
16:02:22.053 : N6 M80 *61
16:02:24.637 : N7 M105 *0
16:02:34.514 : N1 M110 *2
16:02:34.514 : N1 M110 *2
16:02:34.514 : N2 M115 *4
16:02:34.514 : N3 M105 *4
16:02:34.514 : N4 M114 *3
16:02:34.584 : N5 M111 S6 *66
16:02:34.584 : N6 T0 *28
16:02:34.584 : N7 M20 *54
16:02:34.594 : N8 M80 *51
16:02:34.694 : N9 M105 *14

Repetier shows no E0 therm. E1 is showing. I was printing on E0.
I have reset the printer, restarted Repetier and I still have no E0 therm.
I switched therm connectors, same indication. So it isn't the therms to the Ramps bd.
I disconnected E1 therm, still shows but it is showing 100.00 °C. It hasn't been on in 2 days.
Reset bd and display shows flash screen then main screen then blank screen w/i 1-2 sec.
Same results on another computer, so it isn't Repetier.
It feels like the firmware got hosed. I reloaded f/w and it is doing the same thing.
The Ramps bd seems too simple to have a defect.
any suggestions?

what is m111 doing?

AnHardt, 2015-05-27:
@ruggb 
M111 activates some debug output on the serial console.
See: https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Marlin_main.cpp#L141
and https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Marlin.h#L233

Your debug log is telling that Marlin thinks, there is no thermometer connected to the second extruder.
This can mean it's very cold, or a miss-configuration, or a defect thermistor, or defect cabling, or ... , or a bug.
Please give us some more information. What RAMPS type EFB, EEB do you use? What is #define EXTRUDERS? What are your TEMP_SENSOR_X settings? What are your MINTEMP settings?
A partial Configuration.h down to the PID Settings may help us here.

The resets tell me, that you are not using the latest 'Development' - fixed May 25, 2015.

ruggb, 2015-05-27:
this was working, with some random hiccups. It stopped a couple of times like the comm was getting interrupted. Then it stopped totally. Since it was running it obviously isn't a config issue nor is it very cold. My list of things I did above would tell me it is not a thermistor or cabling issue. The load I am using I dnld on 5/26 @ 7:51PM - so what r u reading in the resets?

So it might be a bug, but since it was working that may be a long shot unless the bug corrupted memory, which I why I am asking. But I reloaded f/w with no chg.
It is an EEB with 2 extruders and therms defined as 8 (and that is a load I got from a vendor).
Nothing else relative to therms was changed except for default temp for PLA from 185 to 210.

So it is either:
a bug corrupting memory
a defective Mega bd
a defective Ramps bd

it is not:
bad therms or cabling
bad config
old load

ruggb, 2015-05-27:
Here is another clue.
With the ramps bd unmounted Repetier show the following sequence on connection:
E2 shows 183° immediately
E1 takes 2 sec to show and both show @ 129
about 2 sec later both jump to 183

With the Ramps bd mounted - no other connections made- IE therms NOT connected
E2 shows 183 immediately
E1 never shows.

With therms connected the same thing shows.

I have verified there is no issue on the Ramps bd.

So, IF it is a problem in your f/w then it has to do with setting the I/O pins
If u can confirm that this is NOT an issue, then I must have a bad mega bd.

ruggb, 2015-05-28:
and maybe another clue
If I chg extruders to 1 and temp_1 to 0
it appears to work. nothing else is changed.
I can move all axis and heat E0.
That would suggest a problem with temp_1 but it was temp_0 that didn't show up prior to this chg.
I am past the point of confusion.

AnHardt, 2015-05-28:
@ruggb 
Had a extended session with the configuration you gave us.
My sensors (1, 5) give very low temperatures when used with table 8, but besides of that work.
I can not reproduce any of the problems you told us here.

I can reproduce the "tic, tic, tic, blank screen"-error when resetting the board with a not connected but configured sensor (detected as MINTEMP ERROR), i misinterpreted earlier as a reset. I'll have an eye on that.

ruggb, 2015-05-28:
yes. my table 8 is not the same as default. I get the same low readings using default table 8.
I sent my table 8 in another thread. Not really an issue.

well at least my issue (whatever it is) seems to be finding other bugs.

I dnld a new f/w file last night
set EEB, Extruder 2 and temp_0 & 1 to 8 -- along with mech changes which worked previously.
Repetier shows no Et1 temp and Et2 temp is at 100° which is totally false, and it will not run, which I would expect if there is a problem with T2 or its i/o or even Marlin

I went thru all combinations with Extruders = 1, temp_0 = 8 (my 8), and temp_1 = 0 and my conclusion is there is something wrong with i/o pin 15. 

pins settings--header placement==result
t0=13, t1=15--headers t1 & t2==runs - Et1 = ~26°, Et2 = 100°
t0=13, t1=15--headers t2 & t1==runs - Et1 = ~26°, no temp RISE indication --Et2 = 100° - so both therms r good
t0=15, t1=13--headers t1 & t2==nothing - Et1 not showing, Et2 = 100° - I would expect Et1 to show something. Why doesn't it?
t0=15, t1=13--headers t2 & t1==nothing - Et1 not showing, Et2 = 100°

The ??? is why is Repetier showing 100° when f/w is set for E=2 and showing the same when set for E=1? Repetier obviously should be set for E=1 but it is curious that 100 show in both cases.
What info should Marlin send if there is a problem with that i/o? Pin voltage is 4.80 for T1 and 4.73 for T2 @ room temp. (E=1) Open pin voltage is 4.98 for T1,T2 and T3.
Is there a way I can test pin 15 with a marlin config arrangement, w/o a shields bd and loading their test s/w?

AnHardt, 2015-05-28:
@ruggb 
No problem if you have a potentiometer or 2 resistors at hand.
Connect the outer pins of the potentiometer to GND and AREF and the middle to A15.
With resistors connect one to GND, one to AREF, twist together the free lags and connect them to A15.
The value is uncritical ~ 1K Ohm or more.

AnHardt, 2015-07-18:
@ruggb 
Do you still experience problems here?

ruggb, 2015-07-18:
@AnHardt 
I chgd the mega bd and had other issues so I really don't know what the problem was and have not ventured to try to figure it out so I guess u can close.
I keep getting strange things happening and it appears that if anything gets out of sync everything stops-usually resetting both repetier and the bd fixes it, but??

thx

closed 2015-07-18 by AnHardt