MarlinFirmware/Marlin #2189: Accidental presses

LazyPrinter, 2015-05-28:
I am using a full graphic LCD, and the problem has done the same on my Ramps 1.4 and Rumba. During a print, a single press of the encoder would momenterally goto the menu, then go back to the print screen. Many times I've accidentally changed the print speeds.

Been careful is a temporary fix, but not pratical.

ntoff, 2015-05-29:
Yeah latest dev version sometimes I have to press the button 5 times before the menu will show up and other times it shows up on the first press.

clefranc, 2015-05-29:
@LazyPrinter It happens too often to be ignored, perhaps an option to disable adjustment of the feedrate while in the main screen. After all, there is already a menu to access feedrate (and other option) while printing.

@ntoff  I saw this strange behavior too...

ntoff, 2015-05-29:
The feed rate doesn't adjust until a certain amount of "dead zone" has passed, at least for me anyway. I have to turn my encoder quite a few times while on that main screen before the feed rate will move.

@clefranc Do you see it happen more if you are deep within the menu system and just let it sit idle and return to the main menu on its own?

boelle, 2015-06-01:
if any of you can do code to fix this then please submit a PR....

thinkyhead, 2015-06-09:
@boelle I'll debug this pretty soon. I had it working really well at one point, but it got broken later.

shampine1, 2015-07-08:
I'm seeing this on my Panelolu2 also; it takes 10+ presses to get to the menu.  I couldn't tell if it is momentarily flashing back and forth, though.

thinkyhead, 2015-07-09:
Definitely aiming to debug this multiple-click issue tonight, once I get through reading all these notifications…

thinkyhead, 2015-07-25:
@shampine1 Testing with the latest code I didn't run into any points where clicks were giving me issues, but I haven't run through a full session, homing and printing, yet. I'll continue to watch for debug output on my next full test and if ignored clicks start to occur again I should have a better idea where the problem lies.

shampine1, 2015-07-25:
Scott,
That’s the post 2513 version?  I’ll try it tonight!  Thanks!!
Thanks,
Rod

Rod Shampine
Coiled Tubing Engineering Advisor
555 Industrial Blvd. MD-13
Sugar Land 77478
Office: 281-285-8479
Cell: 713-825-6531
Fax: please don’t!

shampine1, 2015-07-26:
Scott,  I don't see anything different on 2513; it still ignores button pushes.  When you press the button and the printer is NOT printing, it flashes "y." momentarily.  If you push and turn at the same time it goes into the menu most of the time.

thinkyhead, 2015-07-27:
@shampine1 If the screen is showing "y" that's rather weird. Make sure all the connectors between your board and your controller are super-tight.

shampine1, 2015-07-27:
@thinkyhead The "y." flashes for a VERY short time after the press; under 1/10 second.  Its like the first menu option gets selected immediately.  Yes, all the connectors are fully plugged in.  All three printers (running Printrboard Rev D and Panelolu 2) exhibit the same symptom, which is why I suspect firmware.  If I turn and push then it usually goes to the menu (and usually a few items down the list).

thinkyhead, 2015-07-27:
> Panelolu

Hmm, maybe this is a Panelolu thing… I will keep messing around with my controller anyway (the G3D panel).

closed 2016-05-04 by jbrazio