MarlinFirmware/Marlin #2197: Conditionals.h: No such file or directory

Hat3, 2015-05-29:
Hi

In file included from Configuration.h:820:0,
                 from Marlin.h:22,
                 from BlinkM.cpp:5:
Configuration_adv.h:4:26: fatal error: Conditionals.h: No such file or directory
 #include "Conditionals.h"
                          ^
compilation terminated.
Error compiling.

Arduino IDE 1.6.4

Wackerbarth, 2015-05-29:
Exactly which version of the source are you using?  In general, the Marlin code does not compile using the Arduino IDE 1.6.3 unless you are using the AddOns from the Development branch.

Hat3, 2015-05-29:
Hi Wackerbarth
where i can find marlin's version?
i just downloaded from github

Wackerbarth, 2015-05-29:
"downloaded" ? What did you download?
Or, preferably, you cloned the git repository.
In that case, ```git rev-parse --verify --short HEAD``` will tell me the version that you checked out.
But, after that, did you make ANY changes (like copying some example file to Configuration.h)

Also what board did you select in the IDE?

Hat3, 2015-05-29:
Download zip (https://github.com/MarlinFirmware/Marlin on the right, under clone indesktop)

The problem is the Configuration.h and Configuration_adv.h for K8200
https://github.com/MarlinFirmware/Marlin/tree/Development/Marlin/example_configurations/K8200

AnHardt, 2015-05-29:
@Hat3 
If you did not changed the branch (top-left near the green button) you got the 'Release' branch. The `Configuratin.h` from 'Development' can not work for that.

Hat3, 2015-05-29:
@AnHardt
Thank you for suggestion!

i restart from the original Configuration.h and Configuration_Adv.h and modify like original 3drag conf
i made only few change

Wackerbarth, 2015-05-29:
I think that you will find that "Release" will not build on the Arduino IDE 1.6.
You will have to revert to the Arduino IDE 1.0.5 OR Build from a Development release as I indicated previously.

AnHardt, 2015-05-29:
@Wackerbarth 
Would you please be so kind to explain what is going wrong with Arduino 1.6.3 and Marlin 1.0.2? I do it all the time now and can't find any obvious problem.

Wackerbarth, 2015-05-29:
What board are you selecting?

The MarlinSerial code hacks into the interrupt service routine based on an old Arduino 1.0 Serial model.
I have it building properly if you are using the AddOn changes that I made in the Arduino 1.5.x folder.
However, that build environment is not a “stock” build for Arduino boards under their 1.6 release.

AnHardt, 2015-05-29:
@Wackerbarth 
And what are the symptoms? Maybe i patched something and can't remember.

Hat3, 2015-05-30:
@Wackerbarth 

i selected
Board Arduino Mega or Mega 2560
Processor ATMega2560

and now compiling is done, i'll try to flash it and check the printer

CONSULitAS, 2015-05-30:
@Hat3 

If you only need a working but recent version check https://github.com/CONSULitAS/Marlin-K8200 .

Here you will find branches named `K8200_stable_20xx-xx-xx` which are tested and working on my K8200. So you can download and just compile/upload without any additional steps. This is based on the recent Development branch but working for the K8200.

If you like to contribute, you should switch to the Development branch on this respository. But be aware that there are (sometimes unknown) bugs.

AnHardt, 2015-05-30:
@Wackerbarth 
Got it. There are differences between the Windows and Linux versions of the IDE.

closed 2015-05-30 by AnHardt