MarlinFirmware/Marlin #2203: 1.0.2-1 features

ruggb, 2015-05-31:
Apparently, quickhome, as a #define, got removed, HOWEVER, it works by default. THANK YOU!!!
That was the only reason I was messing with dev versions - which I seemed to have too many issues with. So far this ver works well - only had it running a few hours though. However, it is on a new mega bd so I really don't know if the issues were the board yet. But I did start out with the dev ver on the new bd and it gave me similar but different problems.
Thx again for the update to 1.0.2

AnHardt, 2015-05-31:
@ruggb 
https://github.com/MarlinFirmware/Marlin/blob/Release/Marlin/Configuration_adv.h#L219 ???

ruggb, 2015-06-01:
don't know what u r looking at but what I dnld as 1.0.2-1 for configuration.adv.h does not have quickhome in it. I like it that way. Apparently it is default - maybe just for CoreXY.

ruggb, 2015-06-01:
My apologies......I did a search for the wrong thing.
I did not even realize or remember I set that, but apparently I did Since I marked it with //**

boelle, 2015-06-01:
will close this one as it seems solved

closed 2015-06-01 by boelle