MarlinFirmware/Marlin #2204: Marlin Google Hangout • Saturday 6 June 2015 at 21:00pm UTC • Topics TBD • Be there!

boelle, 2015-05-31:
```
Name             Zone    UTC-7 Offset               online weekdays      online weekend
@thinkyhead      UTC-7
@Wackerbarth     UTC-5
@Jnesselr        UTC-5                        
@clefranc        UTC-4   + 3 hours           
@alexborro       UTC-3   + 4 hours           
@fmalpartida     UTC+1   + 8 hours           
@AnHardt         UTC+1   + 8 hours           
@Wurstnase       UTC+1   + 8 hours
@CONSULitAS      UTC+1   + 8 hours
@paclema         UTC+1   + 8 hours          
@boelle          UTC+1   + 8 hours                  13:00 - 20:00        ~8:00 - 22:00
@emartinez167    UTC+8   +15 hours = -9      
@ntoff           UTC+10  +17 hours = -7      
```
The above UTC+ values change at summer/winter time by one hour

"Official" Clock :  http://www.timeanddate.com/time/aboututc.html

<a href="https://plus.google.com/hangouts/_/gxn3wrea5gdhoo223yimsiforia" target="_blank">Hangout</a>

link to googledoc with topics:

<a href="https://docs.google.com/document/d/1GmKjDVOKLj6f6s76ZL99LvCPOwy_PkYK57M88Q2oNdM/edit?usp=sharing" target="_blank">Topics</a>

AnHardt, 2015-06-05:
Got a last minute chance to visit http://makerfairehannover.com/
I'll __Make__ a Marlin clinic.
[image]
[image]

If i can join the hangout depends on finding a reasonable WLAN.
If one wants to talk with me, i'll be in the hangout today Fr. from 18:00 - 22:00  UTC and if i can't join the normal hangout on Mo. 18:00 - 22:00  UTC.

See you

CONSULitAS, 2015-06-06:
@AnHardt 
GREAT! I am on the way to the maker faire too! Just sitting in the train with my son. See you there.

boelle, 2015-06-06:
We are so lonely in the hangout :-P

Wackerbarth, 2015-06-09:
Scheduling conflict on that date. Unfortunately, I won't be attending.

AnHardt, 2015-06-09:
50th birthday of my wife - big party - impossible for me.
How about a week before, or later?

fmalpartida, 2015-06-09:
How about if we move it one week?

boelle, 2015-06-10:
i can one week after, the week before i'm in london and the weekend planned
i have just landed

boelle, 2015-06-12:
i have moved the date one week ahead...

gralco, 2015-06-20:
If you guys don't have any topics for the discussion I would like to cover the https://github.com/MarlinFirmware/Marlin/pull/2175 pull request I've submitted. Specifically with respect to the https://github.com/MarlinFirmware/Marlin/issues/2125 issue that @alexrj submitted.

I find it strange to have all G-Code queued by the planner whilst only some LCD functions such as [ultralcd.cpp#L443](https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/ultralcd.cpp#L443) use proper G-Code (which are queued in the buffer) and others such as [ultralcd.cpp#L508](https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/ultralcd.cpp#L508) do not (which are performed immediately). This is somewhat understandable since an LCD controller should be able to change some things dynamically; although since some calls enter the queue, such as [M600](https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/ultralcd.cpp#L511), while others do not, such as [babystepping](https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/ultralcd.cpp#L508), it produces some undesirable inconsistency. This forces the use of my G5 XYZ code to wait in queue if it is retrieved by MSerial, even with the lack of any st_synchronize() calls.

I think that some G-Codes such as M600, M220, M221, and maybe even M104 are preferred to be performed asynchronously and skip the planner queue altogether. Maybe there should be an argument that could follow any G-Code as an argument so that lookahead can handle it as priority?

thinkyhead, 2015-06-23:
@gralco Definitely looking forward to dissecting some code on Saturday…

> I find it strange to have all G-Code queued by the planner whilst only some LCD functions such as ultralcd.cpp#L443 use proper G-Code (which are queued in the buffer) and others such as ultralcd.cpp#L508 do not (which are performed immediately)

They are both performed immediately. The first on is a function because it needs to do more than just queue a G-Code command. It also exits the sub-menu. The second one could be a function, but it's all G-Code, so it doesn't need to be.

Technically all commands are processed "immediately." `G1` means "add a move to the planner," and that happens right away, while the move itself of course takes however long it takes. About the only other command besides movement that gets queued in the planner is a change in fan speed.

gralco, 2015-06-23:
@thinkyhead
I may need to perform another test with the current Development branch but when I tested sending M600 via serial I always had to wait until it got its turn in the queue. As a matter of fact this was the case regardless of whether printing (sending G-Code) from serial or SPI, and regardless of whether I selected the menu item from an LCD or sent it via serial. I tried all four combinations IIRC. I used a bogus test G-Code file I made that moved the printer back and forth from MIN/MAX on one axis thousands of times over.

thinkyhead, 2015-06-23:
> M600 via serial I always had to wait until it got its turn in the queue

As it should be. There's no way around it.

gralco, 2015-06-23:
But what if one wants to have an asynchronous response from sending M600 like you say should happen with the LCD selection?

Roxy-3DPrintBoard, 2015-06-23:
> I think that some G-Codes such as M600....

I don't think it makes sense to 'reschedule' the M600.   Where ever the user puts that in the file, that is where it should happen.  M600's pretty much only happen because somebody edits the GCode file.  The firmware should not be second guessing the user.

gralco, 2015-06-23:
@Roxy-3DPrintBoard 
I totally agree; but what about in the circumstance where one uses a microswitch to trigger filament change mode? e.g using [enquecommand_P(PSTR("M600"));   or   enquecommand(PSTR("M600"));](https://github.com/3DP-Unlimited/3DP-Marlin/blob/3DP1000X2/Marlin/Marlin_main.cpp#L1433) (albeit this is a outdated code base)

Also if one were to use the G5 code I produced (via serial) they would want an immediate response just like using babystepping from an LCD controller.

Roxy-3DPrintBoard, 2015-06-23:
Perhaps this should be moved to a separate thread dedicated to the subject (instead of hijacking this one)!   But in the case of a micro switch that triggers a filament change there would still need to be some filament left for use.   I think I would argue nothing should move either way past a M600.  However a M600 happens, it acts like that barrier you put down between customers at the grocery store check out lane.   It isn't clear which item will get picked up first on either side of the barrier, but nothing moves from one side to the other side.

boelle, 2015-06-23:
yep, move to other thread.

this one is about the hangout itself only

boelle, 2015-06-23:
if you want something added as topic for the hangout there is a link for a googledoc with topics... everyone should be able to add comments even without login

closed 2015-06-26 by boelle