MarlinFirmware/Marlin #2213: People cant request accounts at http://www.marlinfirmware.org/

boelle, 2015-06-01:
last in this issue they say they cant request accounts: https://github.com/MarlinFirmware/Marlin/issues/2090

@scotty1024 any idea what is going on?

ps. password reset does not work either

thinkyhead, 2015-06-10:
I have a web hosting account and would be happy to move the site there if no one else has time to manage it.

scotty1024, 2015-06-10:
The only two people that have requested accounts are yourself and Scott.

Scotty

boelle, 2015-06-10:
ok, there was one more that tried, hence the reason that i made this one

just cant rember who, but it was a person wanting to do documentation

epatel, 2015-06-10:
I have tried to apply for an account. Never got any response.

clefranc, 2015-06-10:
I've just tried it, so far no response...

closed 2015-06-29 by boelle