MarlinFirmware/Marlin #2214: Testers NEEDED - With touch probe

boelle, 2015-06-01:
We are in the process of getting the issues list down to 0, or as close we can.

For this we need a lot of people, with as much different machine setups as we can, that are willing to load the latest development and report back any faults, the goal is not to do features but just to test for faults.

Testers so far:

@MagoKimbra -  one Prusa I3 with Ramps 1.4 and 4 extruder and autobed level with servo & one Kossel mini with ramps 1.4.

@a4jp-com - Scoovo X9H

@darkjavi - PrusaI3 with megatronics V2 & coreXY printer with ramps1.4

@fmalpartida - H-Bot mechanics (with auto level) and  AT90USB1286 controller

@gregrebholz  - Rumba on RigidBot

@avluis - MakerFarm Prusa i3v 12" - RAMBo w/heated bed & dual extrusion (Dual Hexagon set-up for > 300C), ABL (servo+micro switch), LCD + SD (RepRapDiscount Full Graphic Display).

@vandarin - Phoenix with RAMBo, 1 extruder, heated bed, switch touch probe

@Grogyan - RAMPS 1.2 derivative with on board MAX6675 & test bed electronics is RAMPS 1.4 with a RepRap discount full graphic display, Max6675 

@thawkins - Scratch built machine with autoleveling inductive probe, ramps, 330x200 bed, based on zentoolworks cnc chassis, dual jhead hotend, qubd thermistors. Ballscrews. Reprap discount Full graphic display. Heatbed. & Scratch built machine with autoleveling servo and microswitch, ramps, 210x210 bed, jonbot, printed chassis. Single ubis hotend, qubd thermistors. Gt2 belts. Reprap discount 20x4 display. Heatbed.

@arno-millenaar - Rumba with external DM422C drivers (will add the extra delay in the stepper code)
LCD + SD (RepRapDiscount Smart Controller)
Bed Auto-Leveling
Bulldog XL Extruders with E3D v6 hotends (not sure this is relevant :) ) setup for > 300C
Bed 500W with temperature reach of 185C (or more not tested)
Thermistor for Bed and Thermocouples (AD597) for the hotends

@Lukelectro - I have a Mendel90 (Melzi electronics), gt2 belts and modified (low power, run a 24V hotbed on 12V) hotbed, otherwise standard)

@pacaj2am - Rebelix with Ramps 1.4 and Heated Bet.

@JoshyuNL - MendelMax 1.5 custom, Rambo 1.3, Heated bed + dual extruder, Filament Diameter Sensor

@alhirzel - modified MTU MOST delta printer on Sanguinololu 1.3a

@thinkyhead - Prusa i3 with Mega2560, RAMPS 1.4, Gadgets3D (HD44780-based) 20x4 LCD. &
Custom Deltabot with a Brainwave board.

@Booli CreatrHS

@CONSULitAS Vellemann K8200

@PxT Printrbot Simple Metal, Revision D board

@ProjectEnder Rostock, RAMPS 1.4, 1 Extruder, J-Head HotEnd, Nichrome Wire Heated Bed, RepRapDiscount Full Graphic Display Smart, and Filament Diameter Sensor

@clefranc Prusa i3 EiNSTeiN Variant, RAMPS 1.4, Dual Extruder, J-Head Hotend, Heated Bed, RRD Fan Extender, Filament Diameter Sensor, RepRapDiscount Smart LCD Controller or RepRapDiscount Full Graphic Smart LCD Controller

Boots Industries BI V2.5 (Delta), RUMBA, Dual Bowden Extruder, Pico Hotend, 500°C Thermistor, Heated Bed, RepRapDiscount Full Graphic Smart LCD Controller

@boelle Mendel90 Sturdy version. OMC Board w/DSM Drivers

@Natealus Modified Rostock Max V2, 280 mm circular diameter X 360mm height, E3D Kraken 4 nozzle hotend, M3 screw-in thermistors, water pump for Kraken, 4 Airtripper V3 BSP Extruders slightly modified, Onyx 120C Heated Bed, borosilicate glass print surface, Azteeg X3 Pro Printer Controller, servo z-probe, and Viki 2 Graphical LCD.

@Sniffle 12" Makerfarm Prusa i3V, Rambo Electronics, bed auto-leveling with servo and switch, RepRap Discount full graphics Display, E3DV6 Dual extruder

Test object: http://www.thingiverse.com/thing:525888

boelle, 2015-06-01:
@fmalpartida, @avluis, @vandarin, @thawkins, @arno-millenaar,  @Natealus , @Sniffle are candidates to test out if auto leveling is working

boelle, 2015-06-01:
anyways... recreated this to print to the top....

how are the lavest DEV branch working? is autolevel working as it should?

fmalpartida, 2015-06-01:
Apart from cura not playing ball, dev branch is almost ready for promotion.

ntoff, 2015-06-01:
I don't have any probes or anything but I do have a printer that I built specifically for fiddling with that uses a mixture of A4988 drivers and DRV8825's that usually has a reasonably recent version of the dev branch on it. 

Currently printing a squirrel, will throw the test object at it once that's done.

clefranc, 2015-06-01:
@boelle I've just added a **levelling probe** to my delta, can you update my description? I also have a **RepRapDiscount Smart LCD Controller** I can switch with the full one on my delta.

@thinkyhead I'm totally new to auto bed levelling, do you think you can update the wiki? I found an older Marlin Documentation folder with still BedLeveling.md inside.

fmalpartida, 2015-06-01:
@boelle, could you update my profile with the current setup I am doing some testing:
- Controler: SAV MkI (AT90USB1286), integrated uSD card reader.
- LCD: SAV 3D LCD (Shift Register LCD panel with rotary encoder, kill and home buttons)
- Mechanics: CoreXY
- Extruders: 1
- Inductive sensor + Autobed leveling
- Prints via: SD Card
- Host interface: CURA, Octoprint

Alternative setup:
- LCD: SSD1306, rotary encoder, kill and home pins
- Mechanics: i3
- Extruders: 1
- Hotbed: 1

PxT, 2015-06-02:
With a Printrbot Metal Simple (inductive probe):
I compiled leaving all the settings at default other than what was needed for my board, sensors, probe points & endstop directions.

M851 is at Z-0.7, that offset works perfectly with my Printrbot-fork firmware.  With the current dev branch, the printer probes all points as expected but then drops the nozzle a little more just as the print starts, driving it into the bed.  Homing by use of a software button works fine. My start gcode is:
    `G28 X0 Y0 Z0`
    `G29`

Let me know if you need any more info on my setup.

chris-bo, 2015-06-02:
3 point leveling with sled mounted probe works fine on my i3.
Electronics are Sanguinololu 1.3a with ATMega 1284p running at 20Mhz and Full Graphics LCD.
Grid leveling does not fit into progmem. If necessary I could test it without LCD.

I tested the current Marlin 35af5ef.

boelle, 2015-06-02:
3 point is enough

if the bed is bent etc the grid method could do some tricks.... but really
then i would rather get a flat bed

geekgarage, 2015-06-02:
I got a Prusa i3 frame with RUMBA board (dip-switch version) with inductive sensor for auto level, dual Z-axis step drivers and reprap discount full graphics, e3D v6 hotend with direct drive  and there is a few questions I can't find an answer to, as this is my first printer and i got it used prebuilt 2 days ago:

What does this do?
#define ENDSTOPPULLUP_ZPROBE 
Can't really see any change in my printers behavior

Also in advanced config i enable
#define Z_DUAL_STEPPER_DRIVERS
since i got my Z step motors running on their own driver as a single driver don't have enough power to run two motors (this is tested)
in the code i see it asks for Z_DUAL_ENDSTOPS that i don't have, and i get a compile error if i comment it out (that do make sense tho after looking at the code)
But should i just ignore this or should i install two endstops as i plan on adding Z max end stops?

I don't have a Z-min end stop switch, i'm using the inductive sensor for the Z-axis stopping.
and i only have X and Y min end stop switches tho I will be upgrading shortly with XYZ Max end stop switches hence the Z endstop question

Here is my configs
Configuration.h
http://pastebin.com/uLhsW89M

Configurations_adv.h
http://pastebin.com/vwikWtBk

Sorry in advance if this is a bit noobish and shouldn't be asked here, but i'm stuck for now trying to get the last bit to work. If you have a better site to ask these questions please refer me.

fmalpartida, 2015-06-02:
@boelle @thinkyhead hi folks, tests on the development branch look good and in shape. I think it is a good moment to move the code to the integration branch and get it ready for testing and eventual release.

fmalpartida, 2015-06-02:
What I then suggest is to freeze the integration branch for new features and only fix in the integration branch defects that the testers report on it that make it inviable for release.

Wackerbarth, 2015-06-03:
WAIT, I think that you are rushing things! There are still some pull requests that affect the configuration files and features for the delta printers that need to be integrated.

thawkins, 2015-06-03:
Speaking of which, what is the best practice for stopping local changes to
Configure.h from leaking into pull requests. How do folks normaly ensure
that is not happening.

boelle, 2015-06-03:
i noramlly make a copy of the folder that is used to sync to github.... and
then place the folder somewhere else where its easy to get at but far from
the other folder

ie on windows the desktop is my choice

epatel, 2015-06-03:
@thawkins I use https://www.sourcetreeapp.com that gives a good overview of what goes into a commit. It also makes it very easy to commit single lines. And is easy to diff different branches (i.e. I use a local branch for the private dev before merging into anything that goes to github). But also, trying to be disciplined. Like carefully look over the Pull Request carefully before pressing the button on github.

fmalpartida, 2015-06-03:
OK, how about up issuing the integ branch and continue with the dev branch. That way people can start testing it. In the mean time continue with dev to troubleshoot the delta's auto bed leveling.

boelle, 2015-06-03:
is auto leveling only an issue for Delta's now?

i thought there where trouble also with normal printers

i cant be on pc since we will have electricians shortly and they will cut
power to everything....

Sniffle, 2015-06-03:
I was of the same thought... I thought the cartesian leveli g was still
messed up or did something change suddenly?

BuzzBumbleBee, 2015-06-03:
@boelle ill check current dev release and auto bed level on my set up tonight (its a 3DR with FSR auto level probe)

boelle, 2015-06-03:
ok, i got a electronics board from @fmalpartida, but not quite there yet.
need to connect endstops and servo

Wackerbarth, 2015-06-03:
@epatel -- "But also, trying to be disciplined. Like carefully look over the Pull Request carefully before pressing the button on github."

I am in total agreement.  Further, I feel that EVERYONE should submit ALL changes in the form of a PR and have someone else actually perform the merge (at least two sets of eyes)

BuzzBumbleBee, 2015-06-03:
@boelle @Wackerbarth depending on the contributors you could use a code review system like gerrit ;)

thawkins, 2015-06-04:
@epatel nice app, but not available for linux...... dont have any win or mac machines....

thinkyhead, 2015-06-10:
> I'm totally new to auto bed levelling, do you think you can update the wiki?

@clefranc Sorry I've been doing a big move to a new city, so just getting back in the loop after many days away. I will do a basic overview page about bed leveling soon. Lots to do in that wiki.

clefranc, 2015-06-11:
@thinkyhead Thanks, no rush, I'm 2 weeks away of printer stuff.

shampine1, 2015-06-30:
I'm interested in testing!  I have access to three printers, all currently running the T3P3 branch on https://github.com/T3P3/Marlin  
I'm having a lot of problems with it, so I'm motivated to help!
All three printers are running Printrboard rev D with Panelolu2
Emaker Huxley
Solidoodle 2
Solidoodle 3
I don't have any hardware for auto leveling.

Please let me know how I can help!

thinkyhead, 2015-07-04:
@shampine1 You can test the Development branch at any time, then just report any issues you have with the latest code. We'll work with you (at our convenience) to help resolve them for you and for the community as a whole.

Wackerbarth, 2015-07-04:
@shampine1 - But, please, in your reports, if at all possible, reference the last commit that is included in the version of the code that you are testing. "Development" is a moving target and we need to be able to know which snapshot of it you are using.

shampine1, 2015-07-06:
I'm compiling using Arduino 1.0.6 and #1938 on the Development branch.  It compiles so far!

It looks like we should have a comment in Configuration.h is the Panelou area to remind people to install LiquidTWI2.h like Panucatt gets.  This note is currently hidden in conditionals.h.

I think it would be a good idea to modify Pins.h in the printrboard area.  If you have SD support, you have to move the Y endstop to the "Estop" plug; the Y pins are involved in SD communication.
Perhaps:
#ifdef SDSUPPORT
   #define Y_STOP_PIN         37   // {normally 8, must be 37 for SD card support}
#else 
  #define Y_STOP_PIN         8  // 
#endif

I'll upload tonight!

shampine1, 2015-07-08:
I'm compiling using Arduino 1.0.6 and #1938 on the Development branch.
BTW, thanks for putting the custom name in Configuration!
I've adjusted the settings to match my Solidoodle 2
x, y, z, sd and time all show --- on startup
The display shows numbers after homing
Takes about 4 button pushes the first time before anything happens
SD and time show something after you start printing
Later it has taken about 10 button pushes; not a fixed number.
The therrmal behavior seems different
I  tripped thermal runaway once on heatup
The extruder, bed, and fan on lights don't work when you are in manual control, but they do work in SD printing.
Marvin's feet have this weird cyclic stepping thing on the outlines; not smooth at all!  I don't know if this is an artifact of some acceleration setting or not.  I didn't print Marvin on my previous version of Marlin.  The perimeter is a little more smooth.  I can feel the printer vibrating slightly when this is going on.  Linear infill doesn't have this jerky thing.
On the good side, I was pleased to see the menus have improved!
I'll let you know what Marvin looks like in the morning.

shampine1, 2015-07-08:
Marvin failed.  However, I suspect that it is an extruder issue, possibly brought about by the discontinuous motion problem.  I'm printing something less oval as a test right now.

thinkyhead, 2015-07-09:
> Takes about 4 button pushes the first time before anything happens

Looking forward to fixing that soon…

shampine1, 2015-07-09:
Hot end.
I tried Marvin with the acceleration turned down to 1000 (from 3000) and Jerk turned down to 10 (from 20)  That helped quite a bit, but I still have the infamous E3D heat soak extruder problem.  Groan.  I think those settings are too high for most printers.  I would change the default to either 1000 or 1500 for X and Y.
Anything I can do to help?

thinkyhead, 2015-07-10:
> infamous E3D heat soak extruder problem

Is that due to the heat taking a long time to migrate from the heating element to the thermistor?

shampine1, 2015-07-10:
Scott,
No, the E3D hotend uses a large round heat sink with a fan above the heater section.  There is a think stainless steel heat break between them.  If the heat sink gets above the glass transition temperature for PLA, the PLA filament starts changing shape (swelling up) and sticks in the heat break and/or heat sink.  In my case, this is due to running the printer in a VERY hot environment (my un-airconditioned shop in Houston) and a partially blocked air exit.  Now that I finally understand what's going on, I can fix it!

shampine1, 2015-07-10:
I'm compiling using Arduino 1.0.6 and #1938 on the Development branch.
My re-meshed Marvin printed last night and looks very nice!

shampine1, 2015-07-25:
I had a thermal runaway shutdown last night on my Solidoodle 3 running 2474.  I'll load the latest and re-try

thawkins, 2015-07-25:
Has anybody tested the dual thermistor mode, where it uses an extra
thermister to spot runaways.

The only time i have lost an extruder was a jhead where the thermistor
detatched from the heatblock, so it was feeding back a low temp reading. It
melted down the PEEK barrel and the nozzel assembly now at about 350c
dropped into the print.

I think having the second thermistor in would have stopped that.

shampine1, 2015-07-25:
After looking closely, it actually did what it was supposed to do!  I found out that my extruder thermistor had gotten unplugged!

AnHardt, 2015-07-25:
@thawkins 
The last time i tested (Mai) *dual thermistor mode* worked fine. The code is so simple - there can't be much errors in. 
Just use the first thermistor as usual,
Test if the second thermistor is closer than MAX_REDUNDANT_TEMP_SENSOR_DIFF to the first.
If not throw an error and kill the printer.

shampine1, 2015-07-26:
Has anyone else noticed that Marlin runs REALLY rough when printing short segments?  It looks like its stopping between segments rather than following its maximum acceleration/jerk.

Nandox7, 2015-07-26:
Yes I believe I've seen that happening.
I have a STL that was created from a CAD file that was full of splines, despite the fact that most of the lines are almost straight when printing the movement isn't smooth and the printer kinda stutters.
I suspected it could be that and wanted to make a sample part to test it.

giliammc, 2015-08-24:
How do I sign up for testing

thawkins, 2015-08-25:
you just did it, grab the code, compile and upload to your bot and report
any bugs or strangeness you find

giliammc, 2015-08-25:
OK thanks

thinkyhead, 2015-08-26:
@giliammc Tell us about your hardware, electronics, and any other special features of your machine, and I'll post those details at the top of this issue.

Nandox7, 2015-08-26:
If you guys need more.

Prusa i3 Hephestos
Ramps 1.4
Single Extruder
Auto-bed level with Capacitative sensor
Two LCD's to test:
---
REPRAP_DISCOUNT_SMART_CONTROLLER
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
---

giliammc, 2015-08-30:
@boelle @thinkyhead  i have marlin dev working perfectly on a modified printrbot simple 2014 with a printrboard rev D and a ABL servo

thinkyhead, 2015-09-01:
@giliammc "Perfectly" is a word I both love to hear and fear. But I will choose to take the positive part!

thawkins, 2015-09-01:
What tools did you use to compile and upload to the printrboad, did you use
platformio?

shampine1, 2015-09-01:
I use the DFU bootloader and FLIP.  Here's my crib sheet for DFU-fu:

1.You will need a USBtinyISP or similar ICSP   http://www.ladyada.net/make/usbtinyisp/  Mine is avrisp2,not USBtinyISP.
2.Download WinAVR     http://sourceforge.net/projects/winavr/
3.Install AVR dev tools     http://www.ladyada.net/learn/avr/setup-win.html
4.Obtain a compiled DFUbootloader   http://blog.lincomatic.com/?p=548
5.Installing A Bootloader reference: http://reprap.org/wiki/Printrboard
6.Remove the BOOT jumper from the Printrboard. Press the Reset button
7.Connect the 6 pin programming cable to the Printrboard’s ICSP header. 
**** Red wire is next to the SD card slot!!!****
8.Connect your programmer’s USB cable.
9.Run the following avrdude commands,
  avrdude -c usbtiny -p at90usb1286 -U lfuse:w:0xDE:m -U hfuse:w:0x9B:m -U efuse:w:0xF0:m
  avrdude -c usbtiny -p at90usb1286 -U flash:w:BootloaderCDC.hex:i
10.Replace jumper on the Printrboard. Press Reset again.

Re-flashing the printerboard with DFU:
Install the BOOT jumper (for Rev D,E & F boards)
Press and release the Reset button. The AT90USB's bootloader will appear as 

a new USB device the first time you boot into the bootloader. Allow Windows 

to install the USB driver and note the new COM port number.

DFU BOOTLOADER
Open Atmel's FLIP software.
Select the target device: AT90USB1286. Select Communication medium as USB. 

Click Open.
Compile firmware within Arduino. You will need to copy and paste the 

compiled .HEX file into the HID application directory. 
The HEX file is saved in the temp folder, C:\users\{currentuser}\appdata

\temp\build1234567890\Firmware_Name_Here.cpp.hex
Open compiled HEX file within the FLIP software.
Make sure the Erase, Blank Check, Program, and Verify checkboxes are 

checked.
Click the Run button
Remove the BOOT jumper on the board (for Rev D, E & F boards)
Hit the reset button

thawkins, 2015-09-01:
I would be interested if you can make this work under platformio, i
currently dont have a windows machine.

See, platformio.org

There are a set of platformio config files in the root of the marlindev
folder. And there is already a printrboard target in the platformio target.
It would automate most of what you have outlined here, for installing the
toolchains. You do however need to add python support to your machine.

Im trying to bring it up under linux this weekend.

thawkins, 2015-09-01:
Ps: i thought that printrboards already used dfu protocol when in
programming mode. Certainly the ones from printrbot do, im not sure what
some of the clones like geeetech ones have as bootloader.

giliammc, 2015-09-01:
i use linux. so what i did is install teensyduino and compile marlin for
teensy ++2.0 and then use dfu-programmer to flash the hex file. i have herd
that you can flash the lufa cdc boot loader and upload code directly to the
printrboard with the arduino ide i haven't had any success with that.

giliammc, 2015-09-01:
> The CDC/arduino process sucks; that's why I went to the trouble to
> re-flash the bootloader on my old boards to DFU. the FLIP process is quick
> and dead reliable, but irritating to dig out the .hex file.
>
> Yes, the current printrboards from the mothership have DFU. Clones seem to
> have CDC on them. I really wish printerbot would add some dc power pins for
> fans and lights!

What do you want the lights for? Have you tried useing the expansion pins

shampine1, 2015-09-01:
The panelolu2 cable occupies those pins.  The Solidoodles have LED lights inside the case; quite nice.  I have rigged up cables to power them.  The Solidoodle version of printrboard (old, old, old) had pins for the lights.

giliammc, 2015-09-01:
Are you using both of the expansion headers

shampine1, 2015-09-01:
Yes, the Panelolu2 adapter mates with all of the pins.

giliammc, 2015-09-01:
If you are not using the estop pin that might work

[6 more replies]

closed 2016-04-23 by jbrazio