MarlinFirmware/Marlin #2215: Coders still needed

boelle, 2015-06-01:
We have many things to solve here, and only 2-3 people to do the hard work. So if you are able to contribute towards getting the last handfull of bugs fixed let us know

BuzzBumbleBee, 2015-06-01:
@boelle 
I'm new around here but i can try and help out with any delta related bugs (My setup is a 3DR (modified for belts) and FSR auto bed level) Just gotta put the finishing touches to my full delta trimming branch first.

dantman, 2015-06-05:
@boelle A list of the important bugs here might be helpful. I took a look through the bug list and aside from the stuff that is on hold most of the actual bugs are either in special configurations (auto tramming, etc..) lots of us can't test or require one to understand deep and complex parts of the code to fix.

boelle, 2015-06-05:
https://github.com/MarlinFirmware/Marlin/milestones

bug fix round 6 - 8

boelle, 2015-06-05:
take what you can.... every bug fixed counts :-D

thinkyhead, 2015-06-10:
Watch out for [Brooks' law](http://en.wikipedia.org/wiki/Brooks%27s_law).

closed 2015-06-29 by boelle