MarlinFirmware/Marlin #2217: Serial Communication Problem With Octoprint

msutas, 2015-06-01:
We are using several machines some of them loaded with Marlin dated May 2nd, 2015 and one loaded with Marlin dated April 13th, 2015.

The one with April 13 dated Marlin works flawlessly with Octoprint. However, the ones with May 2 dated Marlin have serial communication problem.

The boards used are Arduino Mega and Ramps 1.4 on all machines.

I couldn't figure out what has changed in this period but I believe a change on Marlin created this issue.

Can anyone figure out which change may had an effect on this issue?

The problem on communication screen is as follows:

Send: N2386 G1 X108.266 Y164.050 E150.21252*82
Recv: ok
Send: N2387 G1 X108.749 Y164.222 E150.22088*90
Recv: ok
Send: N2388 G1 X108.953 Y164.430 E150.22563*85
Recv: ok
Send: N2389 G1 X109.241 Y164.520 E150.23054*93
Recv: ok
Send: N2390 G1 X109.445 Y164.627 E150.23430*85
Recv: ok
Send: N2391 G1 X109.681 Y164.593 E150.23818*84
Recv: ok
Send: N2392 G1 X110.226 Y164.617 E150.24707*95
Recv: ok
Send: N2393 G1 X110.419 Y164.610 E150.25022*82
Recv: Error:Line Number out of sequence. Expected: 2390 Got: 239
Changing monitoring state from 'Printing' to 'Error: Line Number out of sequence. Expected: 2390 Got: 239
'
Recv: Resend: 2390
Recv: ok
Recv: Error:No Line Number with checksum, Last Line: 2389
Recv: Error:Line Number out of sequence. Expected: 2390 Got: 2393
Recv: Resend: 2390
Recv: ok
Recv: echo:enqueueing "M84"
Recv: ok

nophead, 2015-06-01:
Looks like there are problems on both side. Marlin OKed line 2390 but later complained the 0 was missing and seemed like it split the line there and started parsing the end of the line with the checksum. The line numbers and checksums should be checked before OK is sent. It also OKed 2391 and 2392 but hadn't seen a correct 2390.

The resend should have been able to recover but Octoprint gives up when it sees the word "Error" from the firmware. @foosel insists this is the correct host behaviour. In which case Marlin should not give an error message for things that are not fatal.

foosel, 2015-06-01:
No, OctoPrint gives up when it sees the word "Error" from the firmware and the rest of the line does not match any known firmware error messages which indicate communication errors.

Marlin should not change error messages that are known to indicate recoverable communication errors. OctoPrint knows that "Line Number is not Last Line Number+1" indicates such a recoverable communication error (and will be followed by a resend request and hence is harmless). OctoPrint does not know that this is also the case for "Line Number out of sequence" because this was only introduced in d43cc2dd5f52c9b71b486e7778d7fb44ea67fad6 and for that I immediately opened #2014 when I saw it (completely by chance btw). It has since been changed back through PR #2015 so with a later Marlin version that also includes that stuff should work again.

foosel, 2015-06-01:
"Close and comment" is not "Comment", sorry for that

nophead, 2015-06-01:
>OctoPrint gives up when it sees the word "Error" from the firmware and the rest of the line does not match any known firmware error messages which indicate communication errors.

That is a very fragile protocol. Virtually every change to Marlin now breaks something for somebody, not to mention every change to the Ardunio IDE.

foosel, 2015-06-01:
> That is a very fragile protocol.

It is not a protocol at all, it's a nightmare. That's what I'm complaining about all the time.

foosel, 2015-06-01:
Also, let me rephrase that: OctoPrint gives up when it sees a line starting with `!!` or `Error:` and the rest of the line does not match any known firmware error messages which indicate communication errors.

If you have a better idea how to do that and which is backwards compatible to all the printers that are already out there and also will not just blindly ignore actual fatal errors, I'm all ears. I don't, since we don't have a protocol.

msutas, 2015-06-01:
Update:

I have switched back the firmware of one problematic printer to April 13th firmware. Now it works fine.

I will check once again with a long print and give an update also.

On one of the printer I will also check the latest development version of Marlin, test and give an update.

thinkyhead, 2015-06-10:
Hey everyone, I found a home, so I'm back online. How's it working today? A couple of bugs in the pre-scanner have been patched in the last couple of days, so it's more forgiving of missing spaces.

querry43, 2015-06-20:
Are there known working versions?  I'm losing about 1 in 4 prints to this communication error.

fmalpartida, 2015-06-20:
I'm all good printing from Octoprint and the latest development branch.

nophead, 2015-06-20:
I think comms errors are hardware / serial chip firmware bugs in the Arduino mega. Other boards don't get any so any bugs in the Marlin / host retry mechanism don't show up.

closed 2015-10-13 by AnHardt