MarlinFirmware/Marlin #2222: Compiling hangs when enabeling Z_DUAL_STEPPER_DRIVERS

geekgarage, 2015-06-01:
I uncommented Z_DUAL_STEPPER_DRIVERS in Configuration_adv.h and Arduino IDE hangs in compile at Marlin_main.cpp

AnHardt, 2015-06-01:
Hangs or throws an error?
If hangs, you likely used Arduino 1.6.4. for Windows. I have had several of this issues and returned to 1.6.3.

The problem is a missing semicolon in line 561 of stepper.cpp. PR is on its way. #2223

closed 2015-06-01 by AnHardt

geekgarage, 2015-06-01:
It just hangs there and nothing happened. But i tried to revert back to 1.6.3 and now it throws errors at me like crazy :) Thank you, and i will download a fresh firmware and redo my settings step by step as i properly have messed up a ton by now :) Reason i will Download a new firmware is that the code in stepper.ccp does not look like the code referred to in #2223.