MarlinFirmware/Marlin #2224: CoreXY axis inversion possibly screwy

i-make-robots, 2015-06-01:
CoreXY "invert axis" affects one of the two XY motors, which reverses BOTH axies.  

to replicate:

#define COREXY
#define INVERT_X_DIR true 
#define INVERT_X_DIR true 
upload and observe behavior
#define INVERT_X_DIR false
upload and observe behavior

Wurstnase, 2015-06-01:
Right, the naming is not correct for coreXY. It should named as INVERT_A_DIR and INVERT_B_DIR. It reverses the motor and not the axis.

AnHardt, 2015-06-02:
@i-make-robots 
You cold use these if one of the motors is mounted top-down.

If you really want to change the direction of only one axis (logically) a good start could be to change the signs here: https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/planner.cpp#L532
But this will probably mess up the end-stopp logic.

Wurstnase, 2015-06-02:
You shouldn't modify anything there.

There are 3 options (2^3 possible solutions):
Invert Motor X Direction (A-Motor)
Invert Motor Y Direction (B-Motor)
Switch Motors (physically on your board or via pin-definitions)

thinkyhead, 2015-06-10:
Adding COREXY causes the X and Y axes to become retronyms, but doesn't change their behavior. We may rename them in future, but for now keeping them the same for legacy reasons.

closed 2015-06-10 by thinkyhead