MarlinFirmware/Marlin #2226: Dev branch status - 2015.6.1 - d4c6fc5d0a121d25e97836cfc71661b94f500400 [d4c6fc5]

fmalpartida, 2015-06-01:
## Test setup:
* SAV MkI
* SAV 3D LCD
* CoreXY mechanics
* Inductive sensor + Autobed leveling
* SD Card
* Host interface: CURA

## Status:
* Homing/Outbed leveling - **PASS/FAIL** (with observation)
* Thermals - **PASS**
* Movement and print - **PASS**
* SD printing - **PASS**
* LCD updates - **FAIL** (print position shown OK for X, Y, after increasing idle time, Z remains as Z ---.--)
* Host printing - **PASS** 
### Observations
**Auto bed levelling/Homing:** strange behaviour while executing predefined start code, could not reproduce from command line. Same behaviour while printing from SD card sliced with Cura.
The following start code skips autobed levelling sequence from cura:
```
G90
M82
M106 S200
G28
G29
G1 Z15.0 F{travel_speed}
G92 E0
G1 F200 E3
G92 E0
G1 F{travel_speed}
M117 Printing...
```
Console print on error:
```?Probe position out of range.```

No string, as if out_of_range_error was called with a null character.

Fixed as follows:
```
G90
M82
M106 S200
G28 X0 Y0
G29
G1 Z15.0 F{travel_speed}
G92 E0
G1 F200 E3
G92 E0
G1 F{travel_speed}
M117 Printing...
```
## Current development branch status
* Fit for integration branch - **YES**.
* Fit for release branch - **NO**.

### Defects found:
* During printing the current print coordinates for Z are not refreshed and remain as "Z---.--"
* Autobed levelling -  strange behaviour of initialisation code before starting prints, both printing from USB or SD card.

AnHardt, 2015-06-02:
@fmalpartida 
Did you get "---" before or after your "fix"?
If after the fix - all is well. The "---" is an indicator for axis_known_position. And tells you: "This axis is not homed." Exactly this happens with "G28 X0 Y0" - Z is not homed.

As far i can remember "homing all axis" was and still is required for a working G29. I'm afraid we'll never get G29 to work when only a few testers will constantly send reports based on not homed systems. (but i'm nor an expert for that)

I'll have a look on "?Probe position out of range." now.

fmalpartida, 2015-06-02:
The "---" was after the fix, X and Y are working well and as expected.

This is a bit of a workaround for the G28 problem that I was seeing. 
However, performing a G29 gets you to do an implicit home of the Z axis as you are probing 4 points in the bed where one (perhaps the first one) can be considered you actual Z home.

There are several things to consider here:
1. if for autobed leveling to work you need to have all 3 axis to work, then the check before executing autoleveling should be:
```
            // Prevent user from running a G29 without first homing in X and Y (and Z).
            if (! (axis_known_position[X_AXIS] && axis_known_position[Y_AXIS] && axis_known_position[Z_AXIS]) )
            {...
``` including the associated error.
2. if only X and Y is required, then there is a need to use one of the first probes as the "virtual home".

If going down the route of 2, then on locking the firts point as the reference home, the Z indicator should then be set to true.

In any case, something is not consistent.

fmalpartida, 2015-06-02:
Hi folks, been trying out the latest branch today and I have to say that it is working just fine with my configuration.
Perhaps the only decision is if we go for option 1 or 2 and force homing of all axis.

I think we can start thinking in promoting this branch as it is to integration.

thinkyhead, 2015-06-10:
@fmalpartida We're getting close. But there's still some oddity with the Z position after auto-leveling that needs to be worked out.

fmalpartida, 2015-06-10:
Indeed we are... I'll update this weekend and report back. I want to try out the latest changes from Richard too.

thinkyhead, 2015-06-12:
#2281 seems like an obvious one, what say ye?

fmalpartida, 2015-06-12:
I am good with the change. I will be updating tonight for testing too.

fmalpartida, 2015-06-14:
Update with test from latest development branch.

closed 2015-06-14 by fmalpartida

thinkyhead, 2015-06-15:
#2292