MarlinFirmware/Marlin #2227: Enabeling MANUAL_BED_LEVELING causes errors in compiling

geekgarage, 2015-06-01:
C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp -o C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp.o 
C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp: In function 'void _lcd_level_bed()':
C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp:2026:44: error: 'MESH_NUM_X_POINTS' was not declared in this scope
         int ix = _lcd_level_bed_position % MESH_NUM_X_POINTS,
                                            ^
C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp:2028:13: error: 'iy' was not declared in this scope
         if (iy & 1) ix = (MESH_NUM_X_POINTS - 1) - ix; // Zig zag
             ^
C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp:2029:9: error: 'mbl' was not declared in this scope
         mbl.set_z(ix, iy, current_position[Z_AXIS]);
         ^
C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp:2029:23: error: 'iy' was not declared in this scope
         mbl.set_z(ix, iy, current_position[Z_AXIS]);
                       ^
C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp:2031:58: error: 'MESH_NUM_Y_POINTS' was not declared in this scope
         if (_lcd_level_bed_position == MESH_NUM_X_POINTS*MESH_NUM_Y_POINTS) {
                                                          ^
C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp:2032:38: error: 'MESH_HOME_SEARCH_Z' was not declared in this scope
           current_position[Z_AXIS] = MESH_HOME_SEARCH_Z;
                                      ^
C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp:2039:38: error: 'MESH_HOME_SEARCH_Z' was not declared in this scope
           current_position[Z_AXIS] = MESH_HOME_SEARCH_Z;
                                      ^
C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp: In function 'void _lcd_level_bed_homing()':
C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp:2062:34: error: 'MESH_HOME_SEARCH_Z' was not declared in this scope
       current_position[Z_AXIS] = MESH_HOME_SEARCH_Z;
                                  ^
C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp:2064:34: error: 'MESH_MIN_X' was not declared in this scope
       current_position[X_AXIS] = MESH_MIN_X;
                                  ^
C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp:2065:34: error: 'MESH_MIN_Y' was not declared in this scope
       current_position[Y_AXIS] = MESH_MIN_Y;
                                  ^
C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp: In function 'void lcd_level_bed()':
C:\Users\RASMU_~1\AppData\Local\Temp\build4839067570237832792.tmp\ultralcd.cpp:2078:5: error: 'mbl' was not declared in this scope
     mbl.reset();

I see alot of "MESH" not declared and i do not have MESH_BED_LEVELING defined, i left that commented out.

It almost don't matter what LCD/keypad i try to compile with then it throws this error
The ones that do not produce this error is:
REPRAPWORLD_KEYPAD_MOVE_STEP

I haven't tested with I2C panels

geekgarage, 2015-06-01:
Issue opened on wrong firmware

AnHardt, 2015-06-01:
@geekgarage 
Lets go back to MarlinFirmware/Marlin https://github.com/MarlinFirmware/Marlin/issues/2227
I'm not an expert for the code for bed leveling. On MarlinFirmware you'll get a much larger audience.

My Development branch is some days delayed most of the time. Please do not download from AnHard/Marlin. Please use MarlinFirmware instead. If there are special branches i'd like to have tested, i'll anounce them on MarlinFirmware.

To your problem. I have the impression it could be caused by copying an older configuration into a newer version.

I can't help you without the knowledge about your branch and your configuration.
Sine i have 17 branches today is much easier to have a new try on MarlinFirmware.
We prefer to find configurations not in the bug-tracker but in a gist or pastebin. It's so difficult to paste a configuration into the bug-tracker without ruining the format (what helps us so much when reading)

Copy from https://github.com/AnHardt/Marlin/issues/4

epatel, 2015-06-02:
Looks like you enabled `MANUAL_BED_LEVELING` but not `MESH_BED_LEVELING`.
`MANUAL_BED_LEVELING` does only work with `MESH_BED_LEVELING` today.
I wrote some short instructions about this feature, that seem to have been moved here http://www.marlinfirmware.org/index.php/Mesh_Bed_Leveling

geekgarage, 2015-06-02:
ENABLE_AUTO_BED_LEVELING <-- That was the issue, i had that one enabled as i do have the sensor for it.

closed 2015-06-02 by geekgarage