MarlinFirmware/Marlin #2237: SD card printing should disable heaters when finished

atunguyd, 2015-06-03:
I have recently switched to printing from SD card and the other night I left the printer printing overnight. The next morning the print was finished and the extruder heater was off, but the bed heater was still on.

I realise that in the Gcode the heaters can be disabled when printing is completed but i think it would be prudent for marlin to switch off all heaters when it knows the printing is complete anyway as my slicer (slic3r) seemed to have defaulted to switching off the extruder only and leaving the bed on.

Looking in void CardReader::printingHasFinished() I see that there is a call to autotempShutdown() but this will only switch off the active extruder and then only if variable autotemp_enabled is set.

I suggest that here a call is made to disable_all_heaters() which appears to disable every heater regardless.

AnHardt, 2015-06-03:
@atunguyd 
Maybe M85 is what you are looking for.
If you want it to default different from 0, you could change:
https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Marlin_main.cpp#L279
to a different value (in milliseconds).
It's not a big thing to make this configurable - if wanted.

atunguyd, 2015-06-03:
Sure you could use M85, but then again that needs to be in the G code does it not, so I might as well ensure that the G code switches off all heaters.

As I said, it is just an extra failsafe to switch off everything when we know that the printing is completed (heaters and stepper motors I would say), then if the GCode did miss something we have this extra backup.

nophead, 2015-06-03:
Some people prefer to leave the bed on between prints to save warm up time. They swap the glass to remove the parts. 

I think the current situation is correct. If you want it to be off turn it off in your gcode. Failsafe doesn't come into it because the bed staying on isn't a safety issue. In normal operation the bed can be on with parts on it for many hours. A few more makes no difference except it wastes a few pennies.

atunguyd, 2015-06-03:
Fair enough, i guess I can just use my suggestion in my own firmware then.

AnHardt, 2015-06-03:
@atunguyd 
Would you pleas be so kind to read more than the first two lines of my first comment.
Think about the consequences of changing the default value.

atunguyd, 2015-06-03:
Yes mate I did read everything, my suggestion was for a change to the Marlin code base and in my opinion suggesting to change the default timeout to a different value would open a whole other can of worms (trying to agree with everyone on what this should be).

For a personal fix sure I could change the default timeout value, but that is a bit of a long way around addressing my immediate need which is for the heaters to be switched off when printing from SD is completed. For me it would be far more elegant to disable_all_heaters() from the routine printingHasFinished() then to rely on a timeout mechanism, which in future changes could be disabled upon printing completed anyway.

ntoff, 2015-06-04:
Why not just have a "Turn printer of when done" option in the LCD menu? That way it shuts down all the heaters and if your power supply permits it, turns that off too.

brainscan, 2015-06-04:
I print from sd and my heaters shutdown at the end, waits for hotend to cool down then switches off the power supply. Doesn't it depend how you end your gcode?

atunguyd, 2015-06-04:
Brainscan, thats exactly what I said, one my PC I use slic3r and I have not made any changes to the custom G-Code so ti looks like it defaults to not switching off the bed, only the extruders.
Having Marlin take care of this seems to me to be a good idea, but then thats the beauty of open source, if I want it I can add it myself.

grob6000, 2015-06-05:
Yup, you sure can. Enjoy! :)

I've had a think about this idea, and maybe not to be dismissed entirely?

* I slice with slic3r via Repetier-Host, but print with octoprint on a raspberry pi.
* The shutdown gcode is configured in slic3r.

While it works predictably, the niggle here is the same as with SD printing - once you've sliced and uploaded, the shutdown routine is fixed in that file. Then you have to remember whether the file was a shutdown one or not!

Ideally I'd like to be able to choose whether to shutdown after a print without having to reslice - it would make it much easier to reuse gcode files (at the moment I often reslice+reupload just to change the start and end behaviour). This is only a mild annoyance on a full slicing host, but when you have an additional step of  saving and uploading the g-code file, it can quickly become time-consuming and, through laziness, unpredictable.

Similar to @ntoff, I'd suggest an auto-shutdown script feature rather than enforcing the "bed heater off on finish" - run a specific script on the SD card is run when a print is terminated. The function could then be enabled/disabled with a pair of gcodes, so you can control whether it will happen from the host and/or LCD. I think that would achieve @atunguyd's goal, and maybe others would find it useful too.

On the other hand, maybe the firmware isn't the place for this: from a host-side, having selectable start and end codes could be just as good, maybe as would allowing arbitrary queuing of gcodes. Think something like: "start_cold.gcode" --> "yodahead.gcode" --> "clearbed.gcode" --> "benchy.gcode" --> "shutdown.gcode", then lunch!

clefranc, 2015-06-05:
I've just tried a small SD print, all heaters were left ON after print finished. Did I missed something?

@atunguyd I agree with your OP.

It seems to please only those that swap glass (majority or minority?) that the bed must stay on after a print?

@nophead Yes, the current situation is correct, it was programmed this way, but I think the current behavior and implementation are not. If someone want to keep its bed on after a SD print then a proper option **#define** should be created to do so. For all of us who *presumes* (logically) that when a print has finish, all heaters should shutdown, the code must be modified.

What's the point to use the **autotempShutdown** function after SD print finish, as I could argue that I can use gcode to do the job, but what if I don't. I can see a nice option like ```#define NO_AUTOTEMP_SHUTDOWN_BED``` included in the routine, for those who need that their beds heat all the time.

@ntoff Good idea.

@brainscan A 3D printer should never be left unattended, but when a print is finished (from host or SD), I can assure you that many new users will never suspect that the bed is still heating. I totally respect your comment, but why tempt the devil, just gives the option to those who have specific needs (and knowledge) to use heaters as they wish, for all of *casual* users, shutdown all heaters please.

@alexrj Can I bring you to the discussion? Why only the hotend heater is shutdown in the default **End G-code**? What was the argument at the time? Can the bed's heater shutdown command be added to the default config? I was going to suggest to add the **M0** command, but I just tried it and the Marlin behavior is not what I expected. Do you think a new command should be added to the reprap wiki to signal the printer that the print is complete? Like an **EOP** (End Of Print) that you can add at the end of all G-code generated? And why not a **SOP** too? 

Right now, when a print is complete from host, we have a nice **Heating done** left on the LCD. Hosts has enough cpu power to calculate and display some stats, even shutdown heaters if configured properly. I'm sure that situation can evolve a bit from a firmware perspective. We can't solely rely on hosts software and slicers to protect us. A 3D printer has high potential to harm its user and people around. Safety should be on top of the feature list.

@grob6000 Interesting comment, good suggestions. But I still think the firmware should be the last line of defense against slicers G-code mis-use and/or bad hosts configuration. Let *advanced* users use their printer in any way they want.

SD print terminates with some info on the LCD, but as I wrote, the latest dev still keeps all heaters ON after print completion.

Many collaborators and active members here have years of experience and expertise in 3D printing. Manually typing G-code was once (and is still in some extend) the only mean to control a 3D printer, so they must had learn to master them. I don't want to blame anybody, but if someone come here and says its printer almost killed his family, I don't want to be the one to say *Yeah, we knew that heaters can be harmful, but we thought the unenlightened users like you had to be aware of heating behaviors of Marlin...*

dantman, 2015-06-05:
I like @ntoff's suggestion. Though I think the opposite of that might be a better idea. @nophead's mentions that there are people who keep their beds warm to swap prints. But that's one group of people and only applies when they're doing multiple prints. If they're finished printing or only doing one print then even that group doesn't really have a reason for the bed to stay on.

Here's my idea (at least for when SD printing with an LCD controller, I don't know enough about the different hosts to suggest what to do with them):

- The default is for the heated bed to turn off when printing is finished.
- In the LCD menu there is a "Keep bed warm after prints" (if that's too long, shorten it)
- Using that menu item will change the behaviour to keep the bed on when printing is finished (maybe we'll even ignore the command to turn it off at the end if the user selects that option).
- When the behaviour is changed that menu item will change to "Cooldown bed after prints" which will change the behaviour back to normal and run a cooldown if a print is not in progress.
- ((If possible maybe we'll add a warning about the heated bed being left on to the LCD))
- If an advanced user doesn't like having to select the "Keep warm" option after they turn on their printer. We'll have a setting in Configuration(_adv).h the can uncomment:
```cpp
#define DEFAULT_KEEP_WARM_BED true
#define DEFAULT_KEEP_WARM_BED true
#define DEFAULT_KEEP_WARM_BED true
```

alexrj, 2015-06-05:
@clefranc, the default End G-code in Slic3r only contains `M104 S0` and not `M140 S0` because including the latter would block those printers not having a heated bed but still having the BED_PIN configured (there are lots out there). So if user has heated bed, they should add `M140 S0` to the End G-code manually. This is not nice, I know. A bunch of checkboxes should be designed :-)
(I need to catch a plane now so I'll jump into the discussion in the next days)

nophead, 2015-06-05:
I strongly believe Marlin should do what the gcode tells it to do. Also
this behaviour has been the same for many years and used by perhaps
hundreds of thousands of people. It should not be changing on a whim.

Adding yet more compile time conditional code and complication is what
makes this an almost unmaintainable unstable nightmare.

ntoff, 2015-06-05:
>I strongly believe Marlin should do what the gcode tells it to do.

If all Marlin did was whatever gcodes told it to do, there'd be no auto bed leveling, no thermal runaway protection, because how do you run those checks in gcodes? Should all that stuff should then be handled by the host, be it repetier or octoprint with Marlin just left alone to do nothing but run gcode commands?

No. Certainly not. Marlin has the ability to run in standalone mode so it NEEDS to have certain features beyond just "doing what the gcode tells it to do".

I don't see how adding a menu item to "switch off when done" could possibly hurt the hundreds of thousands of people who use Marlin. What, you think everyone who uses it is a senile old fool who gets easily confused when extra things show up in the menu?

atunguyd, 2015-06-05:
@nophead that is a terrible attitude for a developer to have. All the other reasons provided above have been valid but to reject a valid change based on  "Adding yet more compile time conditional code and complication is what makes this an almost unmaintainable unstable nightmare" is just a poor attitude in my books.

Imagine if the linux kernel team just stuck to x86 CPU's and said "we cant port this to ARM or MIPS because it make things too unmaintainable" - we would not have Android today or the multiple CE router products we have today.

dantman, 2015-06-05:
Thanks @ntoff, you put it better than I could.

Additionally, others have already explained the value of code to automatically turn the bed off and ignoring the gcode.

The additional 'compile time conditional code and complication' of my idea beyond basically amounts to one menu item and a boolean to track the state.
----
Also 'because we've always done is this way' is not a valid excuse. Not when everyone else has already explained multiple reasons why the change is warranted and why the current gcode situation is wrong.

This even goes beyond that. From the sounds of it Slic3r would like to made bed shutoff the default behaviour. But it can't because that would break printers. That alone makes this a problem that should be solved and can only be solved within the firmware.

To reiterate and expand:

Saved gcode on an SD card is not the place to indicate whether to switch a bed off at the end or not is. The former is practically immutable. The later is related to the user's intent and can change on the fly.

The portion of those hundreds of thousands of people expecting the current behaviour to stay as is is likely not as large as you might think.
- Cura seems to include gcode to turn the bed off by default. So any Cura user with the defaults won't see a difference.
- Not all of those people even have a heated bed, so those without are unaffected.
- Any user that added the command to their gcode wouldn't be affected either.

And to further reduce that a good portion of those people who would notice a change didn't explicitly chose it and aren't expecting or desiring it to stay that way.

Wackerbarth, 2015-06-05:
@atunguyd —
Although @nopphead may have a “terrible attitude”, he does have a point.
The code base that is so full of conditional compiles, particularly with their patchwork addition,
and the git repository full of merges “from here, there, and ‘yond” does create a code base that is VERY difficult to maintain.

It is not the functionality of supporting all of these behaviors, but the manner in which they are implemented.
You should look at some of the history of coding practices. This code seems to have many of the bad aspects of coding style from the early 1960’s.

But on the point of adding additional M codes for each little variation, I think that it is a bad idea.
Instead, I feel that we need to adding “hook” capabilities in a manner that allows the user to specify behavior in a custom manner.

There are two schools of thought in controlling a printer, or other CNC device. In one school, you have a simple “engine” that handles the real-time constraints of moving the parts in concert. The complexity of “where and when” is left to the “host”. In the other school, the printer is viewed as a stand-alone device. It is given the “goal” and it is up to the device to handle all of the complexities.

The present Marlin code is somewhere in between. As a result, there are always pressures to pull the design to satisfy each particular user’s “itch”.  The whole project could benefit by better implementing a top-down design rather than continuing the "patchwork of warts” development.

> On Jun 5, 2015, at 4:04 AM, atunguyd <notifications@github.com> wrote:
> @nophead <https://github.com/nophead> that is a terrible attitude for a developer to have. All the other reasons provided above have been valid but to reject a valid change based on "Adding yet more compile time conditional code and complication is what makes this an almost unmaintainable unstable nightmare" is just a poor attitude in my books.
> 
> Imagine if the linux kernel team just stuck to x86 CPU's and said "we cant port this to ARM or MIPS because it make things too unmaintainable" - we would not have Android today or the multiple CE router products we have today.
>

dantman, 2015-06-05:
@Wackerbarth
Could you give an overview of what you think a top-down design would look like for Marlin, ideally in another bug we could discuss it. It's hard for anyone to jump in and help cleanup Marlin without an idea of what it would look like.

I'm also interested to know what you think "hook" capabilities would look like.

Wackerbarth, 2015-06-06:
By top-down, I fundamentally mean "decide how the machine should behave" before you start adding the implementations of particular ways that might achieve imagined goals.  The other thing that I feel we need is a more modular coding style. The "one huge flat union of all the possible code fragments" style makes it very difficult to maintain -- particularly with respect to understanding what the code is intended to do.

By hook capabilities, I mean to add user configurable descriptions of actions to be taken in response to certain events.

Taking the present issue of printing from SD card as an example, the slicer has the capability to add a configurable prelude and postlude wrapping the actions translated from the stl description of the object to be printed.

In addition, it might be appropriate to have a similar capability in the printer itself.

Then each user could specify the behavior that they desire.

Now, I know that to do this, I need to be able to store a g-code sequence and have it activated as a part of the startup/ending sequence. But, before I go off and implement code to do that, we should first decide how it fits in the bigger picture.  For example, does the "print from SD card" need to call this code before executing the code on the SD card? Or should we expect the command to execute the code to be provided in the prelude which the slicer inserts? We should also consider whether, or not, it makes sense to have this as a more general capability. Where do we store this g-code? How does the user change it?, etc.

As I have said, we need design before heading out to implement.

eboston, 2015-06-06:
@Wackerbarth your last comment gave me an idea.  What if there were two special file names like SDPRE.G and SDPRO.G that if on the SD card, the system would send them before/after any file sent from the SD card?

Wackerbarth, 2015-06-06:
@eboston  -- That is certainly a reasonable possibility to be considered. It does address the "where is it stored?" question. But, is that scheme flexible enough to satisfy the users?

We might also consider a more generic approach making the "print from SD card" recursive and allow the user to trigger it from his intro text.

As I have said, before rushing into the writing of code, we need (1) The requirements. (2) A Design of the overall implementation. And then (3) Specific specification of the "commands" that will be implemented.
Only then (4) Code implementing the process described in the preceding steps.

dantman, 2015-06-06:
I like the idea of a pre and post gcode SD file. In fact I wish all the printer hosts also had a way to specify some pre and post gcode. This general practice we have of putting things like telling the printer to do auto tramming, reset position, etc... all in the slicer and having them embedded into every gcode file generated feels like an awful hack.

Though for this particular issue it leaves us right back where we started. In fact @grob6000 already suggested splitting out the gcode. The problem is that one shouldn't have to manually create a gcode file in order to enable what should be the default behaviour (shutting off the hotends and heated beds after printing). We need to find some way to invert that. If not an LCD menu option, making the people that want to swap glass put something in their pre/post gcode to tell the printer to stay warm for the next print.

Wackerbarth, 2015-06-06:
@dantman -- I get the feeling that everyone is trying to solve some specific issue when a more generic approach would better serve the situation. What if, in a manner that I am not currently specifying, we have a set of events. Associated with each event, we have some way to redefine the action taken.

I envision three levels of "default" for each such action.
(1) The ultimate default action (perhaps just a "nop")
possibly overridden in the program memory by
(2) A firmware-specific configuration which replaces the default action for the device
and 
(3) A user configurable replacement that, when supplied, replaces the action specified above.

dantman, 2015-06-07:
Ok, to get an idea of the scope we should come up with a list of events and actions we think may happen in those events.

Here are some of the things I see in Slic3r and/or Cura as default start/end gcode that would probably be better placed in events (firmware default, firmware config, user config).

- print start
  - Run auto tramming setup
  - Homing the axis and lifting the nozzle?
  - Squirting out a few millimetres of filament before the print (head cleaning? Cura’s defaults do this)
  - Print something on the LCD
- print end
  - Turn off extruders and heated bed. 
  - Home one or more axis (and maybe 'home' to max on the y axis for some printers)
  - Turn off steppers.

Cura’s default start.gcode also seems to set the coordinate system to metric units and absolute coordinates then turns off the fan initially (G21, G90, M82, and M107). These are directly related to the generated gcode. i.e.: They’re directly connected to what coordinate system the G0 and G1 commands Cura generated were in and the initial state of the fan which is also something controlled within the print. IMHO Cura should be generating these implicitly instead of them being part of the start.gcode. Sli3er looks like it correctly handles generating this stuff itself.

----

I don’t think we can simply do this event system as a simple list of g-code commands. The firmware built-in default for both of these events should include a number of these actions in the defaults. And it should be simultaneously possible for a higher level (the user configurable / host based config outside the firmware config) to override an event without inadvertently disabling something like cool down of all hot printer parts and to explicitly disable things like turning off the heated bed.

One idea on how to deal with that might be to define multiple named actions/behaviours. The set of g-code commands would be assigned to those. And multiple named actions/behaviours would be assigned those to events. (ie: start.home, start.print, end.cooldown, end.home, etc...) then the host can override and add individual sets of commands.

Wackerbarth, 2015-06-07:
OK, I think that we are thinking in the same direction. There are some events e.g. POWER_UP, START_PRINT and END_PRINT.  Each of these has a sequence of actions that should be performed.

Hard coded in the machine is:

On END_PRINT_EVENT, Execute EndPrintScript

We define a default script which is nothing more than calls to individual scripts for the more basic actions.
EndPrintScript:
  Execute WithdrawExtruder
  Execute InitiateCoolExtruderAtEndOfPrint
  Execute InitiateCoolBedAtEndOfPrint
  Execute RetractFilament
  Execute ParkEffector
  Execute ReleasePositioners
  Execute PowerDown

If the custom firmware/user wishes to redefine the sequence, they could do so.

However, since each of those actions is really just a call to another script, They would more likely need to override only some aspect. For example, a default might be:

InitiateCoolBedAtEndOfPrint:
  Execute InitiateCoolBed

InitiateCoolBed:
  SetTargetTemperatiureForBed 20

By adding that extra layer of indirection, we have a "hook" that the user can override without destroying the actual definition of InitiateCoolBed (which they will eventually need after the last print)

If the user wants to skip the bed cooling, they come override with

InitiateCoolBedAtEndOfPrint:
  Continue

Since storage space is VERY limited, all of these scripts can be precompiled into some kind of byte code representation.

dantman, 2015-06-07:
I'm not quite certain how configurable scripts would be implemented in Marlin.

But I'll make a related note about events. Because it's the only place that reliably knows, I think the host should be responsible for triggering the print start and print end events. For LCD+SD prints Marlink itself would be considered the host.

If for any reason we find we need to behave differently depending on whether an actual printer host supports these events or not. Then it seems that others have already designed a nice way to [negotiate protocol feature support](http://reprap.org/wiki/Protocol_Feature_Negotiation) between the host and firmware which I think we should use.

In fact I think the syntax and trick used by that to offload string processing onto the Host could potentially be useful when defining scripts/actions.

atunguyd, 2015-09-15:
Hi, I just wanted to add to this issue that I raised a while ago. Last night I did a print and went to go have dinner, I returned a few hours later to find the printer stuck in the middle of the print, all motors still and temperatures still set to the values that would be used to print. The screen on my graphic display said something along the lines of waiting for user input.

I eventually gave up and killed the print. Upon further inspection I discovered that the SD card did not have the complete gcode file on it and the gcode just stopped.

Again the printer should have gotten to the end of the file and determined that the print is over and switched off everything.

Wackerbarth, 2015-09-15:
@atunguyd -- Perhaps what would work better is an "Idle Shutdown" where the printer parks the printhead and shuts down all heaters and steppers if it does not receive any instructions within a (configurable) fixed time. (eg. 5 minutes)

atunguyd, 2015-09-15:
Yes quite possibly an idle shutdown would work - I would say any intelligence that ensures that the printer doesnt sit there braindead waiting for a command that will never come.

nophead, 2015-09-15:
Well I expect my printers to "sit there braindead waiting for a command" in between builds and manual commands. I often use the heated bed to cure JB-weld so I turn it on with OctoPrint and expect it to stay on until I turn it off. I certainly don't want it to suddenly decide to move the head, I might be working on it.

The firmware is a slave. It should never do things without being told to do so. If you send it half a gcode file it should do half the job.

Wurstnase, 2015-09-15:
I'm 100% with nophead.

Wackerbarth, 2015-09-15:
> The firmware is a slave. It should never do things without being told to do so. If you send it half a gcode file it should do half the job.
> 
I guess that all depends on how you define “the job”.  There is certainly a reasonable argument for having watchdog timers, particularly if they are associated with unattended machines.  Now, I will agree that various users will want different behaviors depending on their current use.

However, I see no harm in having the watchdog shut things down when its timer expires.
And you should not complain that the feature is present if you have the option to set that timer to your desired interval. In your case, that interval might be “Never Expire”.

Wackerbarth, 2015-09-15:
OK, let's also disconnect ALL limit switches, thermal runaway tests, etc......  After all, those aren't part of "the job" (until something unexpected happens).

nophead, 2015-09-15:
Well I do run with limit switches disabled after homing (don't most people) and don't use thermal runaway tests. I don't like nanny firmware.

Wackerbarth, 2015-09-15:
That is your choice. But there are obviously others who want a bit more "protection".  I certainly don't think that you should impose your "daredevil" attitude on everyone.

BTW, on my printers, I, also, use the sensors for alignment only. However, I do have real limit switches on my metal lathe.

ntoff, 2015-09-15:
>Well I do run with limit switches disabled after homing (don't most people) and don't use thermal runaway tests. I don't like nanny firmware.

I run with software limits disabled and min/max hardware limits permanently on. I prefer it that way.

>OK, let's also disconnect ALL limit switches, thermal runaway tests, etc...... After all, those aren't part of "the job" (until something unexpected happens).

Now you're just being dumb. Saying let's remove all the safety stuff because someone doesn't want their printer doing things without their permission is like saying ok let's remove the airbags and abs and other safety stuff from cars because someone doesn't like their headlights automatically turning on at night and off at dawn because they might want to run with them on 24/7.

It's one thing to expect a printer to shut itself down in an emergency should the heater to come loose, it's another thing to purposefully leave the bed on to provide heat for an acetone chamber and come back to suddenly find Marlin has turned it off.

Wackerbarth, 2015-09-16:
@ntoff -- Perhaps you should read about "Reductio ad absurdum".

ntoff, 2015-09-16:
Perhaps I know you were being sarcastic and that sort of thing helps no one. Even going there has no point at all and is like a small child saying no they're not going to eat their dinner because they're not allowed to play with their toys first.

Hear both sides out, offer a  solution that gives everyone what they want *if it's possible* and if not, go with the majority.

Wurstnase, 2015-09-16:
But why producing new code for something you can do with just some gcodes in your file?

atunguyd, 2015-09-16:
I am not suggesting that the firmware always shut itself down if left unattended, if you manually set the bed temp then I agree it should be left that way. What I am saying is that when a print terminates either because the print is finished or something went wrong then the printer should do something more than just freeze there. In my case the head stopped mid print and sat there oozing ABS all over a half completed print.
I cant fathom anyone here arguing that this is a good thing? Had it moved the head away from the print and shut off the heaters I would have been able to work out from the gcode where the printer stopped and then fed it a new gcode file to complete the job. However the presence of a big blob of ABS in my print precluded this option.

My suggestion is this:

Implement a EndOfPrint() function that is called when the printer detects anything that it assumes means the print is complete.
In EndOfPrint() we can issue commands like move head to a location, switch off heaters etc. Then each of these commands can be wrapped around #ifdef and these defines placed in configuration.h that way anyone that wants their printer to do something more than sit there at the end of a print can  enable this as they want.

atunguyd, 2015-09-16:
@Wurstnase - because things can go wrong - look at my case, the gcode was corrupted on the SD card, or in my opening post I changed slicers and the new one does not by default disable the heaters afterwards. 
Putting fail-safes into products is not a new idea. In fact Marlin already has plenty, I was just suggesting an area where it is not done and in my opinion, it should be. 

As I said earlier this is the one advantage of opensource, if the manufacturer (developers) think it is a stupid idea you can go ahead and just implement it yourself,.

Wurstnase, 2015-09-16:
Yes, maybe. I see some risks. If you handle it like EndOfPrint() and use this for the shutdown/end/error or whatelse. Someone will make something like M104 R50. What will happen on a temp-error?

Wackerbarth, 2015-09-16:
@atunguyd -- The problem in doing as you suggest is that, AFAICT, there is no way to distinguish between "failure" and intended action, or in-action. For whatever scenario you present, someone like @nophead will point out another situation where that is the desired behavior.

If things work properly, then there is no reason to have some special code. Any shutdown sequence can be built into the "program" driving the printer.

To handle the unexpected, as I have previously suggested, having a programmable watchdog (where the user can set the duration, including disabling it) is the only way that I see to respond to "lack of instruction".

closed 2016-05-04 by jbrazio