MarlinFirmware/Marlin #2240: Z position problem

paulnd, 2015-06-03:
I've been trying to resolve this problem for about 3 days but can't find my error.

I'm using DC42's IR probe on a Mendel90 and a Ramps 1.4 with the probe offset set. I also use Kisslicer and Repetier host.

When I G28 and G29 P3 and then G0 Z0 manually the hotend is at the correct height for printing (paper thickness). However when I start printing with homing and autobed levelling in the gcode, the first layer is at +0.36 not 0 and doesn't adhere to the bed, the only way to overcome this is by adding a -0.36mm offset in the slicer. I have tried different probe offsets but seem to be unable to get the hotend printing closer than 0.36mm from the bed unless I add an offset in the slicer.

I'll be grateful for any suggestions or advice please.

nophead, 2015-06-03:
Z0 should be nozzle just touching the bed, not paper thickness away.

Paper thickness is not the correct height for printing unless your paper just happens to be same thickness as your first layer. E.g. if you layer height is say 0.2mm then the gcode should do the first layer with Z = 0.2mm and that should be 0.2mm from the bed.

The way to check the Z calibration is to measure the height of the skirt and compare it with the Z value the skirt was extruded at.

AnHardt, 2015-06-04:
Apparently dubbing #1682

closed 2015-06-04 by AnHardt

lrpirlet, 2015-06-05:
I know this is closed, yet I think I can gather some knowledge here... 
@paulnd , could you please describe your printer... or better, would you mind to tell me if you have a X axis made of 2 smooth rods placed in a vertical plan with all the weight on one side of this plan (typically a Prusa I3, I have that sort of printer).
I have been working on the G29 (Homing & Leveling Bugs #2040) and got a fairly good result, measuring only the last known distance tip-bed. I downloaded the version 110 (development version a couple of days ago) where my modification was removed... using that 110 version, I end-up with a constant offset of +.15 mm...
BTW, I googled about g29/autolevel and found that this behaviour (small offset at the end of g29) was not uncommon and was solved using g1 Z-<offset> followed by g92 to reset the zero on Z... 
My theory, is that the algorithm used in G29 assumes that the distance between the hot-end tip and the bed is CONSTANT... This is not necessarily the case for a Prusa I3 where the torque is big enough to distort the X axis and render the tip-bed distance variable along the axis...
The real solution, for me, is to change the hardware configuration and/or to simply tram (level) the bed as good as I can, then use the mesh-bed levelling...

paulnd, 2015-06-05:
My printer is a Mendel90, the x axis is 2 horizontal smooth rods.

I haven't looked at mesh-bed levelling yet and may give it a try as this is very fustrating!

paulnd, 2015-06-05:
Disabling Z_SAFE_HOMING seems to have provided a temporary fix

lrpirlet, 2015-06-06:
@paulnd Many thanks, 
First you killed that low probability cause I described.
Then you reported that disabling Z_SAFE_HOMING improves the behavior...
 Disabling Z_SAFE_HOMING means that you have executed the G28 Z on ANOTHER SPOT on your bed... With Z_SAFE_HOMING, G28 probes the Z axis in the middel of the bed... Without Z_SAFE_HOMING, (I guess the probe is over the bed when the hot-end is at home) the  probing is done a lot nearer what I call the "rotation point"

thinkyhead, 2015-06-08:
#2040 is the place to get into the subtle leveling bug we are still tracking. For now mesh bed leveling or no bed leveling may be the most frustration-free approach.