MarlinFirmware/Marlin #2244: Workflow Discipline and Conventions for Automatic Versioning

Wackerbarth, 2015-06-03:
The infrastructure that has now been introduced for automatic versioning can be made to handle all of the version number generation automatically from branches and tags in the github repository.

However, for it to work optimally, a particular workflow needs to be followed when promoting Release Candidates and Releases.

It seems to work best if a tag is placed in the tree at the root of each minor release.
Thus, there would be a tag for "1.0", and another tag for "1.1".
The process is to build a single line of merges, one for each { point release or release candidate }

Thus, the latest release will be found at the head of the release branch.
Beyond that point, there will be a string of release candidates for the next release.  (Your "implementation" branch which I would rename "RC" for reasons that should become apparent)
Now, nothing gets released unless it first passes through RC testing. With hopefully only a few iterations, the head of the RC branch might be a commit that is known as, for example, 1.1.3 RC4. This commit would be exactly 4 merges later than the current release head (which, in this example, is known as 1.1.2)
To execute the release,  we `git checkout Release` and  `git merge --no-ff RC`. Thus, all of the Release Candidates that went into 1.1.3 appear as a loop between the 1.1.2 and 1.1.3 nodes. Think of the "local" and "express" stops along a rail route.  To continue development, we move the RC head up to this common point and then begin another "local" loop of Release Candidates.

Now, you should note that the RC "loop" can, and should have the same structure. Various "features" are promoted from the development branch and then merged into the RC branch.

It should be noted that the Development branch should be restarted with each Release candidate by rebasing the changes to the previous candidate before merging them into the chain that will become the next candidate.

If you are willing to accept this workflow, or something similar, I will modify the versioning scripts to implement it.

fmalpartida, 2015-06-03:
I like it and support it. Nice job!

thinkyhead, 2015-06-10:
Makes sense to me. Marlin needs proper versioning.

closed 2016-04-23 by jbrazio