MarlinFirmware/Marlin #2245: Z_DUAL_STEPPER_DRIVERS enable compile error

geekgarage, 2015-06-03:
Stable Release 1.0.2
Arduino compile log
http://pastebin.com/W0sZpRzx

I tried searching related issues but without luck, all the solutions refer to code that do not exist any more or has been fixed. I first tried this with the DEV release but the probing of bed did not work, it just moved in a square and that was it, no probing of the bed at all, it didn't lift the printhead. (inductive sensor), so i moved back to stable release.

Configuration.h
http://pastebin.com/g8ivucRA

Configuration_Adv.h
http://pastebin.com/zebccnHK

Wurstnase, 2015-06-04:
Where do you define the ```Z2_ENABLE_PIN```?

geekgarage, 2015-06-06:
Sorry for the late answer i had a family emergency. I've looked over the pins.h file and can see that Z2_ENABLE_PIN isn't defined for RUMBA board, i could define it and see if it will work I just don't know what pin i should use, but i do not have a secondary stop switch and i really do hope that this is going to be optional in the next firmware.

Wurstnase, 2015-06-06:
Have fun with it :)

closed 2015-06-06 by Wurstnase

thinkyhead, 2015-06-08:
Probably for all the boards that have enough stepper connections we'll patch up the code to do something like this instead of using the pin numbers:
```
#define Z2_STEP_PIN E2_STEP_PIN
#define Z2_DIR_PIN E2_DIR_PIN
#define Z2_ENABLE_PIN E2_ENABLE_PIN
```