MarlinFirmware/Marlin #2246: Travis CI Testing with Arduino 1.6.3+

Wackerbarth, 2015-06-03:
In reality, the testing done in the Travis CI process does not actually run the Arduino IDE. Instead, it has a "command line replacement" called `ino` which wraps the build process. However, the Arduino IDE is loaded into the build space in order to be able to extract the core files and compiler tools.

Unfortunately, Travis still runs on a very old version of Ubuntu 12.04 LTS.

1) I have managed to build using compilers and core files from the Arduino IDE 1.6.4. However,
I did so by including it in my github repository.  I certainly don't want for THAT code to ever get drawn into the MarlinFirmware repository because it doesn't really belong here and would bloat EVERYONE's copy of the Repository. ("Revert" removes the files that get checked out. However it doesn't purge them from the repository.)   Does anyone know of a copy of the Arduino IDE 1.6.4 that runs on Ubuntu 12.04 and that we can simply apt-get?

2) The file structure for the Arduino IDE changed with 1.5. `ino` doesn't handle the new structure.
I have been hacking into it to make it compile sketches.

3) `ino` doesn't really process the platform files associated with board selection. As a result, build options provided by "extra flags", etc. and the pre-build hook recipe never get processed. I'm working around this by translating those functions into the Travis script.  However, that means that, within the Travis tests, we are not, and, in reality, never have, actually tested Arduino IDE building. Any such testing will need to be performed by actual users.

Wackerbarth, 2015-06-06:
Another part of the issue is that, for its builds, Travis doesn't clone the branch and tag information from the git repository. As a result, the automatic versioning does not work as designed, or as it would for a typical user.

There are two ways to consider this:
1) We can add some code to the Travis setup which populates the currently active branches and tags.
2) We can ignore the situation and, rather than having an automatically created version number, discard, rather than distribute, the resulting .hex file.

AnHardt, 2015-06-09:
To the first message:
1) It should be possible to store your modified core files in an other reposity (yours), clone that during the travis build and copy if to the Marlin-build-directory on the fly. No. i don't know about a source for a newer arduino for Ubuntu 12.04. Even the source we try today seems not to include any arduino version.
2) Even here we could copy your stuff in on the fly. But - yes - an awful hack.
3) - or the developers. Ideally the requester as well as the merger gave the PR some compiles with __relevant__ configurations. Here we are much more intelligent than the script to make some __useful__ tests. The hope to put all possibilities into the travis script is illusory. It can only test some typical cases.
About your second message:
It's worse. Even the .zip files we provide on github do not include the information we need to produce the version numbers (.git folder).
1) Will this produce an endless loop?
2) Is impossible. We simply can't distribute some thousand .hex files for all the possible combinations.

Wackerbarth, 2015-06-09:
1) I have actually gotten thinks to work using a REAL Arduino IDE 1.6.4 distribution (Pushing soon) No need to have it anywhere other than their public repository.
2) I am no longer using ino at all. This has the advantage that it also processes the boars.txt file and the platform.txt (and platform.local.txt) which allows my versioning code to operate automatically.

I think that I may be able to perform the copy automatically by using the pre-build hooks that I added to IDE 1.6.3

boelle, 2015-06-29:
did we not merge this one at the hangout?

Wackerbarth, 2015-06-29:
Yes. It indicates "merged" above. I guess that we need to close issues as we verify that the merged fix solved the issue.

closed 2015-06-29 by Wackerbarth

boelle, 2015-06-29:
doh, me going blind :-D