MarlinFirmware/Marlin #2251: LCD Control > Motion menu cannot effectively control some large values

dantman, 2015-06-04:
The "Control > Motion" menu seems to allow you to control a number of settings and test out other calibrations.

Various values have various levels of specificity (some increment by 100, others 1, others 0.1) and the values themselves come in various exponents. However sometimes you can end up with a value that is both very large and very fine. For example I have a zsteps/mm of 2000.0 that is controlled at an increment of 0.1, meaning to test calibration at 1000.0 using the LCD I'd have to turn the knob 10000 ticks.

I think the control menu (at least the motion one) should use an increment selector like the one used in axis movement. Except after instead of before. ie: You select 'Zsteps/mm: +####.#' then select one of "+/- 1000", "+/- 100", "+/- 10", "+/- 1", "+/- 0.1". (With units if possible, and some unnecessary exponents omitted depending on context).

((I understand I can use serial commands or upload new firmware to test this; But I filed the bug since this feature does exist in in Marlin and I think this is important for that feature to be worthwhile))
----
For context, I've got a RAMBo with digital micro-stepping controls and I wanted to test various micro-steps vs. steps-per-mm and my acceleration/speed to see what the optimum way of handling two issues with my z-axis configuration.

AnHardt, 2015-06-04:
I guess what #1521 does is pretty much what you want. It's already in the Development branch.
The feature is called ENCODER_RATE_MULTIPLIER. The configuration is in Configuration_adv.h

dantman, 2015-06-05:
If I find some time I'll re-assemble my printer, upload Development to it, and see how much that improves the ability to change `2000.0` to `4000.0`.

dantman, 2015-06-07:
Correction, `1000.0` to `2000.0`. And unfortunately the encoder rate multiplier doesn't help enough. It barely helps when you're trying to change increments of > 200 (and it's probably fairly taxing on the encoder to continuously be spun so fast). And it's of course useless when you have to do a change that requires 10000 increments (1000.0 -> 2000.0 in 0.1 steps).

AnHardt, 2015-06-09:
ENCODER_RATE_MULTIPLIER is currently not implemented for any value in the Control-menu.
I tried to change `Zsteps/mm` with the multiplier - it's not good, but,at least improved - for my taste.

You can try it by simply replacing `MENU_ITEM_EDIT` by `MENU_MULTIPLIER_ITEM_EDIT` in line 1061 of ultralcd.cpp.

If wanted, the discussion about what menu-items should get the multiplier, will take more time than actually implementing it.

closed 2016-05-04 by jbrazio