MarlinFirmware/Marlin #2252: LCD Control > Motion menu canot control digital microsteps and digipots

dantman, 2015-06-04:
This one is a feature request. ((Probably simple enough I could write it, but marlin looks to be in a feature freeze right now))

Some boards have stepper drivers where the microstepping and the current trimpots are controlled digitally. And these can be changed live as serial commands already exist to do so.

The Control > Motion menu allows for testing of different values of other settings that are used in conjunction with these. So I think it would make sense for Control > Motion to also allow these calibration settings to be tested.

----

Side note, the whole reason I first looked for this in the control menu was because the documentation on `M350` wasn't detailed enough to tell me how to target a specific axis/stepper.

AnHardt, 2015-06-04:
Yes, we could make this settable by the menu

But even the permanent inclusion of steps/mm in the menu is of limited sense.
How about a 'setup-menu' what can be disabled in the configuration after a initial setup phase. Things like microstepping and steps/mm you'll likely never touch again when you have the correct settings for your machine. This could make room in the memory for other features. Maybe ABL+ LCD will than fit in for the 128K processors.

dantman, 2015-06-04:
I could agree with that. I had already thought there might be some reason to have something like a DEBUG_(MODE|...) configuration.

There are probably a number of other things we could add to the LCD that would help make it easier to get things working but generally are worthless otherwise which we could add under a setting like that. 

For example, a "Debug > Encoder pulses" screen that gives you information about what encoder pulses Marlin is seeing. This could be helpful for identifying what pulses per step and inversion settings are needed for a particular LCD. But afterwards that screen would be mostly useless.

closed 2016-05-04 by jbrazio