MarlinFirmware/Marlin #2253: Hotbed issue

ruggb, 2015-06-05:
I just added a hotbed and it won't compile with a thermistor defined.
from a good configuration.h, I set extruders to 2 and set E1 therm to 8 (E0=8) and bed therm to 8 and it does not compile.
If I set bed therm back to 0 it complies OK.
It does not compile with them set to 1 either.
I did not uncomment the thermal runaways for the bed
they are uncommented for the extruders
what did I miss?

this error
temperature.cpp: In function 'void manage_heater()':
temperature.cpp:566: error: 'THERMAL_RUNAWAY_PROTECTION_BED_PERIOD' was not declared in this scope
       thermal_runaway_protection(&thermal_runaway_bed_state_machine, &thermal_runaway_bed_timer, current_temperature_bed, target_temperature_bed, 9, THERMAL_RUNAWAY_PROTECTION_BED_PERIOD, THERMAL_RUNAWAY_PROTECTION_BED_HYSTERESIS);
                                                                                                                                                      ^
temperature.cpp:566: error: 'THERMAL_RUNAWAY_PROTECTION_BED_HYSTERESIS' was not declared in this scope
       thermal_runaway_protection(&thermal_runaway_bed_state_machine, &thermal_runaway_bed_timer, current_temperature_bed, target_temperature_bed, 9, THERMAL_RUNAWAY_PROTECTION_BED_PERIOD, THERMAL_RUNAWAY_PROTECTION_BED_HYSTERESIS);
                                                                                                                                                                                             ^
'THERMAL_RUNAWAY_PROTECTION_BED_PERIOD' was not declared in this scope

ruggb, 2015-06-05:
this is 1.0.2-1

ruggb, 2015-06-05:
it will compile if I uncomment those items
Is that by design? If so maybe there should be a note.

dantman, 2015-06-05:
It looks like that section of code tests the `THERMAL_RUNAWAY_PROTECTION_PERIOD` then uses `THERMAL_RUNAWAY_PROTECTION_BED_PERIOD`. So it's a coding mistake. (If you want to workaround wile using Release changing the `#ifdef` to use the right constant would work)

It only shows up when you have the bed on because that section of code only runs when the bed is enabled and you have runaway turned on for your hotends.

Looking at Development this section of code has been changed. Now instead of testing for the general period it tests for `THERMAL_PROTECTION_BED`.

So this bug looks like it should already be fixed in Development.

thinkyhead, 2015-06-10:
@AnHardt Worth patching such egregious bugs in the 1.0.2 tag (which may in future become a branch as versioning goes forward)?

AnHardt, 2015-06-10:
@thinkyhead 
Yes. If its so simple to fix it like in #2266.
Lets collect some of this before making a new tag.
In the meanwhile the fixes are available when using the Release branch - not the tag.

thinkyhead, 2015-06-10:
@AnHardt I'd like to go through all the PRs (or commits) since 1.0.2 was tagged anyways so that I can fill out the Change Log. I'll make note of which PRs (or commits) include fixes for bugs like this one. I knew (vaguely) that this bug existed in 1.0.2, and I vaguely recall that there exist at least one or two other behavior-related bugs that will need relatively small patches.

ruggb, 2015-06-12:
I commented out both sets of those in config and changed the line in temperature
won't compile
Both uncommented = compiles
bed pair uncommented = wont compile
hotend pair uncommented = compiles

???

ruggb, 2015-06-12:
curious - I just noticed
    #ifdef THERMAL_RUNAWAY_PROTECTION_BED_PERIOD && THERMAL_RUNAWAY_PROTECTION_BED_PERIOD > 0
those factors are both the same - that doesn't seem right

ruggb, 2015-06-12:
but then there are lots of others so I guess I don't understand the code.

Wackerbarth, 2015-06-12:
There are three possible states for a variable of this type:
1) Undefined
2) Defined, but 0
3) Defined, > 0

We choose to give states (1) and (2) the same behavior.

However, you cannot simply test XXX >0 because that test will encounter an error if the variable (XXX) is undefined.

Therefore, you will see the construct that you reference fairly often in the code from many projects.

The compiler is smart enough to test the first condition and, finding it false, realize that the result will be false, thus avoiding the

thinkyhead, 2015-06-12:
@Wackerbarth Well, strictly speaking the preprocessor will turn any undefined symbols referenced in `#if` directives into a 0 integer. So @ruggb is technically correct. That line could be `#if THERMAL_RUNAWAY_PROTECTION_BED_PERIOD > 0` and it would function just fine. The more serious problem with that line is that it starts with `#ifdef` instead of `#if defined(…`. There are probably a few lines like that still hanging around in 1.0.2 that could stand to be patched up.

AnHardt, 2015-06-12:
@ruggb 
I doubt the usefulness of TRP for the bed only - but #2284 is the solution.

ruggb, 2015-06-12:
I'm confused again - surprise!
I took a clean copy  NO CHANGES, it compiles OK
then I changed ONLY
58 #define EXTRUDERS 2
it does not compile

R U saying if I chg line 565 and 1016 in temp-cpp
and line 165 in temp-h
this will all work properly??

It just seems like if changing 1 thing not associated with TRP breaks it - there may be something else going on.

ruggb, 2015-06-12:
and the answer is - it does not compile with those changes to temp
ANOTHER ISSUE????

emartinez167, 2015-06-12:
I can confirm the issue with failure to compile with EXTRUDERS = 2. Seems to be an array definition out of bounds.

Regards,

Ernesto

Wackerbarth, 2015-06-12:
What is “breaking”?

In particular, if you specify a MB that does not have two extruders defined, TRP, or no TRP, the compile will fail.

ruggb, 2015-06-12:
i set 
#define MOTHERBOARD BOARD_RAMPS_13_EEB
and it still doesn't compile
-- with or w/o chg to temp files and with extruders =2

AnHardt, 2015-06-12:
Indeed.
With Arduino 1.6.3 i get this sweety:

>temperature.cpp: In function 'void manage_heater()':
temperature.cpp:665:1: error: unable to find a register to spill in class 'POINTER_REGS'

This is an internal compiler error.
The error is completly independent of the TRP stuff.

Compiles great with 1.0.6.

This is outside the scope __I__ want to fix here.

Wackerbarth, 2015-06-12:
Can you push it to some github repo and give me the SHA of the commit?
I’ll take a look at it on a different version of the compiler.

AnHardt, 2015-06-12:
#1603

ruggb, 2015-06-12:
I am using 1.6.4

ruggb, 2015-06-12:
When I do get this to run by having both uncommented, it bombs on an over temp error about  2 min into the print.
Is this related to this issue or have I got something not set right.
I did not chg any of the param on the defines and the temp curves don't indicate overtemp though the hotend oscillates more at 230 than 210. The hotbed is steady but has a hard time getting to 110.

AnHardt, 2015-06-12:
@Wackerbarth 
It's the plain Release Branch with more then 1 extruder, matching board, matching sensors.
But don't waste time to test - it's fixed with #1603. But #1603 is to big to oversee what new errors where introduced by that. Therefore __my__ decision that __I__ will not do the patch for the Release branch.

Wackerbarth, 2015-06-12:
> On Jun 12, 2015, at 12:16 PM, AnHardt <notifications@github.com> wrote:
> 
> #1603 <https://github.com/MarlinFirmware/Marlin/pull/1603>
> 
/Users/Projects/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy -O ihex -R .eeprom /var/folders/qq/sc5wp5896yg_3xm169_4fw040000gn/T/build5268376626209386576.tmp/Marlin.cpp.elf /var/folders/qq/sc5wp5896yg_3xm169_4fw040000gn/T/build5268376626209386576.tmp/Marlin.cpp.hex 

Sketch uses 51,896 bytes (20%) of program storage space. Maximum is 253,952 bytes.
Global variables use 3,204 bytes (39%) of dynamic memory, leaving 4,988 bytes for local variables. Maximum is 8,192 bytes.

No problem on my Mac OS X.  (Now I learn that it was PRE-1603 where you had the issue. I’ll not bother unless you change your mind about it.)

a4jp-com, 2015-10-10:
Did this get fixed?

ruggb, 2015-10-10:
in Marlin-RC2-91b5156b3e
both uncommented = WARNING
either uncommented = WARNING
niether uncommented = WARNING

d:\Users\Bill\Documents\Arduino\libraries\U8glib\src\clib\u8g_rot.c:48:1: warning: initialization from incompatible pointer type [enabled by default]
==========================
 u8g_dev_t u8g_dev_rot = { u8g_dev_rot_dummy_fn, NULL, NULL };

 ^

d:\Users\Bill\Documents\Arduino\libraries\U8glib\src\clib\u8g_rot.c:48:1: warning: (near initialization for 'u8g_dev_rot.dev_fn') [enabled by default]
========================

Guess there is more work to do someplace. I have upgraded Arduino & u8glib and I don't recall if I tested this on the old version of u8glib, Since it is just a warning, it does compile. 
So is there a problem with u8glib?

I am running with one uncommented which I couldn't do before.

KiteLab, 2015-10-10:
@a4jp-com 
As in this thread we talked about several different issues - what is the "this" in your question.

ruggb, 2015-10-11:
when i ran it previously it would not compile.
It now compiles but there appears to be an error in u8glib.
and I don't know if I tested all the variations with the current Arduino & u8glib.
I suppose I did not notice the warning previously because of the small window.
I saw it complete and never scrolled it back to see if there was a warning.
Don't know why I would since it did compile.

I don't suspect that the u8glib issue has anything to do with Marlin, so I guess this is fixed.
At least I don't see an issue - for what that is worth.

KiteLab, 2015-10-11:
@ruggb 
When i compiled Marlin, with all warnings on, some weeks back, with u8glib 1.17, i saw a couple of warnings.
At the moment, with the standard settings of the IDE, i cant see any warnings about u8glib 1.18.

I guess the difference in you compiles is the setting about warnings.

As 1.17 was tested intensively, in all aspects we use it, ignore the warnings until we can see, or hear, about problems related to u8glib 1.18. 
The changed directory structure in u8glib 1.18 is already respected by to newer versions/branches of Marlin. I see no chance to do this in 1.02/release (ask @Wackerbarth why `release` delivers code worse than it could and used to do).

KiteLab, 2015-10-11:
@a4jp-com 
The `THERMAL_RUNAWAY` bugs where fixed in 'release' but you can find that fixes now only in the `1.0.x` branch.
The "don't compiles with 2 extruders in combination with some other configuration options"  under Arduino IDE 1.5.x, 1.6.x is fixed for the newer branches but not for `release` and `1.0.x`.

Let's concentrate on the RC and then make the current 'release' as inaccessible as possible.

ruggb, 2015-10-11:
@KiteLab
u8glib 1.18 ???? -- all I see on https://code.google.com/p/u8glib/downloads/list is 1.14.
where is 1.18??

KiteLab, 2015-10-11:
https://github.com/olikraus/U8glib_Arduino

ruggb, 2015-10-11:
thx
I think I got it. Actually, I think it updated itself to 1.18 when I updated Arduino to 1.6.6, but since I only found 1.14 that is what I thought it was.

KiteLab, 2015-10-12:
Updating the links in the configs (and the doku) to:
https://github.com/olikraus/U8glib_Arduino
could be a good idea.

a4jp-com, 2015-10-12:
@KiteLab - The compiling bug/error.

KiteLab, 2015-10-12:
@a4jp-com 
You are funny. There are two.
Please read:
https://github.com/MarlinFirmware/Marlin/issues/2253#issuecomment-147179421

KiteLab, 2015-10-12:
At least the Milestone for this issue could be removed.

AnHardt, 2015-10-12:
@KiteLab 
That's not that easy.

I will close this issue because:
The title problem is/was solved.
The second problem is known how to avoid it.
The u8glib warnings are harmless.
Appending more and more issues to an issue is no good style.

@KiteLab 
Please open a new issue for the u8glib - link problem.

closed 2015-10-12 by AnHardt