MarlinFirmware/Marlin #2256: Z-axis moves really slow after updating new marlin

slanwar, 2015-06-06:
- Using the new firmware the Z-axis runs too slow and for example if I move the Z up 10 mm moves maybe 2 or 3 mm but very slow, I have the folger prusa i3 and I have (and tried) 3 sets of ramps and arduinos (2 legit's and the other one Chinese), same problem. The first time I setup the auto leveling everything worked fine excepted the hot end was about 5-8 mm up and was printing in the air. Tried to set the extruder with the probe and takes me forever to move z-axis to almost touch the glass and when I typed M114 I got -88.60 for z when the first time I setup was about -10.
- Also all of this problem started after the first print out after I installed auto leveling "almost" successfully and was a huge mess when printing without supervising.
- I updated the original old config file from folgertech and works fine except I can't have my LCD as well.

brainscan, 2015-06-06:
What do you have your feed rates set to and has your steps per mm changed? Also are you using the same microstepping in firmware as you have set on your drivers?

slanwar, 2015-06-08:
I had all the feed rates wrong, when I copied over the settings from old config file I overlook a part which was causing all the problems. Because you asked about the feed rates I looked and I realized I never changed.... Thanks

closed 2015-06-08 by slanwar