MarlinFirmware/Marlin #2259: Probe Area defined but not working correctly

geekgarage, 2015-06-06:
Hi All,

I got the probe area setup and i'm doing it like Thomas in the video
https://youtu.be/EcGFLwj0pnA?t=593

But here is where it's going wrong. I do the homing as instructed and then run the leveling procedure with the G29, but insted of probing the area defined it's messing up in a manner that i can't seems to figure out why, it does probe but not at ill in the area i defined. See the picture
[image]

My configuration
http://pastebin.com/sAgwYX33

My configuration_adv
http://pastebin.com/ErrYKM00

My Pins
http://pastebin.com/jSXn5U8v

geekgarage, 2015-06-07:
Seems like i have to do G28 X0 Y0 and then G29 and then it probes in the correct locations. Insted of starting probing in the center ogf the bed it homes to 0,0 and then i can start the probing. Is this the correct way of doing it or is some thing wrong with the setup??

geekgarage, 2015-06-07:
closing issue as there is no help to find here

thinkyhead, 2015-06-08:
Hi @geekgarage I just moved from Seattle to Portland, so haven't been around to help out. Sorry for that. I have some things to put together if you want to help debug, just updating my debug branches, which print extra output to the console regarding bed homing, probing, and leveling.

geekgarage, 2015-06-08:
I would love all the help i can get. As of now, i can get it to level within the area if i do a G28 X0 Y0 first so it doesn't home to center, but when it's done leveling it's off on X and Y axis by 10+ mm. I'm not at at the printer right now, so i can't tell the exact amount it's off.

But after failing to get it fixed i moved on to the DEV branch, and here i got an issue regarding homing Z. In the configuration.h i did the following
// @section homing

// coarse Endstop Settings
//#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#ifndef ENDSTOPPULLUPS
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  // #define ENDSTOPPULLUP_XMAX
  // #define ENDSTOPPULLUP_YMAX
  // #define ENDSTOPPULLUP_ZMAX
   #define ENDSTOPPULLUP_XMIN
   #define ENDSTOPPULLUP_YMIN
  // #define ENDSTOPPULLUP_ZMIN
   #define ENDSTOPPULLUP_ZPROBE
#endif

But after doing this it will home X and Y normally but Z will move down to the bed with normal speed, then pause for 20-30 seconds, then it completes the Z homing by moving up and then down slowly.

As of this point i'm ready to try anything to get it up and running.

thinkyhead, 2015-06-08:
One thing I've seen lately is that in the development version it helps to make sure you've disabled the max endstops (if they are not disabled already).

geekgarage, 2015-06-08:
I will do that later today when I get off work, as they aren't disabled.

Can you explain what this ENDSTOPPULLUP_ZPROBE actually do, should it be enabled when i have an inductive sensor for bed probing?

brainscan, 2015-06-08:
I think if you're using mechanical end stops they need the pull-up resistors on to get a signal. I've only used Hall effect sensors and a capacitance probe so I've not had to use them but unless you're using mechanical switches it doesn't matter if it's on or off. 

Hopefully someone will correct me if I've got this wrong.

Wackerbarth, 2015-06-08:
Typically, mechanical end stops are just a switch -- Two wires. They require pull ups, either externally, or internally from the mpu.  So, it all depends on what circuitry is on the interface board. Some of the industrial ones have components to protect the rest of the circuit (the mph in particular) from stray electrical events.

However most of the inexpensive ones connect directly to the processor pins.

YMMV

Nandox7, 2015-06-08:
That's odd.
In my start gcode I use:
---
G28
G29
---
And it works fine, the only issue is that I had to add a G1 X0 Y0 after the G29 if not it would start printing
from the far corner where it stopped the G29 process.

Question: As we can't quite define the movement for the G29 process , wouldn't it make sense for the printer to return to the X0 Y0 point as it would happen when doing G28 in a printer without the sensor?

AnHardt, 2015-06-08:
@geekgarage 
The first time it took __me__ a while to understand that the coordinates for the probing area describe the positions of the sensor - not the nozzle. ?_PROBE_OFFSET_FROM_EXTRUDER is taken into account automatically. This could be related.

geekgarage, 2015-06-09:
When i got home yesterday my internet connection was down... and it still is this morning, as soon as it get up i can post a video of how it reacts after the proposed changes.

geekgarage, 2015-06-09:
@brainscan @Wackerbarth 
I only have endstop at X and Y min, is there a way for me to use  multimeter and see if i need pullup or not?

@Nandox7 
When you run the G28 does it center your probe on the bed?

@AnHardt 
Yep and all the probe locations are set after the nozzle location not the probe:) But thank for the info

@thinkyhead 
After i disabled the max endstops i got the same result, as you can see in this video
https://youtu.be/oDVvk-Rsvlg

My Configuration.h used in YT video
http://pastebin.com/gipX0EPJ

My Configuration_adv.h used in YT video
http://pastebin.com/rb2z4sqB

pins_RUMBA.h used in YT video
http://pastebin.com/36QtjzQj

Since I don't have //#define Z_PROBE_ENDSTOP uncommented can i expect the firmware to use the probe connected to Z-min on the board for probing the bed?

BTW to all of you, thanks for all the inputs, i really appreciate it!

Nandox7, 2015-06-09:
@geekgarage yes it does.
This was one of the first tries: https://goo.gl/photos/VjxRFef7SgzGmUBQ6
I had to tweak the probing points as the bed was hitting the max and the nozzle cleaning was done in the opposite corner.

geekgarage, 2015-06-09:
@Nandox7 i don't follow?? I was asking if your homing centers your probe on the bed? I know you are just trying to help, but adding non related stuff to the conversation just confuses me more, right now i'm trying to understand why the homing acts like it does on the video. But thx for the input.

Nandox7, 2015-06-09:
@geekgarage I replied to that question. Yes it does center the probe in the bed.
And you can see that in the video.
when I run the G28, it firsts goes to the min positions for both X and Y and after it does the Z min position after centering the probe in the center of the bed.

Wurstnase, 2015-06-09:
Simple activate each pullup of your endstops. It is not necessary to deactivate it.

Anyway, to test it just restart your printer. Send M119. Trigger an endstop and send M119. Untrigger send M119. If it's still triggered you need a pullup.

But you don't need to deactivate the pullups.

AnHardt, 2015-06-09:
@geekgarage 
No you don't need a  multimeter to test the endstops. Just ask Marlin by a M119 what 'Marlin sees and trigger the switches manually to see the difference. Normally an inductive (active) probe does not need a pullup, but normally the pullups do not hurt (as they have a pretty high resistance). Depending on your sensor (NPN/PNP) you have to play with *_ENDSTOP_INVERTING.

Where G28 makes the z-probe, and ends, depends on your settings. Z_SAFE_HOMING tells the printer to home with the nozzle in the center of the bed. Especially for your setup, with the probe 6mm left of the nozzle this makes sense - else the probe may not hit, or only partially hit, the bed.  Z_SAFE_HOMING can only be defined together with ENABLE_AUTO_BED_LEVELING - else is assumed the z-min-switch is not mounted close to the nozzle but on an axis.

The order of the definition is not important here. The evaluation is done much later.

It used to work pretty good without the definition of Z_PROBE_ENDSTOP.

AnHardt, 2015-06-09:
@geekgarage 
You told us you'v build the sensor with the information of Tom's video. If you used resistors with much higher (factor 10 or more) values than he used - turning off the pullups may be necessary.

thinkyhead, 2015-06-09:
@geekgarage That video shows some weird behavior, alright.

Before testing with 1.0.3 you have to update your configuration files to the latest versions, otherwise some things simply won't work. Start with clean copies of the configs from 1.0.3 dev and then make your customizations based on your old config. Then see if the results improve.

geekgarage, 2015-06-09:
@thinkyhead 
My config that I link to are from Development Branch v1.0.3 not the Dev branch

Going to test some of the inputs that you guys have posted

thinkyhead, 2015-06-09:
> My config that I link to are from Development Branch v1.0.3 (not 1.0.2)

@geekgarage Ah, so they are. I'll have to look closer to see what's wrong. It looks like the nozzle/probe is only centering your X but not centering your Y when you do safe homing.

AnHardt, 2015-06-09:
@thinkyhead 
With the probe 70mm behind the nozzle - the probe could be centered.

geekgarage, 2015-06-09:
alright so after looking once more at the code i see since i have 2 z axis steppers each on their own step driver i've enabled the #define Z_DUAL_STEPPER_DRIVERS. That enables the z2_max and according to M119 it's always triggered. I have no z2_max endstop switch and would like to be able to disable it, just to make sure that it's not the cause to the issue, but i have no clue as to whare or what i have to change to do so.

The code also define

#define DISABLE_XMAX_ENDSTOP              //Better to disable the XMAX to avoid conflict. Just rename "XMAX_ENDSTOP" by the endstop you are using for Z2 axis.

can i use this some how to disable the Z2 instead?

geekgarage, 2015-06-09:
@thinkyhead  @AnHardt 
UPDATE: After messing around with the pullups it's now open (as i don't know how to disable the Z2_MAX) but even after it says open now, it still have 30+ sec og hang time before finalizing the Z axis homing. X and Y homing is working fine.

AnHardt, 2015-06-09:
Some stupid questions.
When your sensor is in mid air - is the LED (at the end of the sensor) on or off?
Is in on or off during the big pause?
In the video i hear a knocking sound when the motion stops. Is the nozzle or sensor then touching the bed?
What material is your bed? I have seen a PCB-Glas bed working everywhere very well with an inductive probe, but failing at the center - where the hole for the thermistor is.

geekgarage, 2015-06-09:
@AnHardt 
Not stupid questions at all
When the sensor is in mid air the LED is off, and in the big pause it's on, the pause starts and movement stops the second the LED turns on.
The knocking is just me chair, it's not from the printer.
My bed is aluminium, original MK3 heatbed (not wired up yet) with BuildTak like surface from BQ.

Just tried to disable safe home, as my sensor is still within the bed even tho it is 6mm to the left of the nozzle, and it's still doing the big pause, so I'm keen to think that it might have some thing to do with the dual Z axis step driver enabled thing. I will try and disable it and move the leads from one z axis motor and mount it in parallel and hope that it fixes the issue

geekgarage, 2015-06-09:
Tried again to convert to a single stepper for both Z Axis motors but my drivers aren't powerful enough to drive both motors even tho the power is maxed out, so i can't test if it's the dual z axis enabled that the cause of this issue

Wurstnase, 2015-06-10:
```
#ifdef Z_DUAL_STEPPER_DRIVERS

  #define Z_DUAL_ENDSTOPS

  #ifdef Z_DUAL_ENDSTOPS
    #define Z2_STEP_PIN E2_STEP_PIN           // Stepper to be used to Z2 axis.
    #define Z2_DIR_PIN E2_DIR_PIN
    #define Z2_ENABLE_PIN E2_ENABLE_PIN
    #define Z2_MAX_PIN 36                     //Endstop used for Z2 axis. In this case I'm using XMAX in a Rumba Board (pin 36)
    const bool Z2_MAX_ENDSTOP_INVERTING = false;
    #define DISABLE_XMAX_ENDSTOP              //Better to disable the XMAX to avoid conflict. Just rename "XMAX_ENDSTOP" by the endstop you are using for Z2 axis.
  #endif

#endif // Z_DUAL_STEPPER_DRIVERS
```

This is from the config_adv and should be similar with yours. Please change this to:

```
#ifdef Z_DUAL_STEPPER_DRIVERS

  #define Z2_STEP_PIN E2_STEP_PIN           // Stepper to be used to Z2 axis.
  #define Z2_DIR_PIN E2_DIR_PIN
  #define Z2_ENABLE_PIN E2_ENABLE_PIN

  //#define Z_DUAL_ENDSTOPS

  #ifdef Z_DUAL_ENDSTOPS
    #define Z2_MAX_PIN 36                     //Endstop used for Z2 axis. In this case I'm using XMAX in a Rumba Board (pin 36)
    const bool Z2_MAX_ENDSTOP_INVERTING = false;
    #define DISABLE_XMAX_ENDSTOP              //Better to disable the XMAX to avoid conflict. Just rename "XMAX_ENDSTOP" by the endstop you are using for Z2 axis.
  #endif

#endif // Z_DUAL_STEPPER_DRIVERS
```

geekgarage, 2015-06-10:
After changing the code as suggested, i moved X, Y and Z away from the end stops so none should be active and i ran the homing and it went perfect, no more big pause, so it seems like there is an issue when activating the z2 endstop :)
```

Wurstnase, 2015-06-10:
> when activating the z2 endstop

do you have 2 z-endstops or just 1?

geekgarage, 2015-06-10:
i only have one, I'm just stating that when activating the Z2 endstop it will cause this error with the big pause.

Wurstnase, 2015-06-10:
Ok, got you. 

I will make a PR for that. The second endstop should be deactivated by default also when you using dual z stepper drivers. The rest of the code is already right, just the config_adv needs an update.

geekgarage, 2015-06-10:
yes or at least make it possible to to deactivate it like the change you posted.

Wurstnase, 2015-06-10:
#2277 solved

closed 2015-06-10 by Wurstnase