MarlinFirmware/Marlin #2267: Builds in Travis Test Suite Disabled -- TestOptions1.h -- Two Extruders

Wackerbarth, 2015-06-09:
For this test to work, we need to specify the correct motherboard.

The command "cp Marlin/example_configurations/ramps/TestOptions1.h Marlin/Options.h" exited with 0.
5.62s$ DISPLAY=:1.0 ~/bin/arduino --verify --board marlin:avr:mega  Marlin/Marlin.ino
Loading configuration...
Xlib:  extension "RANDR" missing on display ":1.0".
Initializing packages...
Preparing boards...
Verifying...
In file included from Configuration_adv.h:600:0,
                 from Configuration.h:802,
                 from Marlin.h:22,
                 from watchdog.cpp:1:
SanityCheck.h:310: error: #error HEATER_1_PIN not defined for this board
       #error HEATER_1_PIN not defined for this board
        ^
#error HEATER_1_PIN not defined for this board
The command "DISPLAY=:1.0 ~/bin/arduino --verify --board marlin:avr:mega  Marlin/Marlin.ino" exited with 1.

AnHardt, 2015-06-09:
For example a 
    #define MOTHERBOARD BOARD_RAMPS_13_E __E__ B
will do the trick.
To make the test meaningful also a second thermistor must be defined. That implies #ifndef clauses around the temperature sensors in the config. 

Every not commented out option in the config needs a #ifndef clause to be  settable by Options.h.

Wackerbarth, 2015-06-09:
Using macros, I think that we can get around the issue once we get PR #2169 (or similar) available.

closed 2015-06-10 by Wackerbarth