MarlinFirmware/Marlin #2268: Allen Key Probe lifted twice on Delta Auto Bed Leveling

schlotzz, 2015-06-09:
Hi folks,

I'm very new to GitHub and this is my first opened issue, so please be kind ;-)

Using a Kossel Mini for about 1.5 years I tried switching to the latest Marlin Development version (last commit 0c71bdd14ff3647198a347c9a27f43eb295d2e74). Having configured the delta printer for using AutoBedLeveling with an allen key I was wondering, why the probe keeps being lifted twice after each bed contact. It looks like Z_RAISE_BETWEEN_PROBINGS is being applied twice.

With my rusty C++ knowledge I tracked it down to the following:

**Z_RAISE_BETWEEN_PROBINGS applied the first time**
```
inline void gcode_G29() {
...
z_before = probePointCounter ? Z_RAISE_BETWEEN_PROBINGS + current_position[Z_AXIS] : Z_RAISE_BEFORE_PROBING;
...
}
```

**Z_RAISE_BETWEEN_PROBINGS applied the second time**
```
 static float probe_pt(...) {
...

    #if Z_RAISE_BETWEEN_PROBINGS > 0
      if (probe_action == ProbeStay) {
        do_blocking_move_to(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS); // this also updates current_position
        st_synchronize();
      }
    #endif
...
}
```

Removing the second block it seems to function right. Maybe this block shouldn't be applied or should only be used for non-deltas?

Looking forward hearing from you.

thinkyhead, 2015-06-12:
@schlotzz I think you're right. I noticed this a few weeks ago and apparently forgot to patch it. #2281 should do the trick!

Wackerbarth, 2015-06-23:
Can we close this one?

schlotzz, 2015-06-23:
Yes, bug seems fixed.

closed 2015-06-23 by schlotzz