MarlinFirmware/Marlin #2275: Testing Marlin 1.0.0 on Scoovo printer and calibrating for the first time.

a4jp-com, 2015-06-10:
I'm going to calibrate my Scoovo printer using the original files then after that try a newer version of Marlin.

Marlin 1.0.0.
Windows 7 64-bit

My Marlin.ino file links to 48 files in marlin / Arduino 1.6.4, so I guess it's a lot of work to check through all the files.

I have seen a lot of talk about the newest version of Marlin and Arduino 1.6.4 so I'd like to know if I can update the firmware using Arduino 1.6.4 or if I have to use an older version of the software.

This is my first time flashing the software so I don't wanna get anything wrong.

My prints aren't square so I guess my settings here are wrong:

#define DEFAULT_AXIS_STEPS_PER_UNIT   {71.6,71.6,1067.8,97.5}  // default steps per unit for ultimaker 123 //2014/02/28 105->100 105->90

I printed a 10mm cube and got 
x 10.21mm
y 10.40mm
z 10.27mm

What is to best way to get almost perfect results?

a4jp-com, 2015-06-10:
A copy of the original marlin files with all the default settings can be found below. The original maker sold the printer to a factory in Japan so they don't know that much about the printers and they can't really offer software support so I will have to do everything. Sorry for posting here. I will start with Marlin 1.0.0 and then do the other versions to make sure everything is bug free.

http://a4jp.com/3d-printer-marlin-firmware/

AnHardt, 2015-06-10:
Arduino 1.6.4 is a bit unstable on Wiindows 7 64. Instead of throwing errors it often hangs. For now 1.6.3 is the better choice - at least for Windows 7 64.
For the most boards (including RAMBO) Arduino 1.6.x is ok to compile Marlin in every version.

 DEFAULT_AXIS_STEPS_PER_UNIT is a property of the hardware do not change the values.
Play with other parameters. 
https://www.youtube.com/watch?v=VIHgDiCCY0Y
http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide

a4jp-com, 2015-06-10:
I watched the video and saw the calculation formulas and also looked at the calculation guide but don't understand why my X,Y & Z figures are off. Mmmmm. If the figures are right what other settings should I adjust to get better results?

Wurstnase, 2015-06-10:
Backlash, Z-Wobble, just 2 options. But it's not Marlin.

a4jp-com, 2015-06-10:
They are the only values I can easily see that change the X,Y & Z settings. Where are the other settings? I watched the video and know about the belt and pulley.  

Maybe the company used a number that was a little bit wrong.

They said 71.6

but this is maybe the real calculation:
(200 * 16) / (2.032 x 22) = 71.58196134574087

How accurate does the number have to be?

Do you think this slight variation would throw off the X and Y values a lot?

Also this value (71.58196134574087) is the same for both X and Y but the measurements are different when printing. Where can I adjust the difference in the Marlin settings?
 
----------------

Nozzle diameter: 0.4mm + Filament diameter: 1.75mm

MXL Timing Belts - .080" & 2.03 mm Pitch

B 76 MXL
pitch 6.08 inches, 154.432mm
76 teeth

B 295 MXL
6mm width
pitch 23.06 inches, 599.44mm
295 teeth

http://www.powertechno.co.jp/p_03_06.html
http://www.sdp-si.com/D265/PDF/D265P1003.pdf

z-thread 14mm diameter
2mm pitch

ShinanoKenshin Step Motor

Step Angle 1.8 degrees
Resistance / phase: 2.6 Ohms

1.8°/STEP 2.6Ω
STA-42D3023 (Fake number to hide real details of the motor by Shinano Kenshin)

a4jp-com, 2015-06-10:
Thank you for your advice AnHardt and Wurstnase ^^

I joined the Reprap forum and will ask mechanical questions there and just setting or bug related Marlin questions here if that is okay.

Are there any settings to deal with backlash in Marlin?

a4jp-com, 2015-06-10:
I just changed the numbers from 71.6 to 71.58 and uploaded the firmware and now my LCD screen isn't working. Any ideas on fixing this?

a4jp-com, 2015-06-10:
Can anyone help? I just wanted to set the axis steps, print a test cube then do some 3D work but now my LCD screen isn't working everything is almost impossible (T-T)

AnHardt, 2015-06-10:
Please help us tho help you.
Did you use a "," instead of a "."?
What means: "LCD screen isn't working"? No display at all? No reaction on the nob? "---" instead of coordinates?
What version are you currently on?
Did you define a panel at all?

It's possible to write the calculation instead of the result - to have all digits for the steps.

a4jp-com, 2015-06-10:
This is a copy of the files.
http://a4jp.com/3d-printer-marlin-firmware/

My screen has lights but no words showing at all.
It used to have a startup message and then options. Now it is just blank.

I can use Pronterface to move the head so I didn't completely kill my machine ^^

a4jp-com, 2015-06-10:
The original settings of Marlin 1.0.0, I think. 
[image]

KiteLab, 2015-06-10:
Did the original files worked before?
What did you change? diff is your friend.

KiteLab, 2015-06-10:
Have you disconected the display - and crossed the cables?

a4jp-com, 2015-06-10:
I never tried them before. This is my first time I tried them but they are the files the printer's maker had to send to me when they were told they can't keep the Marlin files secret because of the GPL.

a4jp-com, 2015-06-10:
Changing cables is difficult as the back of the machine has a few tamper safe stickers on it. If I open the back to get to the motherboard I will void the warranty. I hope I can get the screen working again. Everything else seems okay.

KiteLab, 2015-06-10:
If you did not disconnect the display - don't do it - all is well but the software.

a4jp-com, 2015-06-10:
That's okay. I won't disconnect the display. I just need to get it working again and everything is good.

a4jp-com, 2015-06-11:
Is there anyone here that can help me get the display working again? I've been trying for hours and hours but can't figure it out.

a4jp-com, 2015-06-11:
I think I found out the problem. The LCD screen needs the addon for Arduino so the rAMBO motherboard can be selected. I put the folder for the driver in the 1.6.3 hardware folder but the board option didn't show. I'll try Arduino 1.0.5 and see if that works.

a4jp-com, 2015-06-11:
I got the LCD working but can only do it with Arduino 1.0.5. Will this be okay when I update the driver to the newest version 1.0.2 or 1.0.3?

Notes to help others: Open the folder named Marlin with an uppercase M and double click on the Marlin.ino file. Opening the file from Arduino 1.0.5 directly gives you a strange folder name message even though the folder name is correct.
Place the RAMBo folder in the hardware folder of C:\Program Files (x86)\Arduino\hardware. You don't need the README.md file from the Arduino_1.x.x.zip file.

Wackerbarth, 2015-06-23:
If you want to try even "newer" things, you can look at pull request "Prepare for IDE1.6 #2306"
and/or the Development branch.  But, hopefully, you have gotten past your initial issues.
Can we close this thread?

boelle, 2015-06-29:
@a4jp-com can we close this one?

a4jp-com, 2015-06-30:
Yes. 1.0.0 is installed and working. I'm just waiting to install the current version that is going through the bug fix.

closed 2015-06-30 by a4jp-com