MarlinFirmware/Marlin #2279: Remove DELTA_CALIBRATION_MENU?

thinkyhead, 2015-06-11:
Would anyone miss this option? It adds this menu, which is not generalized for all deltas:

```
  static void lcd_delta_calibrate_menu() {
    START_MENU();
    MENU_ITEM(back, MSG_MAIN, lcd_main_menu);
    MENU_ITEM(gcode, MSG_AUTO_HOME, PSTR("G28"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_X, PSTR("G0 F8000 X-77.94 Y-45 Z0"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_Y, PSTR("G0 F8000 X77.94 Y-45 Z0"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_Z, PSTR("G0 F8000 X0 Y90 Z0"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_CENTER, PSTR("G0 F8000 X0 Y0 Z0"));
    END_MENU();
  }
```

thinkyhead, 2015-10-12:
Still needs to be generalized by adding some configuration options.

Roxy-3DPrintBoard, 2015-10-12:
Actually...  My thoughts are it is going to be replaced very soon with real code to assist in physically leveling any bed that is supported by 3-Screw-Mounts with the origin at (0,0).   I'm doing the code to help with my Delta calibration but it has value for Cartesian machines too.  It is actually not difficult to handle the case where the origin is not at the center but I wanted to keep the code small until we had a chance to get feedback on it.

I have some messages that need to be put in the Language.h files but I'm going to be bringing that up and asking for guidance in a separate thread.  (ie.  Do I change each Language.xx.h file and just put English versions of the message in there until somebody that can translate it does the work?  I don't know how to handle that.)

RobotCaleb, 2015-12-14:
I vote to not remove it. (I added it)
I use it fairly regularly as I'm tweaking my printer fairly regularly.

thinkyhead, 2016-02-19:
@RobotCaleb Unfortunately, as implemented this feature is specialized for one specific machine, and it cannot be altered without modifying the code. It will need to be replaced with something more generalized that can apply to _any_ Delta. It should also be more self-explanatory. When a user clicks "Calibrate X" they will expect more than just a movement to some mysterious position. What should they do next? This feature offers no guidance and prints no messages.

```cpp
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_X, PSTR("G0 F8000 X-77.94 Y-45 Z0"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_Y, PSTR("G0 F8000 X77.94 Y-45 Z0"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_Z, PSTR("G0 F8000 X0 Y90 Z0"));
```

As the creator of this feature, would you like to try making it more generalized and submitting a new PR for it…?

closed 2016-04-23 by jbrazio